mbentley/docker-omada-controller #137: Synology Setup

crgable, 2021-11-16:
Do you have any sort of instructions for installation on Docker on a Synology NAS? I have the image set up and even copied the settings recommended here (https://www.epidemiology.tech/virtualization-containers/docker/tp-link-omada-controller-on-docker-in-synology-nas/) but I can't get anything to show up when I try to access the container via IP and port number. 
Can you help me?

crgable, 2021-11-16:
I get a HTTP ERROR 404 when trying to access it...any direction you can provide would be helpful.

christianbormann, 2021-11-16:
I use the following settings on my DS720+ and can access the website via https://[ip-of-synology]:8043

Ports:
[image]

Environment:
[image]

Volumes:
[image]

mbentley, 2021-11-16:
I'm not sure how Synology's implementation of Docker is different.  Might be better if you explain how you're running the container in Synology and how you're accessing it in your browser.

Most likely issue without knowing anything more: Don't try to map the ports to different ports via Docker unless you also set them via the environment variables as explained in the README.

crgable, 2021-11-16:
[image]
Here are my current settings...

When I try to pull up the UI by typing http://192.168.1.2:49232/login I get the 404 error...all the other ports I tried pull up an invalid response. (ERR_SSL_PROTOCOL_ERROR)

mbentley, 2021-11-16:
Yup, that's why.  The ports inside the container (listed in the "Container Port" column) must match the ones exposed on the host (listed in the "Local Port" column).  They must match.  If you do change them, you must use the environment variables that are shown there (the `*_PORT` variables) to match what you map them to.

It's just how the application is written that the behavior works that way.  Not 100% sure why it does that but that's what is required.

closed 2021-11-16 by mbentley

crgable, 2021-11-17:
Ahh...that was it. The container defaults those ports to "auto" which means random numbers get assigned to them. Getting all the ports manually assigned to the same ones in the environmental variables fixed everything. Thank you for your help.

mbentley, 2021-11-17:
Good deal.  I thought the error looked familiar 😄