mbentley/docker-omada-controller #138: Failed to start mongo DB server - Failed to set up listener: SocketException: Permission denied

SlothCroissant, 2021-11-17:
**Describe the bug**
I've used this container previously and it worked great - for some reason however, it's failing to deploy in my docker swarm node setup, and I can't for the life of me figure out why. details below.

**To Reproduce**
Steps to reproduce the behavior:

* Docker Version: `Docker version 20.10.10, build b485636`
* Ubuntu Version: `Ubuntu 20.04.3 LTS`
* Host info: VMWare ESXi 6.7

``` bash
docker volume create omada_data
docker volume create omada_work
docker volume create omada_logs
```

Then, start the stack via Portainer 2.10 (docker-compose.yml is below).

**Expected behavior**
MongoDB starts, binds correctly, and Omada starts using the deployed MongoDB database

**How you're launching your container**

* Via Docker Swarm / Portainer 2.10 and the following Docker Compose file:

``` yml
version: "3.8"

services:
  app:
    image: mbentley/omada-controller
    environment:
      - TZ=America/Chicago
      - SHOW_MONGODB_LOGS=true
    volumes:
      - data:/opt/tplink/EAPController/data
      - work:/opt/tplink/EAPController/work
      - logs:/opt/tplink/EAPController/logs
    ports:
      - 8088:8088
      - 8043:8043
      - 8843:8843
      - 29810:29810
      - 29810:29810/udp
      - 29811:29811
      - 29811:29811/udp
      - 29812:29812
      - 29812:29812/udp
      - 29813:29813
      - 29813:29813/udp
    networks:
      - my-network

networks:
  my-network:
    name: "my-network"
    external: true

volumes:
  data:
  work:
  logs:
```

**Container Logs**
```
omada_app.1.t134z6ludmft@docker02.lan    | INFO: Time zone set to 'America/Chicago'
omada_app.1.t134z6ludmft@docker02.lan    | WARNING: owner or group (1000:1000) not set correctly on '/opt/tplink/EAPController/data'
omada_app.1.t134z6ludmft@docker02.lan    | INFO: setting correct permissions
omada_app.1.t134z6ludmft@docker02.lan    | WARNING: owner or group (1000:1000) not set correctly on '/opt/tplink/EAPController/work'
omada_app.1.t134z6ludmft@docker02.lan    | INFO: setting correct permissions
omada_app.1.t134z6ludmft@docker02.lan    | WARNING: owner or group (1000:1000) not set correctly on '/opt/tplink/EAPController/logs'
omada_app.1.t134z6ludmft@docker02.lan    | INFO: setting correct permissions
omada_app.1.t134z6ludmft@docker02.lan    | INFO: Starting Omada Controller as user omada
omada_app.1.t134z6ludmft@docker02.lan    | tail: cannot open '/opt/tplink/EAPController/logs/mongod.log' for reading: No such file or directory
omada_app.1.t134z6ludmft@docker02.lan    | tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
omada_app.1.t134z6ludmft@docker02.lan    | tail: '/opt/tplink/EAPController/logs/mongod.log' has appeared;  following new file
omada_app.1.t134z6ludmft@docker02.lan    | tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] MongoDB starting : pid=56 port=27217 dbpath=../data/db 64-bit host=01259058d534
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] db version v3.6.3
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] git version: 9586e557d54ef70f9ca4b43c26892cd55257e1a5
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1  11 Sep 2018
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] allocator: tcmalloc
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] modules: none
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] build environment:
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten]     distarch: x86_64
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten]     target_arch: x86_64
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 E STORAGE  [initandlisten] Failed to set up listener: SocketException: Permission denied
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] now exiting
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:03:00.664-0600 I CONTROL  [initandlisten] shutting down with code:48
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17 13:03:00 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17 13:03:00 [main] [INFO]-[SourceFile:91] - going to start local mongod.
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17 13:04:40 [main] [WARN]-[SourceFile:151] - Fail to start mongo DB server
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17 13:04:40 [main] [WARN]-[SourceFile:78] - com.tplink.omada.start.a.a: Fail to start mongo DB server
omada_app.1.t134z6ludmft@docker02.lan    | com.tplink.omada.start.a.a: Fail to start mongo DB server
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.b.d.a(SourceFile:152) ~[omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.b.b.e(SourceFile:75) [omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.OmadaBootstrap.d(SourceFile:224) [omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.OmadaBootstrap.r(SourceFile:309) [omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.OmadaBootstrap.a(SourceFile:101) [omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.OmadaBootstrap.e(SourceFile:246) [omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.OmadaLinuxMain.b(SourceFile:80) [omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    |      at com.tplink.omada.start.OmadaLinuxMain.main(SourceFile:25) [omada-start.jar:?]
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.754-0600 I CONTROL  [main] ***** SERVER RESTARTED *****
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] MongoDB starting : pid=88 port=27217 dbpath=../data/db 64-bit host=01259058d534
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] db version v3.6.3
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] git version: 9586e557d54ef70f9ca4b43c26892cd55257e1a5
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1  11 Sep 2018
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] allocator: tcmalloc
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] modules: none
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] build environment:
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten]     distarch: x86_64
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten]     target_arch: x86_64
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 E STORAGE  [initandlisten] Failed to set up listener: SocketException: Permission denied
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] now exiting
omada_app.1.t134z6ludmft@docker02.lan    | 2021-11-17T13:04:42.761-0600 I CONTROL  [initandlisten] shutting down with code:48

```

**Additional context**
Logs above recycle every few minutes as the app keeps trying to start, but it never does start.

mbentley, 2021-11-17:
Hmm, could you tell me about the network that you have created that is external?  How is it created and with what driver?

mbentley, 2021-11-18:
I see a lot of details from searching about `Failed to set up listener: SocketException: Permission denied` regarding permissions on `/tmp`.

Could you exec into the container and get me the output of `ls -lan /tmp` in the container?

SlothCroissant, 2021-11-19:
Hey there @mbentley, thanks for taking a look. 

First off, I don't use zfs (any more, at least) - that's a hold-over from a previous build that was using ZFS, but since I moved into ESXi and got rid of that specific ZFS deployment, I just kept the mount point the same for the new virtual disk. Sorry for the confusion there :)

As far as /tmp is concerned, here's the `ls` output - not sure if there is supposed to be anything in there on deployment, but alas:

```
ryanb@docker02:~$ docker exec -it omada_app.1.y87876mumey6hbaxsy0nfargk ls -lan /tmp
total 8
drwxr-xr-t 2 1000 1000 4096 Nov 12 21:28 .
drwxr-xr-x 1    0    0 4096 Nov 19 07:50 ..
```

mbentley, 2021-11-19:
Sure thing and no worries, I know how that goes.  Sometimes changing a persistent data path name is more trouble that it is worth :)

Something is a bit off; not exactly sure what's going on.  Could you give this a run and grab the output for me?

```
docker exec -it -u omada omada-controller bash -c "ls -lan /tmp; touch /tmp/test; ls -lan /tmp; rm /tmp/test; ls -lan /tmp"
```

For reference on mine:
```
# docker exec -it -u omada omada-controller bash -c "ls -lan /tmp; touch /tmp/test; ls -lan /tmp; rm /tmp/test; ls -lan /tmp"
total 0
drwxrwxrwt 1   0   0 79 Nov 19 09:06 .
drwxr-xr-x 1   0   0 64 Nov  7 04:25 ..
drwxr-xr-x 2 508 508 15 Nov  7 04:25 hsperfdata_omada
drwxr-xr-x 2   0   0  6 Nov  7 04:25 hsperfdata_root
srwx------ 1 508 508  0 Nov  7 04:25 mongodb-27217.sock
total 0
drwxrwxrwt 1   0   0 91 Nov 19 09:07 .
drwxr-xr-x 1   0   0 64 Nov  7 04:25 ..
drwxr-xr-x 2 508 508 15 Nov  7 04:25 hsperfdata_omada
drwxr-xr-x 2   0   0  6 Nov  7 04:25 hsperfdata_root
srwx------ 1 508 508  0 Nov  7 04:25 mongodb-27217.sock
-rw-r--r-- 1 508 508  0 Nov 19 09:07 test
total 0
drwxrwxrwt 1   0   0 79 Nov 19 09:07 .
drwxr-xr-x 1   0   0 64 Nov  7 04:25 ..
drwxr-xr-x 2 508 508 15 Nov  7 04:25 hsperfdata_omada
drwxr-xr-x 2   0   0  6 Nov  7 04:25 hsperfdata_root
srwx------ 1 508 508  0 Nov  7 04:25 mongodb-27217.sock
```

Is there any chance that there are any specific Docker settings set for the daemon startup or maybe something in the `/etc/docker/daemon.json`?  The output of `docker info` should tell me what I might possible need to know in terms of maybe there being something strange going on.

SlothCroissant, 2021-11-19:
Sure thing. This looks to be interesting (looks like the permissions are from the host's 1000:1000):

```
ryanb@docker02:~$ docker exec -it -u omada omada_app.1.i41239ovnrgxzy5vyz78fc31q bash -c "ls -lan /tmp; touch /tmp/test; ls -lan /tmp; rm /tmp/test; ls -lan /tmp"
total 8
drwxr-xr-t 2 1000 1000 4096 Nov 12 21:28 .
drwxr-xr-x 1    0    0 4096 Nov 19 08:09 ..
touch: cannot touch '/tmp/test': Permission denied
total 8
drwxr-xr-t 2 1000 1000 4096 Nov 12 21:28 .
drwxr-xr-x 1    0    0 4096 Nov 19 08:09 ..
rm: cannot remove '/tmp/test': No such file or directory
total 8
drwxr-xr-t 2 1000 1000 4096 Nov 12 21:28 .
drwxr-xr-x 1    0    0 4096 Nov 19 08:09 ..
```

`docker info` output:

```
ryanb@docker02:~$ docker info
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Build with BuildKit (Docker Inc., v0.6.3-docker)
  scan: Docker Scan (Docker Inc., v0.9.0)

Server:
 Containers: 15
  Running: 11
  Paused: 0
  Stopped: 4
 Images: 29
 Server Version: 20.10.10
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: active
  NodeID: 2d5ifinwbuleourtosgf3rq5v
  Is Manager: true
  ClusterID: vudbl3guuj8ojot43hi2glcwl
  Managers: 1
  Nodes: 1
  Default Address Pool: 172.19.0.0/16
  SubnetSize: 24
  Data Path Port: 4789
  Orchestration:
   Task History Retention Limit: 5
  Raft:
   Snapshot Interval: 10000
   Number of Old Snapshots to Retain: 0
   Heartbeat Tick: 1
   Election Tick: 10
  Dispatcher:
   Heartbeat Period: 5 seconds
  CA Configuration:
   Expiry Duration: 3 months
   Force Rotate: 0
  Autolock Managers: false
  Root Rotation In Progress: false
  Node Address: 10.1.10.9
  Manager Addresses:
   10.1.10.9:2377
 Runtimes: io.containerd.runtime.v1.linux runc io.containerd.runc.v2
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 5b46e404f6b9f661a205e28d59c982d3634148f8
 runc version: v1.0.2-0-g52b36a2
 init version: de40ad0
 Security Options:
  apparmor
  seccomp
   Profile: default
 Kernel Version: 5.4.0-65-generic
 Operating System: Ubuntu 20.04.3 LTS
 OSType: linux
 Architecture: x86_64
 CPUs: 8
 Total Memory: 7.774GiB
 Name: docker02.lan
 ID: FCDY:JSCU:S4H3:EWTX:7RVA:XTY7:DWKW:ULUE:ASWK:7XH2:OM4M:DXRP
 Docker Root Dir: /mnt/zfs/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

WARNING: No swap limit support
```

Only thing interesting I've done is move my docker root dir from the usual `/var/lib/docker` to `/mnt/zfs/docker` (different virtual disk on the host). Not sure if that could impact the /tmp folder within the container, though.

mbentley, 2021-11-19:
Could you update your compose file with this image: `mbentley/omada-controller:4.4-tmp` and give it a shot?

I just added a bit to the entrypoint script to check and change permission on /tmp in the container if needed via https://github.com/mbentley/docker-omada-controller/pull/139.

Let me know if that works for you and I'll merge that and get it in the regular image.

mbentley, 2021-11-19:
And just to mention, I don't see anything unexpected about your output of your `docker info`.  Only thing I can think of could be something with permissions somewhere up the directory tree that might have had an impact on the subdirectories of `/mnt/zfs/docker` but I would think you would be hitting plenty of other issues for anything else that is running as well if that was the case.  I'm running the same version of Docker as you but just on Debian.

SlothCroissant, 2021-11-19:
> Could you update your compose file with this image: `mbentley/omada-controller:4.4-tmp` and give it a shot?

Worked a charm!

> I would think you would be hitting plenty of other issues for anything else that is running as well if that was the case.

That's the bit that confuses me about this too - I'm running ~10 swarm services without issues, aside from this one. But I'm a networking guy, not a storage/linux guy, so I'm generally at a loss for what would cause this.

Thanks for your quick help! Deployment via `4.4-tmp` worked, and I'm up and running.

closed 2021-11-19 by SlothCroissant