mbentley/docker-omada-controller #141: Omada controller docker won't start

nexxusty, 2021-12-10:
I'm having an issue with a Raspberry Pi 4 where docker won't start. Every other container I've tried has worked without issue so at this point I do think it's the Omada controllers problem.

Every version for arm 64 and arm V7 give me the same error which is as follows:

Containers: start omada-controller...code:400 OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: process_linux.go:508: setting cgroup config for procHooks process caused: open /sys/fs/cgroup/docker/2bb1b44e43bf7dcd321febc2219730648dc22dc2fd2f8c696e5949d48709ccb7/io.weight: no such file or directory: unknown

I have tried pretty much everything I can think of at this point so any help would be appreciated.

mbentley, 2021-12-10:
Could you provide your run command or however you're starting the container?

mbentley, 2021-12-20:
Feel free to comment if you continue to have problems.

closed 2021-12-20 by mbentley