mbentley/docker-omada-controller #143: Log4Shell

noelhibbard, 2021-12-11:
Can we get this docker updated to mitigate the new Log4Shell exploit? Should be as easy as adding a command line argument when launching the jar. If you haven't read up in this exploit I urge you to. Everyone in IT needs to take this one serious.

mbentley, 2021-12-11:
Yup, I'll work on that today at some point.  I saw a [post on Reddit](https://reddit.com/r/TPLink_Omada/comments/rdn33i/log4j_exploit/) about how to work around this for now until it is patched.

noelhibbard, 2021-12-11:
For now I simply added this to my docker-compose:

```yaml
environment:
            - LOG4J_FORMAT_MSG_NO_LOOKUPS=true
```

noelhibbard, 2021-12-11:
https://canarytokens.org/generate

This site generates a string which you can paste into the user/password field in Omada (or any app that you think logs) and if the exploit works you will get an email. 

I reverted the simple env var fix I did on my docker-compose and then ran this test and didn't get an email. You can see the bad password attempt (including the string) in the log but it didn't successfully execute. So maybe the OpenJDK used in this container has lookups disable out of the box. 

I'd still like to upgrade log4j to be safe.

mbentley, 2021-12-12:
I submitted #144 which will fix this by patching log4j on all versions of the controller.  Builds are going through now.  I've tested on arm64 for all versions that support arm64 since I am on my m1 macbook right now.

mbentley, 2021-12-12:
I am going to re-open this until I can do some testing with the different versions to verify that they are not vulnerable.  I backported it to all versions published, hoping it just works on the older amd64 builds.

mbentley, 2021-12-12:
OK, I've done some further testing and it looks like things are good.  Older versions look like they use slf4j anyway but they all had the same version (2.8.2) of log4j across all versions of the controller.

mbentley, 2021-12-12:
I was asking some people on the [Reddit thread](https://www.reddit.com/r/TPLink_Omada/comments/rdn33i/log4j_exploit/) if  anyone had been able to successfully exploit a controller previously as I would just like to be able to see a before/after.[](url)

mbentley, 2021-12-12:
> For now I simply added this to my docker-compose:
> 
> ```yaml
> environment:
>   - LOG4J_FORMAT_MSG_NO_LOOKUPS=true
> ```
> 
> I'm just not confident enough in my work around. I'd rather get the updated log4j in here like in the Reddit link you posted. 
> 
> I also need to fire up an LDAP server to use for testing patches. Simply finding everything that uses log4j is going to be a challenge. What a nightmare this is going to be. 

By the way, you probably don't need an actual LDAP server to test it; Im pretty sure you can just run an app that will accept a handshake and log the source to give you what you're looking for.  Pretty sure that's what many of those testing sites are doing.

noelhibbard, 2021-12-12:
Yeah that is all I was going to do. There are a ton of test rigs like this on GitHub already. Canarytokens and a few others I've seen only look for a DNS query which isn't conclusive. I found another site that looks for an actual LDAP connection, or at least the handshake, which is more conclusive.

NathanAdhitya, 2021-12-13:
@mbentley  The new log4shell patch is out. It is beta though. Consider adding it as a beta in the tags? https://community.tp-link.com/en/business/forum/topic/514452

mbentley, 2021-12-13:
Do you know if they publish beta versions anywhere besides this one off for the vulnerability?  Just trying to see how much it is worth the effort.

mbentley, 2021-12-16:
Thanks for that @Tu13es .  Your fix for sha512sum is correct.  They changed the outputted format of the .sha512 file.  I submitted and merged https://github.com/mbentley/docker-omada-controller/pull/145 which bumps the log4j version and fixes the digest verification.

Tu13es, 2021-12-18:
Hm, there's a log4j 2.17.0 out now to fix another vulnerability and the sha512 format has changed to the previous format.  Reverting to the original line resolves the issue.  🙃

mbentley, 2021-12-19:
lol impressive...  I will look at it in a bit.

mbentley, 2021-12-20:
OK, I updated the patching script to automatically retrieve the latest version.  I also didn't realize that the mirrors only have the latest version so I updated it to download from the Apache archive.  They have a daily 5 GB download limit but the log4j binaries are tiny compared to that.

mbentley, 2021-12-27:
I think it is probably ok to go ahead and close this.  If anyone hears about yet another issue, let me know.

closed 2021-12-27 by mbentley

mbentley, 2022-01-07:
Dropping this in this issue: https://issues.apache.org/jira/browse/LOG4J2-3325

I keep having to add multiple checks to figure out how they're writing their checksums as it is now on the third iteration from 2.16.0, 2.17.0 and 2.17.1:

```
# https://archive.apache.org/dist/logging/log4j/2.16.0/apache-log4j-2.16.0-bin.tar.gz.sha512
4d4f44caa138224080a91cedf99d3e207a1ea700382f7a20d2f9c665d3f45a54f8bab826579bf6a35c19d3409ea4f27145fe511b9efdfb2afb9a6b9eb60ecc74  apache-log4j-2.16.0-bin.tar.gz

# https://archive.apache.org/dist/logging/log4j/2.17.0/apache-log4j-2.17.0-bin.tar.gz.sha512
apache-log4j-2.17.0-bin.tar.gz: BB070B6A E01EE6D8 C4979830 902E6DE1 9A4AE84B
                                C1EEFC0D 99654A1D 6E66E9BD B75410CC D73DE079
                                74042C88 92C51720 D8C7679C 80B6A1D3 FC30C5C7
                                06493A2C

# https://archive.apache.org/dist/logging/log4j/2.17.1/apache-log4j-2.17.1-bin.tar.gz.sha512
b7e948df6c6f57d903d990de2cc0270c1537b711285e9b6b91280db6ace38418fced713785b2c20512dd9a4238c2d1d0ceb414d9936df2ca110ff14993ae04dc
```

commit 3d8b9cba (2021-12-12): Merge pull request #144 from mbentley/log4j-patch

Added log4j patch to all versions; fixes #143
commit 63dcc333 (2021-12-12): Added log4j patch to all versions; fixes #143