mbentley/docker-omada-controller #152: Is the hostname for the certificate defined anywhere in the container?

physhster, 2021-12-21:
I've been having issues getting the UI to load in Chrome, most likely because the CN for the cert is 'localhost'. It works in Safari but it's a bit annoying.

Is the hostname defined anywhere in the container, if so, could it be changed to anything else?

[image]

mbentley, 2021-12-21:
By default, the image uses it's own self-signed certificate that was generated as a part of the software build from TP-Link.  I'm able to spin up a controller and use the default certificate by accepting the certificate in Chrome:
[image]

Could you provide the run command you're using?  Anything show up when you open up Developer Tools in Chrome and watch the calls on the Network tab similar to this:
[image]

I suppose that we could generate a self-signed certificate with the appropriate common name but I would have to script it out and add some more logic.  Wouldn't be to terrible to do.

physhster, 2021-12-21:
The error message in Chrome is somewhat unusual... And I cannot whitelist it.
I deployed the container as is, so my run command is whatever is in the DockerFile, made no changes aside from port mappings. Only Chrome seems bothered by the certificate.

This doesn't warrant much troubleshooting if it's outside of the container config, I was just hoping for a place to set the hostname easily and regenerate the cert.

[image]

physhster, 2021-12-21:
Feels like it's Synology's incomplete docker implementation again. Sigh. Sorry about the noise.

mbentley, 2021-12-21:
All good.  If generating a self-certificate with a correct/valid CN would be helpful, let me know and I can see about adding that to the entrypoint and by environment variable.

physhster, 2021-12-21:
I think maybe explicitly generating a cert with a CN and a name that's not localhost could be useful, but my problem could be an edge case so I am not sure it's worth your time.

physhster, 2023-05-21:
As browsers become more and more hardened, it's becoming a bit of a blocking issue. Neither of Chrome, Edge, or Firefox can load the Omada UI, even with exceptions in place for the localhost certificate. Can we revisit switching the default cert not being 'localhost'? I'm pretty sure anything else will do. Thanks

physhster, 2023-05-21:
FWIW, it seems disabling "HTTP redirect to HTTPS" in the controller settings mitigates this issue, while still redirecting HTTP to HTTPS. I'm quite confused.

mbentley, 2023-05-21:
Unfortunately I don’t have the ability to make that change effectively for everyone in a way that it will be guaranteed to work in the future. It needs to be done properly at the app level upstream in the software since it’s generating the certs.

physhster, 2023-05-21:
I guess I'll live with the "HTTP redirect to HTTPS" disabled workaround.

closed 2023-05-21 by physhster