mbentley/docker-omada-controller #155: New Release - Add v5 for Linux

mbentley, 2022-01-07:
https://static.tp-link.com/upload/software/2022/202201/20220107/Omada_SDN_Controller_v5.0.29_linux_x64.tar.gz

closed 2022-01-07 by mbentley

mbentley, 2022-01-07:
Images have been built and are up on Docker Hub.  Please give them a test (making sure to take backups if you do upgrade to test) as there are some differences in how the app is run in 5.0.x since it is using springboot.  I think I caught all of the differences from the `install.sh` script but it would be good to get more eyes on it.  I've only done some quick smoke tests on the amd64 and arm64 builds but I assume that the armv7l build works as well.

bburan, 2022-01-07:
I accidentally updated to this version (I had to restart my docker container and was using the `latest` tag). It's failing to start. I get the following log:
```
Starting omada-controller ... done
Attaching to omada-controller
omada-controller    | INFO: Time zone set to 'Etc/UTC'
omada-controller    | INFO: Setting 'portal.https.port' to 8043 in omada.properties
omada-controller    | INFO: Re-enabling TLS 1.0 & 1.1
omada-controller    | INFO: Starting Omada Controller as user omada
omada-controller    | 01-07-2022 18:08:19.631 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(32): success to load configuration omada.properties
omada-controller    | 01-07-2022 18:08:19.660 INFO [main] [] c.t.s.o.s.OmadaBootstrap(79): going to start local mongod.
omada-controller    | 01-07-2022 18:09:59.729 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Fail to start mongo DB server
omada-controller    | 01-07-2022 18:09:59.730 WARN [main] [] c.t.s.o.s.s.b(145): Fail to start mongo DB server
omada-controller    | 01-07-2022 18:09:59.730 WARN [main] [] c.t.s.o.s.s.a(77): com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
omada-controller    | com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
omada-controller    |   at com.tplink.smb.omada.starter.server.b.a(SourceFile:146) ~[local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.server.a.e(SourceFile:74) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.d(SourceFile:218) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.q(SourceFile:290) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:89) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:245) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:84) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:36) [local-starter-5.0.29.jar:5.0.29]
omada-controller    | 01-07-2022 18:11:41.820 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Fail to start mongo DB server
omada-controller    | 01-07-2022 18:11:41.821 WARN [main] [] c.t.s.o.s.s.b(145): Fail to start mongo DB server
omada-controller    | 01-07-2022 18:11:41.821 WARN [main] [] c.t.s.o.s.s.a(77): com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
omada-controller    | com.tplink.smb.omada.starter.b.a: Fail to start mongo DB server
omada-controller    |   at com.tplink.smb.omada.starter.server.b.a(SourceFile:146) ~[local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.server.a.e(SourceFile:74) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.d(SourceFile:218) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.n(SourceFile:211) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.q(SourceFile:293) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.a(SourceFile:89) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaBootstrap.e(SourceFile:245) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaLinuxMain.a(SourceFile:84) [local-starter-5.0.29.jar:5.0.29]
omada-controller    |   at com.tplink.smb.omada.starter.OmadaLinuxMain.main(SourceFile:36) [local-starter-5.0.29.jar:5.0.29]
omada-controller    | 01-07-2022 18:11:43.823 INFO [mongo-check-thread] [] c.t.s.o.s.OmadaBootstrap(315): mongod was shutdown, restarting it
omada-controller    |
omada-controller    |   .   ____          _            __ _ _
omada-controller    |  /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
omada-controller    | ( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
omada-controller    |  \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
omada-controller    |   '  |____| .__|_| |_|_| |_\__, | / / / /
omada-controller    |  =========|_|==============|___/=/_/_/_/
omada-controller    |  :: Spring Boot ::                (v2.4.3)
omada-controller    |
omada-controller    | 01-07-2022 18:12:13.006 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(55): Starting OmadaLinuxMain v5.0.29 using Java 1.8.0_312 on nutshell with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.0.29.jar started by omada in /opt/tplink/EAPController/lib)
omada-controller    | 01-07-2022 18:12:13.024 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(660): No active profile set, falling back to default profiles: default
omada-controller    | 01-07-2022 18:12:23.267 WARN [main] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
omada-controller    | 01-07-2022 18:12:54.113 WARN [main] [] c.t.s.o.s.a.b(81): Mongo Client connect error while init Jetty.com.mongodb.MongoTimeoutException: Timed out after 30000 ms while waiting to connect. Client view of cluster state is {type=UNKNOWN, servers=[{address=127.0.0.1:27217 [... cut ...] TimeoutException: Timed out after 30000 ms while waiting to connect. Client view of cluster state is {type=UNKNOWN, servers=[{address=127.0.0.1:27217, type=UNKNOWN, state=CONNECTING, exception={com.mongodb.MongoSocketOpenException: Exception opening socket}, caused by {java.net.ConnectException: Connection refused (Connection refused)}}]
omada-controller    | 01-07-2022 18:13:26.810 INFO [main] [] c.t.s.o.s.t.SpringBootStartUpTask(38): Try to reset default certificate and restart springBoot.
```

mbentley, 2022-01-07:
Besides switching to a specific tag and not using latest, could you either add the environment variable `-e SHOW_MONGODB_LOGS=true` so that it will show the mongodb logs or grab the logs from `/opt/tplink/EAPController/logs/mongod.log`?  A `docker cp <container-id-or-name>:/opt/tplink/EAPController/logs/mongod.log mongod.log` would do it.

bburan, 2022-01-07:
@mbentley The logs gave me a clue. There was a permission issue. After fixing it, all is good. Somehow the ownership of one of the files in the docker volume got set to a nonexistent user (likely related to uninstalling/reinstalling docker). Sorry for the trouble! v5 looks great!

Vendo232, 2022-01-07:
Sorry, just opened a new issue before seeing this. My docker won`t start as well, my log is short

```

INFO: Time zone set to 'Etc/UTC'

INFO: Starting Omada Controller as user omada

Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
```

mbentley, 2022-01-07:
No problem @Vendo232.  It's probably best to keep issues separate actually as it will keep things cleaner.  Will respond over in https://github.com/mbentley/docker-omada-controller/issues/160

commit fcee5d05 (2022-01-07): Added 5.0.29; update README; fixes #155
commit 0698495d (2022-01-07): Merge pull request #159 from mbentley/add-5.0

Added 5.0.29; update README; fixes #155