mbentley/docker-omada-controller #160: V5 won`t start

Vendo232, 2022-01-07:
**Describe the bug**

INFO: Time zone set to 'Etc/UTC'

INFO: Starting Omada Controller as user omada

Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain

**To Reproduce**
Steps to reproduce the behavior:

recreated omada docker in portainer and selected use latest revision ( container upgrade process used before ) 
1. foo
2. bar
3. baz
4. qux

**Expected behavior**

Docker would not start

**How you're launching your container**

in portainer by selecting START

**Container Logs**

above

```
logs
go
here
```

**Additional context**

mbentley, 2022-01-07:
I would not recommend running `latest` for the tag - pick a `major.minor` tag and check the changelogs before upgrading to make sure there aren't breaking changes, like when they released 4.x.  You should be able to switch to the `4.4` tag and you'll stay on the 4.4 release.

I'm guessing you're using some sort of appliance like a Synology or something.  It's probably filling in the CMD.  I've had other reports in the past where the CMD of the image changed but it was hard coded in the container to specify a specific command to run even though the user didn't specify it.

The error you're seeing looks like it is using the previous CMD value from 4.4:
https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.v4.4.x#L29

When the image for 5.0 changed the main class:
https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.v5.0.x#L29

If you want to upgrade to 5.0, change the image tag to 5.0 and re-run the container, making sure that it actually re-creates it, otherwise just switch your tag from latest to 4.4 and it'll start right back up.

Vendo232, 2022-01-07:
thank you Sir, changing mbentley/omada-controller:latest to mbentley/omada-controller:5.0 fixed the issue.

Vendo232, 2022-01-07:
I switched to use "latest" because it is convenient to re-create with hit of one button new container revision in portainer.

changing revision in command requires stopping/deleting and re-entering the whole command with each update. Not a big deal, just clarifying use of "latest"

closed 2022-01-07 by Vendo232