mbentley/docker-omada-controller #161: Not starting up after upgrade from 4.4.8 to 5.0.29

rman18, 2022-01-08:
**Describe the bug**

**To Reproduce**
I'm using Synology docker - I'm upgrading from 4.4.8 to the latest - I tried just clearing the original container but that caused a constant crash, I saw on reddit someone deleted the container and tried again so I tried that and it looks like it almost starts before the socket closed error, then it just sits like that in the logs.  I've let it sit for awhile (the container is still running) but it never starts back up.   

**Expected behavior**

**How you're launching your container**

**Container Logs**

```
01-08-2022 01:23:17.295 INFO [main] [] c.t.s.o.c.o.a.b(32): success to load configuration omada.properties
01-08-2022 01:23:17.352 INFO [main] [] c.t.s.o.s.OmadaBootstrap(79): going to start local mongod.
01-08-2022 01:23:21.371 INFO [main] [] c.t.s.o.s.s.b(161): mongodb process id is 35
01-08-2022 01:23:21.373 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Mongo DB server started
01-08-2022 01:23:21.374 INFO [main] [] c.t.s.o.s.s.b(142): Mongo DB server started
01-08-2022 01:23:24.137 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(55): Starting OmadaLinuxMain v5.0.29 using Java 1.8.0_312 on mbentley-omada-controller2 with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.0.29.jar started by omada in /opt/tplink/EAPController/lib)
01-08-2022 01:23:24.142 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(660): No active profile set, falling back to default profiles: default
01-08-2022 01:23:35.219 WARN [main] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
01-08-2022 01:23:40.068 WARN [main] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
01-08-2022 01:23:46.338 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(61): log mq consume task is start...
01-08-2022 01:23:46.344 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(61): Started OmadaLinuxMain in 23.301 seconds (JVM running for 35.18)
01-08-2022 01:23:46.707 INFO [main] [] c.t.s.o.s.d.a(78): Db unchanged. No need to compatible or init data.
01-08-2022 01:23:46.932 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(149): succeed get default omadac OmadacVO(id=458b3104bf13d8e2803a540cb1f2e96c, name=Omada Controller_F2E96C)
01-08-2022 01:23:48.973 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(124): upgrade workGroup core thread num is 2, max thread num is 2
01-08-2022 01:23:49.016 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(151): upgradeSendReq workGroup core thread num is 2, max thread num is 2
01-08-2022 01:23:49.206 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(97): manager workGroup core thread  num is 16, max thread num is 16
01-08-2022 01:23:50.854 INFO [main] [] c.t.s.e.s.c.c(152): start schedule remove expire device... period = 10
01-08-2022 01:23:50.863 INFO [main] [] c.t.s.e.s.c.c(145): update rateLimiterCache, permitsPerSecond = 5.0
01-08-2022 01:23:52.103 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(70): adopt workGroup core thread num is 8, max thread num is8
01-08-2022 01:23:52.776 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(43): discovery workGroup core thread num is 2, max thread num is 10
01-08-2022 01:23:55.130 INFO [main] [] c.t.s.o.p.b.a.c(72): DisconnectRequestServer start
01-08-2022 01:23:55.523 INFO [main] [] c.t.s.o.m.c.d.m.s.d.f(204): Add Dst Schedule true for Site Default on 1647147600000.
01-08-2022 01:23:55.915 INFO [main] [] c.t.s.o.d.g.a(124): discovery Port is 29810.
01-08-2022 01:23:55.916 INFO [main] [] c.t.s.o.d.g.a(125): manage    port v1 is 29811.
01-08-2022 01:23:55.916 INFO [main] [] c.t.s.o.d.g.a(126): manage    port v2 is 29814.
01-08-2022 01:23:55.916 INFO [main] [] c.t.s.o.d.g.a(127): adopt     port v1 is 29812.
01-08-2022 01:23:55.917 INFO [main] [] c.t.s.o.d.g.a(128): upgrade   port is 29813.
01-08-2022 01:23:56.019 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyUdpServer bind /0.0.0.0:29810, export localhost/127.0.0.1:29810
01-08-2022 01:23:56.036 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29812 global traffic shaping, writeLimit: 518979584, readLimit: 0, checkInterval: 1000, max wait time: 15000
01-08-2022 01:23:56.053 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29812, export localhost/127.0.0.1:29812
01-08-2022 01:23:56.054 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29811 global traffic shaping, writeLimit: 518979584, readLimit: 0, checkInterval: 1000, max wait time: 15000
01-08-2022 01:23:56.056 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
01-08-2022 01:23:56.058 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 518979584, readLimit: 0, checkInterval: 1000, max wait time: 15000
01-08-2022 01:23:56.060 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
01-08-2022 01:23:56.063 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 518979584, readLimit: 0, checkInterval: 1000, max wait time: 15000
01-08-2022 01:23:56.069 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
01-08-2022 01:23:56.069 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Device connector server started.
01-08-2022 01:23:56.070 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Omada Controller started
01-08-2022 01:23:56.070 INFO [main] [] c.t.s.o.s.t.SuccessExitTask(23): Omada Controller started
01-08-2022 01:24:03.846 INFO [Thread-7] [] c.t.e.c.c.CloudClient(100): CloudClient is started.
01-08-2022 01:24:03.856 INFO [Thread-7] [] c.t.e.c.c.CloudClient(110): Connect SEF server automatically.
01-08-2022 01:24:04.838 INFO [Thread-7] [] c.t.e.c.c.n(194): The result of connection is true, service host is n-devs-smb.tplinkcloud.com, port is 443.
01-08-2022 01:24:04.839 INFO [Thread-7] [] c.t.e.c.c.CloudClient(114): Close connection to SEF server.
01-08-2022 01:24:04.843 INFO [Thread-10] [] c.t.e.c.c.n(403): javax.net.ssl.SSLException: Socket closed
javax.net.ssl.SSLException: Socket closed
	at sun.security.ssl.Alert.createSSLException(Alert.java:127) ~[?:1.8.0_312]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:324) ~[?:1.8.0_312]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:267) ~[?:1.8.0_312]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:262) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketImpl.handleException(SSLSocketImpl.java:1563) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketImpl.access$400(SSLSocketImpl.java:73) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:973) ~[?:1.8.0_312]
	at com.tplink.eap.cloudsdk.client.p.a(SourceFile:120) ~[cloudsdk-1.0.7.jar:?]
	at com.tplink.eap.cloudsdk.client.o.run(SourceFile:397) [cloudsdk-1.0.7.jar:?]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_312]
Caused by: java.net.SocketException: Socket closed
	at java.net.SocketInputStream.socketRead0(Native Method) ~[?:1.8.0_312]
	at java.net.SocketInputStream.socketRead(SocketInputStream.java:116) ~[?:1.8.0_312]
	at java.net.SocketInputStream.read(SocketInputStream.java:171) ~[?:1.8.0_312]
	at java.net.SocketInputStream.read(SocketInputStream.java:141) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketInputRecord.read(SSLSocketInputRecord.java:464) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketInputRecord.bytesInCompletePacket(SSLSocketInputRecord.java:68) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketImpl.readApplicationRecord(SSLSocketImpl.java:1341) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketImpl.access$300(SSLSocketImpl.java:73) ~[?:1.8.0_312]
	at sun.security.ssl.SSLSocketImpl$AppInputStream.read(SSLSocketImpl.java:957) ~[?:1.8.0_312]
	... 3 more
01-08-2022 01:24:04.857 INFO [Thread-10] [] c.t.e.c.c.n(407): ssl recv thread is stopped.
[... cut ...]
01-08-2022 01:24:16.842 WARN [https-jsse-nio-8043-exec-7] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
01-08-2022 01:30:50.732 INFO [vpn-certificate-clean-workgroup-0] [] c.t.s.o.m.c.d.m.s.g.a.a(98): VpnCertificateClearTask handling omadacId: OmadacId(458b3104bf13d8e2803a540cb1f2e96c)

```

**Additional context**
I tried the latest tag as well as the 5.0 tag - they stop at the same point.

rman18, 2022-01-08:
I changed the network to "use the same network as the Docker Host" and it started working - must be a port conflict.   Sorry for the false alarm.   Thank you!

closed 2022-01-08 by rman18

mbentley, 2022-01-08:
Just in case someone else comes across this; the logs above actually show a successful startup.  The "SSLException" socket error is something I actually normally see after it communicates to the TP-Link cloud connection.  It seems like it doesn't properly close the initial connection for some reason.  If you see `Mongo DB server started` and then `Omada Controller started`, it should be up and running.