mbentley/docker-omada-controller #162: Raspberry Pi Zero

lancer199135, 2022-01-08:
Getting an "exit code 139" and some research is showing it's a memory error. Going to assume a zero isn't going to handle this? I installed arm7 version in default, no port modification and such.

mbentley, 2022-01-08:
Assuming it is a 512 MB of RAM Pi Zero (not sure what else there might be), you're going to struggle to get it running.  You might be able to try to significantly limit the JVM's memory by updating the CMD.  For example, set the command to something like this for the 5.0 version:

`/usr/bin/java -server -Xms128m -Xmx256m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain`

But I have no idea if that will actually run with being limited to 256m for the JVM and I do not know exactly how much memory MongoDB wants to take up.

lancer199135, 2022-01-09:
Looks like it isn't going to work. Thanks for trying to help me. Was looking to remove Omada controller from my Home Assistant raspberry pi because it flags my install as unsupported. Maybe someday RPi's will come back in stock.

closed 2022-01-09 by lancer199135