mbentley/docker-omada-controller #165: Not starting up after upgrade from 4.4.8 to 5.0.29

ilexpl01, 2022-01-08:
Hi,

I was running 4.4.8 version for weeks - no problems at all.

Went for an upgrade to 5.0.29 today and below the log:

<html>
<body>
<h2 style="text-align: center; color: rgb(0, 0, 0); font-family: &quot;Lucida Console&quot;, Consola, Monaco, monospace, &quot;Courier New&quot;, Courier, serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">omada-controller</h2>

date | stream | content
-- | -- | --
2022-01-08 20:16:38 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:16:38 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:16:38 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:15:37 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:15:36 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:15:36 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:14:34 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:14:34 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:14:34 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:13:41 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:13:40 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:13:40 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:13:14 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:13:13 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:13:13 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:58 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:57 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:57 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:50 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:50 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:50 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:46 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:46 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:46 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:43 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:42 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:42 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:39 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:38 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:38 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:36 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:36 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:36 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:35 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:34 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:34 | stdout | INFO: Time zone set to 'Etc/UTC'
2022-01-08 20:12:33 | stdout | Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
2022-01-08 20:12:22 | stdout | INFO: Starting Omada Controller as user omada
2022-01-08 20:12:22 | stdout | INFO: Time zone set to 'Etc/UTC'

</body>
</html>

Any hint ?

mbentley, 2022-01-08:
Completely recreate the container, keeping your persistent data. I bet you're running a NAS like a Synology.  The implementation stores the CMD from the old image in 4.4 and doesn't update it from the image.

ilexpl01, 2022-01-08:
You did bet well :)

Perfect advice, controller is running fine !

Controller Status
Controller Name:WLC_ILEX
MAC Address:xxxxxxxxxxxxxxxxx
System Time:Jan 08, 202221:33:47
Uptime:2m 33s
Controller Version:5.0.29

Thx :)

closed 2022-01-08 by ilexpl01

mbentley, 2022-01-08:
Good deal.  I'll add some detection to the entrypoint so it will output a message as this isn't the first instance of someone running into this.

mbentley, 2022-01-08:
The new images for 5.0.x will now output a message like this when it detects the old classpath:

```
INFO: Time zone set to 'Etc/UTC'

############################
WARNING: CMD from 4.x detected!  It is likely that this container will fail to start properly with a "Could not find or load main class com.tplink.omada.start.OmadaLinuxMain" error!
  See the note on old CMDs at https://github.com/mbentley/docker-omada-controller#synology-users-and-upgrade-issues for details on why and how to resolve the issue.
############################

INFO: Starting Omada Controller as user omada
Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain
```

commit 556b43bf (2022-01-08): Updated ports and added 5.0 entrypoint to detect old CMDs; fixes #165
commit 7d573d4b (2022-01-08): Merge pull request #166 from mbentley/add-warning-for-cmd

Updated ports and added 5.0 entrypoint to detect old CMDs; fixes #165