mbentley/docker-omada-controller #169: Unable to pull data for device updates

shivamg-42, 2022-01-17:
When I try pulling data for device updates I keep getting the following error:

`01-17-2022 04:36:25.242 INFO [Thread-14] [] c.t.e.c.c.q(753): java.net.UnknownHostException: n-devs-smb.tplinkcloud.com: Temporary failure in name resolution
java.net.UnknownHostException: n-devs-smb.tplinkcloud.com: Temporary failure in name resolution`

mbentley, 2022-01-17:
That just looks like a DNS lookup failure.

Check hostname resolution from the host using your default nameserver:

```
dig +short +identify n-devs-smb.tplinkcloud.com
```

Check hostname resolution from the host using your Google DNS:

```
dig +short +identify @8.8.8.8 n-devs-smb.tplinkcloud.com
```

Check hostname resolution from the container:

```
docker exec -it <omada-controller-container-name-or-id> bash -c 'apt-get update && apt-get install -y dnsutils && dig +short +identify n-devs-smb.tplinkcloud.com'

shivamg-42, 2022-01-17:
When running the first command I get the following: 
```
n-use1-devs-smb.tplinkcloud.com. from server 127.0.0.1 in 29 ms.
54.209.246.165 from server 127.0.0.1 in 29 ms.
54.166.59.202 from server 127.0.0.1 in 29 ms.
```

The second command is:
```
n-use1-devs-smb.tplinkcloud.com. from server 8.8.8.8 in 19 ms.
54.209.246.165 from server 8.8.8.8 in 19 ms.
54.166.59.202 from server 8.8.8.8 in 19 ms.
```
When I run the third command I get errors:
```
Err:1 http://ports.ubuntu.com/ubuntu-ports xenial InRelease
  Temporary failure resolving 'ports.ubuntu.com'
Err:2 http://ports.ubuntu.com/ubuntu-ports xenial-updates InRelease
  Temporary failure resolving 'ports.ubuntu.com'
Err:3 http://ports.ubuntu.com/ubuntu-ports xenial-backports InRelease
  Temporary failure resolving 'ports.ubuntu.com'
Err:4 http://ports.ubuntu.com/ubuntu-ports xenial-security InRelease
  Temporary failure resolving 'ports.ubuntu.com'
Reading package lists... Done
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/xenial/InRelease  Temporary failure resolving 'ports.ubuntu.com'
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/xenial-updates/InRelease  Temporary failure resolving 'ports.ubuntu.com'
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/xenial-backports/InRelease  Temporary failure resolving 'ports.ubuntu.com'
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/xenial-security/InRelease  Temporary failure resolving 'ports.ubuntu.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package dnsutils
```
I changed the nameserver in resolv.conf from 127.0.0.1 to my Raspberry Pi IP, and am getting this when I rerun the first command: 
```
n-use1-devs-smb.tplinkcloud.com. from server 192.168.1.132 in 39 ms.
54.166.59.202 from server 192.168.1.132 in 39 ms.
54.209.246.165 from server 192.168.1.132 in 39 ms.

```

mbentley, 2022-01-17:
OK, so all name resolution is failing inside of the container.  Might check to see what the resolv.conf shows in the container.

```
docker exec -it <omada-controller-container-name-or-id> cat /etc/resolv.conf
```

Then try re-creating the container and re-running the above to get `resolv.conf` and then seeing if you can get the 3rd command to work again.

If I remember correctly, when using something like `127.0.0.1` which wouldn't work inside a container, Docker defaults to using Google DNS, or at least it used to back when https://github.com/moby/moby/issues/6388 was talked over but for some reason, DNS isn't resolving so getting the current state in the container and then in a re-created container after you changed the host's settings should give a better picture of the current state.

shivamg-42, 2022-01-17:
Resolv.conf inside the container is showing `192.168.1.132` (IP for Pi I am running the container on). I recreated the container via portainer and pulled the latest image. The interesting thing is that after I recreated my ER605 starting showing 75.75.75.75 for DNS. Before it was showing 0.0.0.0. However when I try running the third command I still get the same errors. After recreation the resolv.conf still shows `192.168.1.132`

mbentley, 2022-01-17:
Is your DNS resolver on the pi actually listening on 192.168.1.132?  Try this from the host:

```
dig +short +identify @192.168.1.132 n-devs-smb.tplinkcloud.com
```

This will test to see if it is accepting connections on it's IP or not.

shivamg-42, 2022-01-17:
I believe so. When I run that I get the following: 
n-use1-devs-smb.tplinkcloud.com. from server 192.168.1.132 in 39 ms.
54.166.59.202 from server 192.168.1.132 in 39 ms.
54.209.246.165 from server 192.168.1.132 in 39 ms.

shivamg-42, 2022-01-19:
After doing some more research this may be an issue with my Pi and the Docker0 network issue. I found this thread with similar issues: 
https://serverfault.com/questions/642981/docker-containers-cant-resolve-dns-on-ubuntu-14-04-desktop-host

mlarivie, 2022-02-20:
I ran the following:

docker exec -it omada-wlc '/bin/bash'

cat /etc/resolv.conf
Looked good

dig +short +identify n-devs-smb.tplinkcloud.com
Looked good

then I did tail -f /opt/tplink/EAPController/server.log and ran the check for update. Looked good.

Then I went into the web interface under Admin > Site and enabled Automatic upgrades. Reran the check for update. Lo-and-behold the rolling upgrades worked. Eureka! 

And now we've all learned how to troubleshoot a microservices container! 👍 🥇

closed 2022-02-20 by mbentley