mbentley/docker-omada-controller #170: Container not running on Raspberry Pi 3b+

deMachin, 2022-01-17:
**Describe the bug**
Container Starts and stays online for a few seconds and exits with code 139 and no logs.

**Additional context**
The pi 3b+ has arm6l architecture, could this be why?

mbentley, 2022-01-17:
139 is usually that a process was killed.  It may be that the pi doesn't have enough memory.

Check the `/var/log/messages` file to see if there are any `oom-kill` messages.  If you're not sure what you're doing, you can do a `grep "oom-kill:" /var/log/messages` which will look for any oom-kill messages.

It may also be that it is armv6l as I only have `amd64`, `arm64`, and `armv7l` images.  You'd normally get a message about not being able to execute the process if it was the wrong architecture though when running the container interactively in the foreground.

mbentley, 2022-01-20:
Feel free to re-open this if you need additional assistance.

closed 2022-01-20 by mbentley