mbentley/docker-omada-controller #171: Omada not start after update from 4.4.6

leeyuentuen, 2022-01-18:
**Describe the bug**

Error log:

**To Reproduce**
Steps to reproduce the behavior:

1. foo
2. bar
3. baz
4. qux

**Expected behavior**

**How you're launching your container**

**Container Logs**

```
==> server.log <==
01-19-2022 00:11:13.204 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(44): Going to init controller!
01-19-2022 00:11:13.220 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(51): Going to start controller!
01-19-2022 00:11:13.232 INFO [main] [] c.t.s.o.c.o.a.b(32): success to load configuration omada.properties
01-19-2022 00:11:13.262 INFO [main] [] c.t.s.o.s.OmadaBootstrap(79): going to start local mongod.

==> mongod.log <==
2022-01-19T00:11:13.312+0100 I CONTROL  [main] ***** SERVER RESTARTED *****
2022-01-19T00:11:13.318+0100 I CONTROL  [initandlisten] MongoDB starting : pid=16768 port=27217 dbpath=../data/db 64-bit host=omada-new
2022-01-19T00:11:13.318+0100 I CONTROL  [initandlisten] db version v3.6.23
2022-01-19T00:11:13.318+0100 I CONTROL  [initandlisten] git version: d352e6a4764659e0d0350ce77279de3c1f243e5c
2022-01-19T00:11:13.318+0100 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1  11 Sep 2018
2022-01-19T00:11:13.318+0100 I CONTROL  [initandlisten] allocator: tcmalloc
2022-01-19T00:11:13.318+0100 I CONTROL  [initandlisten] modules: none
2022-01-19T00:11:13.319+0100 I CONTROL  [initandlisten] build environment:
2022-01-19T00:11:13.319+0100 I CONTROL  [initandlisten]     distmod: ubuntu1804
2022-01-19T00:11:13.319+0100 I CONTROL  [initandlisten]     distarch: x86_64
2022-01-19T00:11:13.319+0100 I CONTROL  [initandlisten]     target_arch: x86_64
2022-01-19T00:11:13.319+0100 I CONTROL  [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2022-01-19T00:11:13.319+0100 I -        [initandlisten] Detected data files in ../data/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
2022-01-19T00:11:13.319+0100 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=256M,cache_overflow=(file_max=0M),session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
2022-01-19T00:11:14.416+0100 I STORAGE  [initandlisten] WiredTiger message [1642547474:416560][16768:0x7f390a4e73c0], txn-recover: Main recovery loop: starting at 23/4864
2022-01-19T00:11:14.640+0100 I STORAGE  [initandlisten] WiredTiger message [1642547474:640131][16768:0x7f390a4e73c0], txn-recover: Recovering log 23 through 24
2022-01-19T00:11:14.763+0100 I STORAGE  [initandlisten] WiredTiger message [1642547474:763833][16768:0x7f390a4e73c0], txn-recover: Recovering log 24 through 24
2022-01-19T00:11:14.873+0100 I STORAGE  [initandlisten] WiredTiger message [1642547474:873982][16768:0x7f390a4e73c0], txn-recover: Set global recovery timestamp: 0
2022-01-19T00:11:14.923+0100 I CONTROL  [initandlisten] 
2022-01-19T00:11:14.923+0100 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
2022-01-19T00:11:14.923+0100 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
2022-01-19T00:11:14.923+0100 I CONTROL  [initandlisten] 
2022-01-19T00:11:14.923+0100 I CONTROL  [initandlisten] 
2022-01-19T00:11:14.923+0100 I CONTROL  [initandlisten] ** WARNING: soft rlimits too low. rlimits set to 127796 processes, 524288 files. Number of processes should be at least 262144 : 0.5 times number of files.
2022-01-19T00:11:14.950+0100 I FTDC     [initandlisten] Initializing full-time diagnostic data capture with directory '../data/db/diagnostic.data'
2022-01-19T00:11:14.951+0100 I NETWORK  [initandlisten] listening via socket bound to 127.0.0.1
2022-01-19T00:11:14.952+0100 I NETWORK  [initandlisten] listening via socket bound to /tmp/mongodb-27217.sock
2022-01-19T00:11:14.952+0100 I NETWORK  [initandlisten] waiting for connections on port 27217

==> server.log <==
01-19-2022 00:11:15.289 INFO [main] [] c.t.s.o.s.s.b(161): mongodb process id is 16768
01-19-2022 00:11:15.290 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Mongo DB server started
01-19-2022 00:11:15.291 INFO [main] [] c.t.s.o.s.s.b(142): Mongo DB server started

==> mongod.log <==
2022-01-19T00:11:15.436+0100 I NETWORK  [listener] connection accepted from 127.0.0.1:45030 #1 (1 connection now open)
2022-01-19T00:11:15.447+0100 I NETWORK  [listener] connection accepted from 127.0.0.1:45032 #2 (2 connections now open)
2022-01-19T00:11:15.452+0100 I NETWORK  [conn1] received client metadata from 127.0.0.1:45030 conn1: { driver: { name: "mongo-java-driver|sync", version: "4.1.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.4.140-1-pve" }, platform: "Java/Private Build/1.8.0_312-8u312-b07-0ubuntu1~18.04-b07" }
2022-01-19T00:11:15.457+0100 I NETWORK  [conn2] received client metadata from 127.0.0.1:45032 conn2: { driver: { name: "mongo-java-driver|sync", version: "4.1.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.4.140-1-pve" }, platform: "Java/Private Build/1.8.0_312-8u312-b07-0ubuntu1~18.04-b07" }
2022-01-19T00:11:15.489+0100 I NETWORK  [listener] connection accepted from 127.0.0.1:45034 #3 (3 connections now open)
2022-01-19T00:11:15.490+0100 I NETWORK  [conn3] received client metadata from 127.0.0.1:45034 conn3: { driver: { name: "mongo-java-driver|sync", version: "4.1.1" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.4.140-1-pve" }, platform: "Java/Private Build/1.8.0_312-8u312-b07-0ubuntu1~18.04-b07" }
2022-01-19T00:11:15.528+0100 I NETWORK  [conn3] end connection 127.0.0.1:45034 (2 connections now open)
2022-01-19T00:11:15.529+0100 I NETWORK  [conn1] end connection 127.0.0.1:45030 (1 connection now open)
2022-01-19T00:11:15.529+0100 I NETWORK  [conn2] end connection 127.0.0.1:45032 (0 connections now open)
[... 33 lines cut ...]
01-19-2022 00:11:36.285 ERROR [main] [] c.t.s.o.s.d.a(90): Controller install failed. Failed to start.
01-19-2022 00:11:36.286 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Data abnormal due to install failed. Failed to start.
01-19-2022 00:11:36.286 ERROR [main] [] c.t.s.o.s.t.FailExitTask(19): Failed to start omada controller, going to exit
01-19-2022 00:11:36.293 INFO [Thread-1] [] c.t.s.o.s.s.b(213): Going to stop mongod which pid is 16768

==> mongod.log <==
2022-01-19T00:11:36.296+0100 I CONTROL  [signalProcessingThread] got signal 15 (Terminated), will terminate after current cmd ends
2022-01-19T00:11:36.296+0100 I NETWORK  [signalProcessingThread] shutdown: going to close listening sockets...
2022-01-19T00:11:36.296+0100 I NETWORK  [signalProcessingThread] removing socket file: /tmp/mongodb-27217.sock
2022-01-19T00:11:36.297+0100 I FTDC     [signalProcessingThread] Shutting down full-time diagnostic data capture
2022-01-19T00:11:36.299+0100 I STORAGE  [signalProcessingThread] WiredTigerKVEngine shutting down
2022-01-19T00:11:36.465+0100 I NETWORK  [conn4] end connection 127.0.0.1:45126 (2 connections now open)
2022-01-19T00:11:36.465+0100 I NETWORK  [conn12] end connection 127.0.0.1:45176 (1 connection now open)
2022-01-19T00:11:36.465+0100 I NETWORK  [conn5] end connection 127.0.0.1:45128 (0 connections now open)
2022-01-19T00:11:36.510+0100 I STORAGE  [signalProcessingThread] shutdown: removing fs lock...
2022-01-19T00:11:36.510+0100 I CONTROL  [signalProcessingThread] now exiting
2022-01-19T00:11:36.510+0100 I CONTROL  [signalProcessingThread] shutting down with code:0

==> server.log <==
01-19-2022 00:11:43.370 INFO [Thread-1] [] c.t.s.o.s.OmadaBootstrap(176): success to shutdown mongodb database
01-19-2022 00:11:43.370 INFO [Thread-1] [] c.t.s.o.s.OmadaBootstrap(181): Omada Controller exited
01-19-2022 00:11:43.370 INFO [Thread-1] [] c.t.s.o.s.OmadaLinuxMain(80): ShutdownHook: service stopped.
```

**Additional context**

mbentley, 2022-01-19:
How are you launching the container?  You mention dpkg so I am confused as to what you're doing.

leeyuentuen, 2022-01-19:
> How are you launching the container? You mention dpkg so I am confused as to what you're doing.

I have use "tpeap start". I have install Linux  ubuntu where there is only install that deb package

mbentley, 2022-01-19:
OK, then in that case, you're not running my docker image.  Please use the [TP-Link Omada Cloud SDN Forums](https://community.tp-link.com/en/business/forum/582) for assistance as this repository is only for help with my Docker image that packages the software.

closed 2022-01-19 by mbentley