mbentley/docker-omada-controller #174: Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain

jaymanp, 2022-01-24:
Hello,

I'm running 4.4.x in docker without issue. Attempted to perform the upgrade to 5.0 by re-creating the container using the new image. Same volumes and network (host) are being used.  As per the CMDs link below, I've done a re-create and the issue persists. 

Error in question....

`############################

WARNING: CMD from 4.x detected!  It is likely that this container will fail to start properly with a "Could not find or load main class com.tplink.omada.start.OmadaLinuxMain" error!

  See the note on old CMDs at https://github.com/mbentley/docker-omada-controller#synology-users-and-upgrade-issues for details on why and how to resolve the issue.

############################

INFO: Starting Omada Controller as user omada

Error: Could not find or load main class com.tplink.omada.start.OmadaLinuxMain

INFO: Time zone set to 'Canada/Eastern'

INFO: Importing cert from /cert/tls.[key|crt]`

mbentley, 2022-01-24:
If the issue persists, then the CMD is still being retained somehow.

jaymanp, 2022-01-24:
> If the issue persists, then the CMD is still being retained somehow from 4.4.x.

Anything to recommend? I've tried creating a new container entirely, mapping to the existing volumes but the issue remains.

mbentley, 2022-01-24:
If you could provide some more detail about what I am assuming is some sort of NAS, how you're actually doing the re-creation with steps, screenshots, anything, it might help.  I only have Docker natively at the command line so I only know what to expect from it's behavior.  Whatever you mean by re-creating the container, it apparently is not fully re-creating it as it is keeping the previous image's CMD in the container spec.

jaymanp, 2022-01-24:
Worked like a charm. Thank you. I simply edited the container in portainer and modified the CMD string.

closed 2022-01-24 by jaymanp

mbentley, 2022-01-24:
Based on your experience, I updated the link in the entrypoint output and added a note about possibly being able to update the command as an option: https://github.com/mbentley/docker-omada-controller#upgrade-issues

jaymanp, 2022-01-24:
That's fantastic. Thank you. I appreciate the effort you've put into this.