mbentley/docker-omada-controller #178: Can't connect to the web UI

luvxinh, 2022-02-07:
**Describe the bug**

Can't access the web UI at <IP address>:8088. After awhile, it gets redirected to :8043, adds "/login/" in the address bar but still failed to connect.

**To Reproduce**
Steps to reproduce the behavior:

1. Using Portainer on a Raspberry Pi4B 4GB ram to spin a container up
2. All port and environment and volumes follow the guidelines.

**Expected behavior**

Web UI should display and I can proceed with managing my EAPs.

**How you're launching your container**
Portainer

**Container Logs**

```
INFO: Time zone set to 'Etc/UTC+7'
INFO: Starting Omada Controller as user omada
02-07-2022 17:20:08.816 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(32): success to load configuration omada.properties
02-07-2022 17:20:08.911 INFO [main] [] c.t.s.o.s.OmadaBootstrap(79): going to start local mongod.
02-07-2022 17:20:12.462 INFO [main] [] c.t.s.o.s.s.b(161): mongodb process id is 47
02-07-2022 17:20:12.467 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Mongo DB server started
02-07-2022 17:20:12.469 INFO [main] [] c.t.s.o.s.s.b(142): Mongo DB server started

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.4.3)

02-07-2022 17:20:15.909 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(55): Starting OmadaLinuxMain v5.0.30 using Java 1.8.0_312 on e1794927e58e with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.0.30.jar started by omada in /opt/tplink/EAPController/lib)
02-07-2022 17:20:15.924 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(660): No active profile set, falling back to default profiles: default
02-07-2022 17:20:47.991 WARN [main] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
02-07-2022 17:20:57.065 WARN [main] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
02-07-2022 17:21:13.025 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(61): log mq consume task is start...
02-07-2022 17:21:13.042 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(61): Started OmadaLinuxMain in 59.167 seconds (JVM running for 67.337)
02-07-2022 17:21:13.907 INFO [main] [] c.t.s.o.s.d.a(78): Db unchanged. No need to compatible or init data.
02-07-2022 17:21:14.370 INFO [main] [] c.t.s.o.s.t.OmadacInitTask(149): succeed get default omadac OmadacVO(id=0047218369393a1640603664bf046d5a, name=Omada Controller_046D5A)
02-07-2022 17:21:18.573 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(124): upgrade workGroup core thread num is 8, max thread num is 8
02-07-2022 17:21:18.669 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(151): upgradeSendReq workGroup core thread num is 8, max thread num is 8
02-07-2022 17:21:19.174 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(97): manager workGroup core thread  num is 16, max thread num is 16
02-07-2022 17:21:24.121 INFO [main] [] c.t.s.e.s.c.c(152): start schedule remove expire device... period = 10
02-07-2022 17:21:24.140 INFO [main] [] c.t.s.e.s.c.c(145): update rateLimiterCache, permitsPerSecond = 5.0
02-07-2022 17:21:26.703 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(70): adopt workGroup core thread num is 8, max thread num is8
02-07-2022 17:21:27.925 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(43): discovery workGroup core thread num is 2, max thread num is 10
02-07-2022 17:21:33.803 INFO [main] [] c.t.s.o.p.b.a.c(72): DisconnectRequestServer start
02-07-2022 17:21:35.043 INFO [main] [] c.t.s.o.d.g.a(124): discovery Port is 29810.
02-07-2022 17:21:35.044 INFO [main] [] c.t.s.o.d.g.a(125): manage    port v1 is 29811.
02-07-2022 17:21:35.045 INFO [main] [] c.t.s.o.d.g.a(126): manage    port v2 is 29814.
02-07-2022 17:21:35.046 INFO [main] [] c.t.s.o.d.g.a(127): adopt     port v1 is 29812.
02-07-2022 17:21:35.048 INFO [main] [] c.t.s.o.d.g.a(128): upgrade   port is 29813.
02-07-2022 17:21:35.345 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyUdpServer bind /0.0.0.0:29810, export localhost/127.0.0.1:29810
02-07-2022 17:21:35.388 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29812 global traffic shaping, writeLimit: 477364224, readLimit: 0, checkInterval: 1000, max wait time: 15000
02-07-2022 17:21:35.425 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29812, export localhost/127.0.0.1:29812
02-07-2022 17:21:35.429 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29811 global traffic shaping, writeLimit: 477364224, readLimit: 0, checkInterval: 1000, max wait time: 15000
02-07-2022 17:21:35.433 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
02-07-2022 17:21:35.438 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 477364224, readLimit: 0, checkInterval: 1000, max wait time: 15000
02-07-2022 17:21:35.444 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
02-07-2022 17:21:35.450 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 477364224, readLimit: 0, checkInterval: 1000, max wait time: 15000
02-07-2022 17:21:35.455 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
02-07-2022 17:21:35.456 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Device connector server started.
02-07-2022 17:21:35.458 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Omada Controller started
02-07-2022 17:21:35.459 INFO [main] [] c.t.s.o.s.t.SuccessExitTask(23): Omada Controller started
02-07-2022 17:21:40.206 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.a.a(332): MANAGE_BY_OTHERS Device 00-31-92-E7-BA-14 on omadac 0047218369393a1640603664bf046d5a is discoveried.
02-07-2022 17:21:43.047 WARN [https-jsse-nio-8043-exec-10] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
02-07-2022 17:21:43.120 WARN [https-jsse-nio-8043-exec-10] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
02-07-2022 17:21:43.995 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.a.a(332): MANAGE_BY_OTHERS Device D8-07-B6-14-A4-10 on omadac 0047218369393a1640603664bf046d5a is discoveried.
02-07-2022 17:26:45.750 INFO [vpn-certificate-clean-workgroup-0] [] c.t.s.o.m.c.d.m.s.g.a.a(98): VpnCertificateClearTask handling omadacId: OmadacId(0047218369393a1640603664bf046d5a)```

**Additional context**
None

mbentley, 2022-02-07:
I know you have said "All port and environment and volumes follow the guidelines." but are you re-mapping the ports from the defaults or are they directly mapped?  You say you're launching the container via Portainer but can you provide any details about how the ports are being mapped and any environment variables that are being set?  It initially sounds like the [issues described in port mapping](https://github.com/mbentley/docker-omada-controller#using-port-mapping) but it's hard to say without knowing exactly how the container is launched.

mbentley, 2022-02-07:
Thanks for the screen shots - that helps clarify that you do indeed mean exactly what you said about using the defaults - I just wanted to make sure.

I *think* what you're seeing is that it appears you might be running another controller somewhere on your network - can you confirm if that is the case?  If I remember right, these are the logs that indicate that sort of issue:

```
...
02-07-2022 17:21:40.206 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.a.a(332): MANAGE_BY_OTHERS Device 00-31-92-E7-BA-14 on omadac 0047218369393a1640603664bf046d5a is discoveried.
...
02-07-2022 17:21:43.995 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.a.a(332): MANAGE_BY_OTHERS Device D8-07-B6-14-A4-10 on omadac 0047218369393a1640603664bf046d5a is discoveried.
...
```

If I recall, what happens is that if the Omada Controller detects another controller running on the network, it will essentially shut itself down to prevent conflicts.  I'll have to try this myself to see if that is indeed the case.

mbentley, 2022-02-07:
Hmm, well I just tried to run another controller on my network and didn't see similar errors - it ran fine for the few minutes I was testing.

Could you run a few commands from a command line to see what exactly might be going on?  If connections are successful (which might happen on the commands where it follows redirects), there might be a great wall of text as it loads the raw HTML so you might append `> /dev/null` to the command to just pipe the standard out to null

Verbose `curl` to http port:

```
curl -kv http://<ip-or-hostname>:8088/
```

Verbose `curl` to http port and follow redirects:

```
curl -kvL http://<ip-or-hostname>:8088/
```

Verbose `curl` to https port:

```
curl -kv https://<ip-or-hostname>:8043/
```

Verbose `curl` to https port and follow redirects:

```
curl -kvL https://<ip-or-hostname>:8043/
```

luvxinh, 2022-02-08:
Hi

I'm sorry for the late reply. I'm at UTC+7 so there's a time gap between us.

I'm still learning about Linux and everything, so I'm sorry I don't understand about appending > /dev/null. But since it's not too much text, here you go.

[image]
 
As for the device already managed, I used to have that error too, but I can just reset the device and it would be fine. This time, I don't think it's that problem yet, since I'm stuck at connecting to the web UI.

mbentley, 2022-02-08:
Well, everything looks like it is working as expected.  When hitting 8088 on http, it redirects to the login page on http, then redirects again to the login page on https which is what I would expect.

What is happening in your browser when you access https://192.168.1.27:8043/0047218369393a1640603664bf046d5a/login ?  You should be getting a SSL warning because it is a self signed certificate by default and then if you accept the self-signed cert, you should be presented with the login page (or the wizard if you haven't run it before).

mbentley, 2022-02-08:
I suppose it is possible that it is something Microsoft Edge related.

To do another really basic quick test, try to just run an nginx container on your pi:

```
docker run -d --name nginx-test -p 8081:80 nginx:alpine
```

Then try to access it at http://192.168.1.27:8081/

If for some reason you already have something running on port 8081, switch it to another unused port in the run command.  If it is successful, you should see something like this:

[image]

When you are done, you can stop and clean up the container:

```
docker stop nginx-test
docker rm nginx-test
docker rmi nginx:alpine
```

That will at least make sure that you can access some sort of web application from your machine.

luvxinh, 2022-02-08:
I haven't run the nginx yet, but for web application, I installed Home assistant and used without problem. Also installed a separate instance of Adguard via Portainer and it runs just fine, including the web GUI and everything.
Does this help? I can get nginx via Portainer or separately in Docker if you think it's important.

luvxinh, 2022-02-08:
My bad. It's definitely Edge. I just got Firefox installed and although it refuses the connection since it's not secure, when I push through, it's there:

[image]

But is it what it's supposed to do? I though the certificate is self-signed and should be valid?

mbentley, 2022-02-08:
A self-signed certificate will not be valid unless you import and trust the root CA certificate that is used in the generation process.  I believe the certificate that they use as the default is not unique at all; I imagine it is this one:

```
-----BEGIN CERTIFICATE-----
MIIDETCCAfmgAwIBAgIEHUfJYjANBgkqhkiG9w0BAQsFADA4MRAwDgYDVQQKEwdU
UC1MaW5rMRAwDgYDVQQLEwdUUC1MaW5rMRIwEAYDVQQDEwlsb2NhbGhvc3QwIBcN
MjAwOTExMDgxNTIxWhgPMjA1MDA5MTEwODE1MjFaMDgxEDAOBgNVBAoTB1RQLUxp
bmsxEDAOBgNVBAsTB1RQLUxpbmsxEjAQBgNVBAMTCWxvY2FsaG9zdDCCASIwDQYJ
KoZIhvcNAQEBBQADggEPADCCAQoCggEBAJi5IlNQVJAVv3KNf3Um7QU9+RS1aUAd
ql2trNzqXt3sZSbjpeNrK9bA6TRJY+fLwhg/au/1NL+F3AXiq3LhVWQiY8PqK8yA
6O9QDNpQMpHhqSy9lHeE5OLs9TLPQApzqTQA9FsXFlx41IWhcohKw8u04ayvAMHZ
nNfV5MKXx77FBxjvrNQQIwvrJLeCgD1G21sCYTMA6mHeC/JGPAlscB7LVkDt70WY
D/dJbntuA7n5aozWR/w9Z7wUn5NEfXZs+yQQEXo11s4DC7RaIS/yU4zsTvNo78eF
c8iM/Yrm1609pcKNtzyy1ficEORnpgmEEWWq6nFFZkDSjWguc3bkvW8CAwEAAaMh
MB8wHQYDVR0OBBYEFOwrLt0YJTBLX9TF0opDNeaVoFL3MA0GCSqGSIb3DQEBCwUA
A4IBAQCOMCseIqosloaHRt98ESH/2SSkycLOfGIBQaXpbvI1wXU8E5MsU+NvdHKY
6KYrHQJEyAF/clFmyS7cj2K78KWDBWh2Zi22kTODjd18apJzTInIR8og0pX7zSFq
xUlbP9E4Moaifj30QX+bPfAmHJBsEOo16YQOZgyXo9ADe6UdFqxO+JcOH3RPbymI
7u4veCwUOuZ5JF1CYYKL8AKl1h+GEaoh3aAlHHEBhxA4TOCpcFzRgtnQ30n/qkDJ
3NUBsUrKpAIP9wz8KICYNaC+PGA87i5mChE8tUCirZKXKwzClyX9HmEGcbBexD08
sWHAEWTBZhj1T1CA8jDXuOs7aQqh
-----END CERTIFICATE-----
```

You _could_ [add it to the certificate to the trust store](https://docs.microsoft.com/en-us/biztalk/adapters-and-accelerators/accelerator-swift/adding-certificates-to-the-certificates-store-on-the-client) but with it not being unique for your environment, it's a bit of a false sense of security.  You're probably better off just bypassing the cert warning if you do not have the ability to generate a valid certificate.  I'm not sure why Edge isn't allowing you to even get a certificate warning but I also currently do not have access to the environment with Edge to use to test as well.

luvxinh, 2022-02-09:
Hi

Thanks for the reply and your explanation. I used to be able to bypass the security block in Edge. I guess with the new Edge, it's gone forever. 

For the time being, I'll resort to Firefox. I am however studying networking and security so definitely will improve in the future.

Thank you for your time.

closed 2022-02-09 by luvxinh