mbentley/docker-omada-controller #180: Running docker compose by stack in portainer but can't connect to the controller

luvxinh, 2022-02-12:
**Describe the bug**
The container is up and running in Portainer but can't connect to the controller

**To Reproduce**

Browser: Firefox latest as at the date of writing this.
openmediavault v6. Storage is created on an external hdd. File system ext4.
Portainer latest installed by openmediavault.
Docker latest installed by openmediavault.
Docker swarm including 2 nodes of Raspberry pi 4.

I use the docker compose file to deploy a stack on Portainer. It spins up and the controller started, but I can't connect to it either on port 8088 or 8043.

**Expected behavior**
Should be connect-able.

**How you're launching your container**
copy and paste the below in the stack deployment of Portainer.

```
version: "3.1"

services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:latest
     environment:
      - TZ=Etc/UTC
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
     network_mode: host

     volumes:
      - /srv/dev-disk-by-uuid-6dafc5d3-83f5-5c47-acbc-d30349bf2aee/omada/data:/opt/tplink/EAPController/data
      - /srv/dev-disk-by-uuid-6dafc5d3-83f5-5c47-acbc-d30349bf2aee/omada/work:/opt/tplink/EAPController/work
      - /srv/dev-disk-by-uuid-6dafc5d3-83f5-5c47-acbc-d30349bf2aee/omada/logs:/opt/tplink/EAPController/logs
      
restart: unless-stopped
#volumes:
#  omada-data:
#  omada-work:
#  omada-logs:

```
**Container Logs**
```
INFO: Time zone set to 'Etc/UTC'
INFO: Setting 'portal.https.port' to 8043 in omada.properties
WARNING: owner or group (0:100) not set correctly on '/opt/tplink/EAPController/data'
INFO: setting correct permissions
WARNING: owner or group (0:100) not set correctly on '/opt/tplink/EAPController/work'
INFO: setting correct permissions
WARNING: owner or group (0:100) not set correctly on '/opt/tplink/EAPController/logs'
INFO: setting correct permissions
INFO: Database directory missing; creating '/opt/tplink/EAPController/data/db'
done
INFO: Starting Omada Controller as user omada
tail: cannot open '/opt/tplink/EAPController/logs/server.log' for reading: No such file or directory
tail: '/opt/tplink/EAPController/logs/server.log' has appeared;  following new file
02-12-2022 12:33:49.645 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(32): success to load configuration omada.properties
02-12-2022 12:33:49.777 INFO [main] [] c.t.s.o.s.OmadaBootstrap(79): going to start local mongod.
02-12-2022 12:33:53.811 INFO [main] [] c.t.s.o.s.s.b(161): mongodb process id is 51
02-12-2022 12:33:53.816 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Mongo DB server started
02-12-2022 12:33:53.819 INFO [main] [] c.t.s.o.s.s.b(142): Mongo DB server started
[... 38 lines cut ...]
02-12-2022 12:36:39.642 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29812, export localhost/127.0.0.1:29812
02-12-2022 12:36:39.645 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29811 global traffic shaping, writeLimit: 477364224, readLimit: 0, checkInterval: 1000, max wait time: 15000
02-12-2022 12:36:39.650 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
02-12-2022 12:36:39.655 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 477364224, readLimit: 0, checkInterval: 1000, max wait time: 15000
02-12-2022 12:36:39.658 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
02-12-2022 12:36:39.676 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(119): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 477364224, readLimit: 0, checkInterval: 1000, max wait time: 15000
02-12-2022 12:36:39.680 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(53): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
02-12-2022 12:36:39.681 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Device connector server started.
02-12-2022 12:36:39.682 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Omada Controller started
02-12-2022 12:36:39.683 INFO [main] [] c.t.s.o.s.t.SuccessExitTask(23): Omada Controller started
02-12-2022 12:46:27.057 INFO [vpn-certificate-clean-workgroup-0] [] c.t.s.o.m.c.d.m.s.g.a.a(98): VpnCertificateClearTask handling omadacId: OmadacId(508486138e337dafa48ddb5a358e2647)
```

**Additional context**
1. It seems like it cannot connect to such ports.

[image]

2. Although the network-mode in the stack file is "host", Portainer created this overlay network. I'm not sure if it's related.

[image]

3. Container up and running.

[image]

mbentley, 2022-02-12:
So last time I checked, you [can't use network_mode: "host" with Swarm mode](https://docs.docker.com/compose/compose-file/compose-file-v3/#network_mode).  It's been a minute since I've deployed anything to Swarm so I forget some of the requirements for mapping ports but you will probably need to statically set ports to expose if you really want to use Swarm mode + Compose.  Make sure that your controller doesn't get scheduled on a node where the persistent storage isn't available as well otherwise it'll start up like a brand new controller as nothing is pinning the container to a node.

luvxinh, 2022-02-13:
I can't believe I missed that in the Docker documentation.

Thanks for your kind patience. I guess for the time being I'll just stick with the deploy container then.

closed 2022-02-13 by luvxinh