mbentley/docker-omada-controller #183: Mongod is shutting down every 24h

mlarivie, 2022-02-26:
Around 0800 EST every morning the controller goes down. I haven't been able to track this one down. Not sure if this is related to enabling auto upgrades on the controller (for the AP's) which I do recall disabling in the past:

mongod.log:
`2022-02-25T05:28:37.861-0500 I CONTROL  [signalProcessingThread] got signal 15 (Terminated), will terminate after current cmd ends`

server.log:
```
[root@z420 /]# zcat /opt/omada-wlc/logs/server_2022-02-25_1.log.gz |less

RROR, which don't need to handle.
02-25-2022 00:05:40.221 INFO [manage-work-group-3] [] c.t.s.o.m.d.p.t.c(129): Device 60-A4-B7-F3-5B-90 OmadacId(7ad6eb8f9083298a5ddad8a7db28f2ec) changed to status CONNECTED_E
RROR, which don't need to handle.
02-25-2022 00:05:40.224 INFO [SpringContextShutdownHook] [] c.t.s.e.s.d(92): going to shutdown ecsp thread pools.
02-25-2022 00:05:40.333 INFO [SpringContextShutdownHook] [] c.t.s.e.p.b.a(226): going to shutdown proxy executors
02-25-2022 00:05:40.344 INFO [SpringContextShutdownHook] [] c.t.s.e.p.a(73): going to shutdown proxy thread pools.
02-25-2022 00:05:47.142 INFO [Thread-1] [] c.t.s.o.s.OmadaBootstrap(176): success to shutdown mongodb database
02-25-2022 00:05:47.142 INFO [Thread-1] [] c.t.s.o.s.OmadaBootstrap(181): Omada Controller exited
02-25-2022 00:05:47.143 INFO [Thread-1] [] c.t.s.o.s.OmadaLinuxMain(80): ShutdownHook: service stopped.

02-25-2022 00:05:49.013 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(32): success to load configuration omada.properties
02-25-2022 00:05:49.041 INFO [main] [] c.t.s.o.s.OmadaBootstrap(79): going to start local mongod.
02-25-2022 00:05:51.052 INFO [main] [] c.t.s.o.s.s.b(161): mongodb process id is 60
02-25-2022 00:05:51.054 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Mongo DB server started
02-25-2022 00:05:51.055 INFO [main] [] c.t.s.o.s.s.b(142): Mongo DB server started
02-25-2022 00:05:52.070 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(55): Starting OmadaLinuxMain v5.0.30 using Java 1.8.0_312 on z420.lariv.lan with PID 1 (/opt/tplink/EAPControll
er/lib/local-starter-5.0.30.jar started by omada in /opt/tplink/EAPController/lib)
02-25-2022 00:05:52.073 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(660): No active profile set, falling back to default profiles: default
02-25-2022 00:05:59.735 WARN [main] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading
 converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
02-25-2022 00:06:02.418 WARN [main] [] o.s.d.c.CustomConversions(260): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading
 converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
02-25-2022 00:06:05.978 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(61): log mq consume task is start...
02-25-2022 00:06:05.982 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(61): Started OmadaLinuxMain in 14.53 seconds (JVM running for 17.915)
```

mbentley, 2022-02-26:
What you're probably seeing lines up with me triggering rebuilds every day plus something on your system that is auto-updating the deployed containers.  Every day at 3am Eastern, I kick off jobs that will rebuild all of the images I maintain.  This doesn't actually rebuild the images every single day though as it uses caching of images so it will only update if:
1. The base image is updated, invalidating the reset of the cache
2. A new version of the software is available and the changes have been merged into the Dockerfile
3. A merged change modifies a file like an entrypoint that would invalidate the cache at some point
4. There are vulnerabilities that are detected in the image and my automation re-triggers the build

I am guessing you have something that is watching for updates and re-creating the container when it detects them or it is just checking every day at 8am (or multiple times a day but the trigger only happens at 8am because that is the last time it is updated between then and the last check).  Is it actually happening every single day or just frequently within the past few days?  If you tell me which image + tag you're using, I can probably tell you when an actual update has occurred in the past couple of weeks.  I've seen some reports of other tools that will look at the last updated time of the image manifests, which are re-written every day but the digests of the images do not change unless there are actual changes to image layers and they will end up re-creating the containers daily because of the manifest re-write.

On my end, there isn't a quick and easy way for me to detect if there are image changes upstream that scales very well with the 150 unique images I rebuild/retest a day to detect an image change and only re-write the manifest if needed but if I know what it is that is monitoring for image changes, I can look at why it is doing it and see if there is a way to fix it.

mbentley, 2022-02-26:
I actually just noticed that the tags listed in the readme weren't correct for the amd64 explicit architectures so I just fixed that but anyway...

One thing you can probably do that would prevent this from happening would be to use the specific architecture image which doesn't have a multi-arch manifest.  So if you're using `mbentley/omada-controller:5.0`, that is just a multi-arch manifest that is built from three tagged images for `amd64`, `arm64`, and `armv7l` architectures.  For `amd64`, the explicit tag is `mbentley/omada-controller:5.0-amd64` and that should only be modified when changes happen to the actual image layers.

mlarivie, 2022-02-26:
The resolution is updating the systemd unit file.. if someone runs into a similar issue this will take care of it 

# systemctl disable podman-auto-update.timer

https://www.redhat.com/sysadmin/podman-auto-updates-rollbacks