mbentley/docker-omada-controller #186: Healtcheck

jonas740, 2022-03-02:
Noticed you added healthcheck for the latest, not an issue but may i ask what was the point with it ? I am not very experianced but i would like to remove it

Impact123, 2022-03-02:
[For `docker-compose.yml` you can use](https://docs.docker.com/compose/compose-file/compose-file-v3/#healthcheck)
```yaml
healthcheck:
    disable: true
```
[For `docker run` you can use](https://docs.docker.com/engine/reference/run/#healthcheck)
```bash
docker run --no-healthcheck ...
```
[Some infomation about healthcheck](https://docs.docker.com/engine/reference/builder/#healthcheck)

mbentley, 2022-03-02:
I was just typing up the same :)

Just to add, orchestrators (like Docker Swarm) or other tools can check the health check status to determine if the container is healthy and act upon the status, such as re-creating the container.

closed 2022-03-23 by mbentley