mbentley/docker-omada-controller #187: Import Docker Container

gavind87, 2022-03-03:
**Describe the bug**
I made an export of the container in Synology Docker.
When I Import it again, the controller starts from scratch with a the setup Wizard

**To Reproduce**
Steps to reproduce the behavior:

1. Export Docker Container
2. Import Docker Container

Is there anyway around this ? I want to avoid starting from scratch and resetting all my devices to have them adopted on my new controller, and due to a disk crash I can no longer access my old docker container to run it and create a backup/restore

mbentley, 2022-03-03:
The data should persist either in Docker volumes or bind mounts to a path on the host. That depends on how you executed the container.

gavind87, 2022-03-03:
I have set it up such that the bind mounts are as they were before. 

Any suggestions on how to debug this further ? 

In essence I just need my device username and PWs again to avoid resetting each individual device and adopting them all over again.

mbentley, 2022-03-03:
If you have bind mounted the directories to the correct locations, it will just resume as if you stopped the service. Check the log files which will be in the logs bind mount, feel free to post them if you want me to look.  I have never seen it "forget" all of the settings.

mbentley, 2022-03-03:
Just to be clear, if you should not be exporting the container - there isn't a reason to do that.  The only data written inside of the container is to temp files that do not matter.  You should be backing up the persistent data if you wish to do a simple file based backup (making sure to back up while the container is stopped so you don't end up with any database corruption) and re-create the container from the Docker image with the data mounted.

closed 2022-03-03 by mbentley