mbentley/docker-omada-controller #189: Controller won't start due to a MongoDB error

FileGo, 2022-03-16:
**Describe the bug**
Upon updating the container to the latest image on Docker hub, a database error seems to prevent the Omada controller to start.

**To Reproduce**
Steps to reproduce the behavior:

1. Pull latest Docker image (
2. Restart the container.

**Expected behavior**
It should work.

**How you're launching your container**
```
version: "3.7"
services:
  omada:
    container_name: omada
    image: mbentley/omada-controller:latest
    network_mode: host
    environment:
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - TZ=Etc/UTC
      - LOG4J_FORMAT_MSG_NO_LOOKUPS=true
    volumes:
      - /data_local/omada/data:/opt/tplink/EAPController/data
      - /data_local/omada/work:/opt/tplink/EAPController/work
      - /data_local/omada/logs:/opt/tplink/EAPController/logs
    restart: unless-stopped
```

**Container Logs**
```
INFO: Time zone set to 'Etc/UTC'
INFO: Starting Omada Controller as user omada
2022-03-16 08:47:15 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
2022-03-16 08:47:15 [main] [INFO]-[SourceFile:91] - going to start local mongod.
2022-03-16 08:47:17 [main] [INFO]-[SourceFile:167] - mongodb process id is 42
2022-03-16 08:47:17 [main] [INFO]-[SourceFile:148] - Mongo DB server started
2022-03-16 08:47:25 [main] [WARN]-[AbstractApplicationContext.java:557] - Exception encountered during context initialization - cancelling refresh attempt: org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'modelFwRepository': Invocation of init method failed; nested exception is org.springframework.dao.DuplicateKeyException: Write failed with error code 11000 and error message 'E11000 duplicate key error collection: omada.modelfw index: compound_model dup key: { : "EAP225(EU) v3.0" }'; nested exception is com.mongodb.DuplicateKeyException: Write failed with error code 11000 and error message 'E11000 duplicate key error collection: omada.modelfw index: compound_model dup key: { : "EAP225(EU) v3.0" }'
2022-03-16 08:47:25 [main] [ERROR]-[ContextLoader.java:312] - Context initialization failed
org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'modelFwRepository': Invocation of init method failed; nested exception is org.springframework.dao.DuplicateKeyException: Write failed with error code 11000 and error message 'E11000 duplicate key error collection: omada.modelfw index: compound_model dup key: { : "EAP225(EU) v3.0" }'; nested exception is com.mongodb.DuplicateKeyException: Write failed with error code 11000 and error message 'E11000 duplicate key error collection: omada.modelfw index: compound_model dup key: { : "EAP225(EU) v3.0" }'
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1778) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:593) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:515) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.AbstractBeanFactory.lambda$doGetBean$0(AbstractBeanFactory.java:320) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.DefaultSingletonBeanRegistry.getSingleton(DefaultSingletonBeanRegistry.java:222) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:318) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:199) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.preInstantiateSingletons(DefaultListableBeanFactory.java:830) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.context.support.AbstractApplicationContext.finishBeanFactoryInitialization(AbstractApplicationContext.java:877) ~[spring-context-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:549) ~[spring-context-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.web.context.ContextLoader.configureAndRefreshWebApplicationContext(ContextLoader.java:400) ~[spring-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.web.context.ContextLoader.initWebApplicationContext(ContextLoader.java:291) ~[spring-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.web.context.ContextLoaderListener.contextInitialized(ContextLoaderListener.java:103) ~[spring-web-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.eclipse.jetty.server.handler.ContextHandler.callContextInitialized(ContextHandler.java:957) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.servlet.ServletContextHandler.callContextInitialized(ServletContextHandler.java:553) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.server.handler.ContextHandler.startContext(ContextHandler.java:922) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.servlet.ServletContextHandler.startContext(ServletContextHandler.java:365) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.webapp.WebAppContext.startWebapp(WebAppContext.java:1497) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.webapp.WebAppContext.startContext(WebAppContext.java:1459) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.server.handler.ContextHandler.doStart(ContextHandler.java:852) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.servlet.ServletContextHandler.doStart(ServletContextHandler.java:278) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.webapp.WebAppContext.doStart(WebAppContext.java:545) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
        at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
[... 202 lines cut ...]
        at org.springframework.data.util.Lazy.getNullable(Lazy.java:141) ~[spring-data-commons-2.0.14.RELEASE.jar:2.0.14.RELEASE]
        at org.springframework.data.util.Lazy.get(Lazy.java:63) ~[spring-data-commons-2.0.14.RELEASE.jar:2.0.14.RELEASE]
        at org.springframework.data.repository.core.support.RepositoryFactoryBeanSupport.afterPropertiesSet(RepositoryFactoryBeanSupport.java:293) ~[spring-data-commons-2.0.14.RELEASE.jar:2.0.14.RELEASE]
        at org.springframework.data.mongodb.repository.support.MongoRepositoryFactoryBean.afterPropertiesSet(MongoRepositoryFactoryBean.java:119) ~[spring-data-mongodb-2.0.14.RELEASE.jar:2.0.14.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.invokeInitMethods(AbstractAutowireCapableBeanFactory.java:1837) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1774) ~[spring-beans-5.1.6.RELEASE.jar:5.1.6.RELEASE]
        ... 40 more
Failed to start omada controller, going to exit
2022-03-16 08:47:25 [main] [WARN]-[SourceFile:20] - WebApplicationContext has not been initialed, failed to get bean dataInitService
2022-03-16 08:47:25 [main] [ERROR]-[SourceFile:51] - Failed to get WebApplicationContext, MetaDataInitTask execute failed!
2022-03-16 08:47:25 [main] [ERROR]-[SourceFile:18] - Failed to start omada controller, going to exit
2022-03-16 08:47:25 [Thread-1] [INFO]-[SourceFile:219] - Going to stop mongod which pid is 42
2022-03-16 08:47:25 [device-discovery-server] [INFO]-[SourceFile:59] - Start device discovery server
2022-03-16 08:47:25 [device-manage-server] [INFO]-[SourceFile:116] - Start device manage server
[... cut ...]
2022-03-16 08:47:27 [nioEventLoopGroup-6-1] [INFO]-[SourceFile:50] - Omada Controller isn't prepared to handle event
2022-03-16 08:47:29 [nioEventLoopGroup-6-1] [INFO]-[SourceFile:50] - Omada Controller isn't prepared to handle event
2022-03-16 08:47:29 [nioEventLoopGroup-6-1] [INFO]-[SourceFile:50] - Omada Controller isn't prepared to handle event
ShutdownHook: service stopped.
2022-03-16 08:47:32 [Thread-1] [INFO]-[SourceFile:181] - success to shutdown mongodb database
2022-03-16 08:47:32 [Thread-1] [INFO]-[SourceFile:186] - Omada Controller exited
```

**Additional context**

Babdef, 2022-03-16:
Same problem here. Was able to start it up with an empty database only to find out that the :latest image is version 4.4.8 now, instead of version 5. That explains the database error.

mbentley, 2022-03-16:
How are the both of you shutting down the containers when you do upgrades or re-create the containers?

mbentley, 2022-03-16:
OK, it's fixed now.  Sorry, I split out the jobs for each major.minor version the other day and that exposed my lack of removing the latest tags from the 4.4 builds so if the 4.4 builds completed after my 5.0 build job, the 4.4 images ended up being tagged as latest.

closed 2022-03-16 by mbentley

Babdef, 2022-03-16:
Yes, it's working again. Thank you!

FileGo, 2022-03-16:
Swift fix, all works fine now here as well. Thanks.

IAmKonni, 2022-03-16:
I was affected, too. Fix is working, thank you for fast fix. :)