mbentley/docker-omada-controller #190: Docker container does not start anymore

rucksman, 2022-03-16:
The container used to run fine, but since the last update it does not start anymore. The error message I get:

```
Creating omada-controller ... done
Attaching to omada-controller
omada-controller    | INFO: Time zone set to 'Europe/Berlin'
omada-controller    | INFO: Starting Omada Controller as user omada
omada-controller    | 2022-03-16 20:40:54 [log4j-thread] [INFO]-[SourceFile:29] - success to load configuration omada.properties
omada-controller    | 2022-03-16 20:40:55 [main] [INFO]-[SourceFile:91] - going to start local mongod.
omada-controller    | 2022-03-16 20:41:00 [main] [INFO]-[SourceFile:167] - mongodb process id is 42
omada-controller    | 2022-03-16 20:41:00 [main] [INFO]-[SourceFile:148] - Mongo DB server started
omada-controller    | 2022-03-16 20:41:14 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
omada-controller    | 2022-03-16 20:41:19 [main] [INFO]-[SourceFile:38] - monitor context initialing...
omada-controller    | 2022-03-16 20:41:20 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
omada-controller    | 2022-03-16 20:41:20 [main] [INFO]-[SourceFile:193] - User do not configure max.device, use defaultValue 1500.
omada-controller    | 2022-03-16 20:41:21 [main] [WARN]-[WebAppContext.java:554] - Failed startup of context o.e.j.w.WebAppContext@7d898981{Omada SDN Controller Software,/,file:///opt/tplink/EAPController/work/omada-web/webapp/,UNAVAILABLE}{/omada-web.war}
omada-controller    | java.lang.NullPointerException: null
omada-controller    |   at java.util.concurrent.ConcurrentHashMap.putVal(ConcurrentHashMap.java:1011) ~[?:1.8.0_312]
omada-controller    |   at java.util.concurrent.ConcurrentHashMap.putIfAbsent(ConcurrentHashMap.java:1535) ~[?:1.8.0_312]
omada-controller    |   at com.tplink.omada.domain.lock.d.f(SourceFile:147) ~[omada-domain-4.4.8.jar:?]
omada-controller    |   at com.tplink.omada.service.site.d.a(SourceFile:63) ~[omada-service-4.4.8.jar:?]
omada-controller    |   at java.util.ArrayList.forEach(ArrayList.java:1259) ~[?:1.8.0_312]
omada-controller    |   at com.tplink.omada.service.site.d.b(SourceFile:63) ~[omada-service-4.4.8.jar:?]
omada-controller    |   at com.tplink.omada.service.site.d.a(SourceFile:39) ~[omada-service-4.4.8.jar:?]
omada-controller    |   at com.tplink.omada.service.ServiceInit.a(SourceFile:60) ~[omada-service-4.4.8.jar:?]
omada-controller    |   at com.tplink.omada.web.ApplicationInit.contextInitialized(SourceFile:48) ~[omada-web-4.4.8.jar:?]
omada-controller    |   at org.eclipse.jetty.server.handler.ContextHandler.callContextInitialized(ContextHandler.java:957) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.servlet.ServletContextHandler.callContextInitialized(ServletContextHandler.java:553) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.server.handler.ContextHandler.startContext(ContextHandler.java:922) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.servlet.ServletContextHandler.startContext(ServletContextHandler.java:365) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.webapp.WebAppContext.startWebapp(WebAppContext.java:1497) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.webapp.WebAppContext.startContext(WebAppContext.java:1459) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.server.handler.ContextHandler.doStart(ContextHandler.java:852) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.servlet.ServletContextHandler.doStart(ServletContextHandler.java:278) ~[jetty-servlet-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.webapp.WebAppContext.doStart(WebAppContext.java:545) ~[jetty-webapp-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:117) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.server.handler.ContextHandlerCollection.doStart(ContextHandlerCollection.java:168) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.util.component.ContainerLifeCycle.start(ContainerLifeCycle.java:138) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.server.Server.start(Server.java:415) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.util.component.ContainerLifeCycle.doStart(ContainerLifeCycle.java:108) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.server.handler.AbstractHandler.doStart(AbstractHandler.java:113) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.server.Server.doStart(Server.java:382) ~[jetty-server-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[jetty-util-9.4.15.v20190215.jar:9.4.15.v20190215]
omada-controller    |   at com.tplink.omada.start.b.c.a(SourceFile:76) ~[omada-start.jar:?]
omada-controller    |   at com.tplink.omada.start.b.b.e(SourceFile:75) ~[omada-start.jar:?]
omada-controller    |   at com.tplink.omada.start.task.JettyStartUpTask.a(SourceFile:20) ~[omada-start.jar:?]
omada-controller    |   at com.tplink.omada.start.task.f.a(SourceFile:13) ~[omada-start.jar:?]
omada-controller    |   at com.tplink.omada.start.OmadaBootstrap.f(SourceFile:321) ~[omada-start.jar:?]
omada-controller    |   at com.tplink.omada.start.OmadaLinuxMain.b(SourceFile:87) ~[omada-start.jar:?]
omada-controller    |   at com.tplink.omada.start.OmadaLinuxMain.main(SourceFile:25) ~[omada-start.jar:?]
omada-controller    | 2022-03-16 20:41:21 [device-discovery-server] [INFO]-[SourceFile:59] - Start device discovery server
omada-controller    | 2022-03-16 20:41:21 [device-manage-server] [INFO]-[SourceFile:116] - Start device manage server
omada-controller    | 2022-03-16 20:41:21 [device-adopt-server] [INFO]-[SourceFile:87] - Start device adopt server
omada-controller    | 2022-03-16 20:41:22 [main] [ERROR]-[SourceFile:108] - DefaultSite already exist.
omada-controller    | 2022-03-16 20:41:22 [main] [ERROR]-[SourceFile:293] - Failed to add default site
omada-controller    | 2022-03-16 20:41:22 [main] [ERROR]-[SourceFile:119] - Failed to init default site
omada-controller    | 2022-03-16 20:41:22 [main] [ERROR]-[SourceFile:62] - Failed to initDefautSite
omada-controller    | 2022-03-16 20:41:22 [main] [ERROR]-[SourceFile:18] - Failed to start omada controller, going to exit
omada-controller    | Failed to start omada controller, going to exit
omada-controller    | 2022-03-16 20:41:22 [comm-pool-0] [WARN]-[SourceFile:142] - get InterruptedException while saving log.
omada-controller    | 2022-03-16 20:41:22 [Thread-1] [INFO]-[SourceFile:219] - Going to stop mongod which pid is 42
omada-controller    | 2022-03-16 20:41:27 [nioEventLoopGroup-5-1] [INFO]-[SourceFile:50] - Omada Controller isn't prepared to handle event
omada-controller    | 2022-03-16 20:41:33 [Thread-1] [INFO]-[SourceFile:181] - success to shutdown mongodb database
omada-controller    | ShutdownHook: service stopped.
omada-controller    | 2022-03-16 20:41:33 [Thread-1] [INFO]-[SourceFile:186] - Omada Controller exited
```

mbentley, 2022-03-16:
Do you have the very latest image?  There was an issue, pointed out in #189 and I am curious if it is related or if this is on the very latest.  Also I will need more info about how you're launching the container otherwise.

rucksman, 2022-03-16:
Deleted container and image, pulled image again, and now the container starts. Thanks!

closed 2022-03-16 by rucksman