mbentley/docker-omada-controller #191: Update image to support 5.1

vd-rd, 2022-03-21:
Thank you for this amazing work.
The new 5.1.17 image is launched, with several bugs fixed. Please update your configs to include this.
Source can be downloaded here: https://static.tp-link.com/upload/software/2022/202201/20220120/Omada_SDN_Controller_v5.0.30_linux_x64.tar.gz
If you don't have the bandwidth to do so, i will open a PR with the changes and submit it for your review.

Regards,
Vadim.

mbentley, 2022-03-21:
That's `5.0.30` that you linked to, not 5.1.17.  My scraper only shows the following versions available right now:

```
https://static.tp-link.com/upload/software/2022/202201/20220120/Omada_SDN_Controller_v5.0.30_linux_x64.tar.gz
https://static.tp-link.com/upload/software/2021/202112/20211217/Omada_SDN_Controller_v4.4.8_linux_x64.tar.gz
https://static.tp-link.com/upload/software/2021/202110/20211011/Omada_SDN_Controller_v4.4.6_Linux_x64.tar.gz
https://static.tp-link.com/upload/software/2021/202108/20210813/Omada_SDN_Controller_v4.4.4_linux_x64.tar.gz
https://static.tp-link.com/software/2021/202107/20210701/Omada_SDN_Controller_v4.4.3_linux_x64.tar.gz
https://static.tp-link.com/2021/202105/20210507/Omada_SDN_Controller_v4.3.5_linux_x64.tar.gz
https://static.tp-link.com/2021/202102/20210209/Omada_SDN_Controller_v4.2.11_linux_x64.tar.gz
https://static.tp-link.com/2020/202012/20201211/Omada_SDN_Controller_v4.2.8_linux_x64.tar.gz
https://static.tp-link.com/2020/202011/20201110/Omada_SDN_Controller_v4.2.4_linux_x64.tar.gz
https://static.tp-link.com/2020/202008/20200805/Omada_SDN_Controller_v4.1.5_linux_x64.tar.gz
https://static.tp-link.com/upload/software/2022/202201/20220125/Omada_Controller_v3.2.16_linux_x64.tar.gz
https://static.tp-link.com/2020/202012/20201225/Omada_Controller_v3.2.14_linux_x64.tar.gz
https://static.tp-link.com/2020/202004/20200420/Omada_Controller_v3.2.10_linux_x64.tar.gz
https://static.tp-link.com/2020/202004/20200409/Omada_Controller_v3.2.9_linux_x64.tar.gz
https://static.tp-link.com/2020/202001/20200116/Omada_Controller_v3.2.6_linux_x64.tar.gz
https://static.tp-link.com/2019/201911/20191108/Omada_Controller_v3.2.4_linux_x64.tar.gz
```

vd-rd, 2022-03-21:
My bad,
You are right, i pasted a wrong link.
It seems 5.1.7 is only released for Windows https://static.tp-link.com/upload/software/2022/202202/20220225/Omada_Controller_v5.1.7_Windows.zip for now.

sarvesh-lad, 2022-03-21:
I know this doesn't help but here is a changelog atleast for the windows version:

If you are planning to upgrade to this version from an old Controller (V3.2.14 or below), please read the Omada Controller Upgrade Guide in the installation package before upgrading the controller.

**Supported device models and firmware**
•	EAP
EAP110_V4 3.20.0 Build 20200525 Rel.36899 and above
EAP115_V4 3.20.0 Build 20200525 Rel.36931 and above
EAP225_V3 2.20.0 Build 20200422 Rel.70513 and above
EAP245_V3 2.20.0 Build 20200423 Rel.36779 and above
EAP115-Wall_V1 1.20.0 Build 20200509 Rel.63986 and above
EAP225-Wall_V2 1.20.0 Build 20200422 Rel.70504 and above
EAP110-Outdoor_V3 3.20.0 Build 20200511 Rel.33388 and above
EAP225-Outdoor_V1 1.20.0 Build 20200422 Rel.70543 and above
EAP660 HD, EAP650, EAP620 HD, EAP615-Wall, EAP610, EAP610-Outdoor, 
EAP265 HD, EAP230-Wall, EAP235-Wall
•	Switch
TL-SG3452P, TL-SG3452, TL-SG3428XMP, TL-SG3428MP, TL-SG3428X, 
TL-SG3428, TL-SG3210XHP-M2, TL-SG3210 V3, TL-SX3206HPP, TL-SX3016F, TL-SX3008F, TL-SG2428P, TL-SL2428P V4, TL-SG2210MP, TL-SG2218, 
TL-SG2210P V3.2 and above, TL-SG2008P, TL-SG2008 V3 and above.
•	Gateway
ER605 (TL-R605) v1/v2, ER7206 (TL-ER7206)

**New Features & Enhancements**
1) Added support for Private Pre-Shared Key (PPSK), which requires upgrading the EAP to the adapted firmware to be released soon. For PPSK without RADIUS, you can create PPSK profiles (Settings -- Profiles -- PPSK).
2) Added the Report page to view, export and email network reports.
3) Added the Heatmap function (Beta).
4) Added support for Facebook Wi-Fi 2.0 for wireless networks, which requires upgrading the EAP to the adapted firmware.
5) Added support for Upgrade Schedule (Settings -- Services).
6) Added support for IPTV function (Settings – Services -- IPTV), which requires upgrading your ER605/ER7206 firmware to version 1.2.0.
7) Added support to submit your feedback.
8) Added support to connect FTP/TFTP/SFTP/SCP file servers for backup and restore (Settings – Maintenance – Backup & Restore).
9) Added TP-Link email service to invite cloud users and reset password.
10) Added support for adding multiple authentication and accounting servers to a RADIUS profile.
11) Added support for the RADIUS accounting function, and NAS ID setting to the MAC-Based Authentication.
12) Added support for the update notification for Omada Software Controller (Settings – Maintenance – User Interface).
12) Optimized the portal customization, added support to adjust the logo size and position and font size, provided a more detailed customized page template.
13) Added the Device Account configuration during Quick Setup and adding a new site. After a device is adopted by the site, its username and password will be automatically set as the site device account, and will be required when re-adopting and logging into the device.
14) Added support to create up to 7 entries for one time range profile (Settings – Profiles – Time Range).
15) Added support for the USB LTE WAN of ER605(UN) V2.
16) Added the Wi-Fi Control function for some wall plate EAPs.
17) Added support to set the end of the Wi-Fi password as the space.
18) Added quick switches to enable or disable many advanced features, such as ACL, Port Forwarding, VPN, etc.
19) Added support to custom the Online Detection Interval, which requires upgrading your ER605/ER7206 firmware to version 1.2.0.
20) Added support to import HTTPS certification in PEM format (Settings – Controller – HTTPS Certificate).
21) Added support to export data from multiple sites (Settings – Services – Export Data).
22) Added a status filter to the device list to filter devices with selected status.
23) Added Controller and Device names for Syslog and Email.
24) Added the requirement for External Web Portal to submit “Access-Control-Allow-Origin” field in HTTP header to enhance security, for detailed information and new template, please refer to FAQ 2912.
25) Optimized PoE utilization widget in Dashboard.
26) Optimized the text formatting to solve problems such as incorrect line breaks.
27) Optimized traffic statistics of EAP.
28) Optimized the default sorting order of the device list.
29) Optimized the Speed Test Statistics page.
30) Removed “DHCP Omada Controller” as it duplicates the “DHCP Option 138” (Settings – Wired Networks -- LAN).
31) Optimized the color of “CONNECTED” marker in the Clients list.
32) Added support to display device Serial Number,

**Bug Fixed**
1) Upgrade log4j version to 2.17.0. Note that Omada SDN Controller is not impacted by the vulnerability CVE-2021-45105.
2) Remove the limitation on the length of the Top-Level Domain Name of SMTP Sender Address.
3) Fixed some UI bugs when configuring the switch with a static IP address or enabling the DHCP server.
4) Fixed the bug where the Client Detail Logs were not sent to the Syslog server when the client connected to the Mesh AP.
5) Fixed the bug that the settings of PAP and CHAP authentication mode are inverse for External RADIUS Server Portal.
6) Fixed the bug that incorrect DHCP Reservation configurations caused the management exception of Omada Gateway.
7) Fixed the bug that MAC address of EAP was displayed in the roaming logs instead of the Device Name.
8) Fixed the bug that SFP+ ports can only work in 1G mode when LACP is configured.
9) Fixed the bug that the channels were incorrectly modified during batch configuration of Tx power.

**Notes**

1) This version of the controller is fully applied to the Omada APP of version 4.2.X or above.
2) Omada SDN Controller can only manage certain devices running the supported firmware. Please confirm that your device is compatible with the SDN Controller.
3) If you are using an old Controller and plan to upgrade to this version, please follow the procedure of the Omada Controller Upgrade Guide.
4) Once upgraded to this version of Omada Controller, you will be NOT able to downgrade to version 3.2.14 or below.

mbentley, 2022-03-21:
Until the new version is available, I am going to go ahead and close this.

mbentley, 2022-03-22:
Well that didn't take long, [`5.1.7`](https://static.tp-link.com/upload/software/2022/202203/20220322/Omada_SDN_Controller_v5.1.7_Linux_x64.tar.gz) is now available.

commit 6e0c528f (2022-03-22): Merge pull request #192 from mbentley/add-5.1.7

Bumped for 5.1.17; fixes #191
commit 7e22a75c (2022-03-22): Bumped for 5.1.17; fixes #191
closed 2022-03-22 by mbentley