mbentley/docker-omada-controller #194: Potentially no need to convert PEM certificate to a Java Keystore from Omada Controller v5.1.7

dli417, 2022-03-22:
**Is your feature request related to a problem? Please describe.**
Currently, the Entrypoint script will convert a PEM certificate into a Java Keystore. This may not be necessary as Omada Controller v5.1.7 supports PEM certificates.

**Describe the solution you'd like**
If PEM certificates can be copied directly into their appropriate directory, the PEM-to-JKS conversion code could be removed from the Entrypoint script.

**Additional context**
From [v5.1.7 release notes](https://static.tp-link.com/upload/software/2022/202203/20220322/Release%20Note_Linux.pdf): "Added support to import HTTPS certification in PEM format (Settings – Controller – HTTPS Certificate)"

mbentley, 2022-03-22:
Good catch.  I'll have to think about a detection method to understand the version of the Omada Controller in the entrypoint.  I'd like to be able to re-use the same entrypoint script for as long as it makes sense in the `5.x` series without having to keep a `5.0.x`  and `5.1.x` version.

dli417, 2022-03-22:
I see. v5.1.7 still supports Java Keystores, so the current entrypoint script should be fine. Thank you for the quick turnaround in creating the v5.1.7 images :)

mbentley, 2022-03-22:
Sure thing.  And yes, it still supports JKS so it's fine.  I wouldn't be terrible to be able to support providing JKS, PEM, and PFX formats since the UI does but that would take time to test and implement; also knowing which versions support which.  At least JKS is the lowest common denominator as I do not know when PFX support was added.

dli417, 2022-03-23:
That makes sense to me. Please feel free to close this feature request.

closed 2022-03-23 by mbentley

mbentley, 2022-06-14:
Just for future reference, I was looking at this today.  It looks like when a certificate is uploaded through the web interface, it is inserted into the MongoDB and not stored in the filesystem so I am not sure that there is a way to handle PEM based certificates prior to startup.  One could automate the API calls to POST the certificate and key after the controller is up:

```
/<controller-id>/api/v2/files/controller/certificate
/<controller-id>/api/v2/files/controller/key
```

With the data looking something like:

```
data: {"cerName":"cert.pem","md5":"7e336a157e3937ee8c2bf18089e53ba1"}
data: {"keyName":"key_nopw.pem","md5":"a03afda2707f36a1dfdffc5c7a74eed7"}
```

But the controller container has to be restarted for the change to take effect as well.  Probably much easier to keep with the java keystore.