mbentley/docker-omada-controller #195: Unable to set custom ports with v3.2?

PatrikStaehli, 2022-04-01:
Hello
Custom ports are working with both 5.1 and 4.1, but the environment variables seem to be ignored with v3.2?
Unfortunately, I'm forced to use 3.2 as my devices (EAP 330 and 225v3) are not supported by higher versions (which also prevents me from adding newer models to the mix and I'm not going to replace 3 perfectly working devices just because tp-link wants to rip me off).
Am I doing something wrong, or is it simply not possible to change the ports for 3.2?
Thanks a lot for providing these containers in the first place!
Kind regards
Patrik

mbentley, 2022-04-03:
Sorry, it isn't documented well in the readme but for v3.2, the entrypoint doesn't support setting the ports.  Are you needing the amd64 image?  If so, I just built a test image that you can try:  `mbentley/omada-controller:3.2-test`

I don't have an extensive test setup for the older images so I am not sure if there are any issues and I could only find variables to support the equivalent of `MANAGE_HTTP_PORT` and `MANAGE_HTTPS_PORT`.  I don't even recall if 3.2 has guest portal functionality or not.

mbentley, 2022-04-03:
Changes are in PR https://github.com/mbentley/docker-omada-controller/pull/196; awaiting testing.

PatrikStaehli, 2022-04-04:
Thanks for the quick response!
Unfortunately, I'm working on a Raspberry Pi 4. 'uname -m' says 'armv7l'.
I guess the error `standard_init_linux.go:228: exec user process caused: exec format error` is the result of running the wrong architecture?
Would you mind providing an arm version?
Kind regards and thank you very much

mbentley, 2022-04-04:
Ah sure.  It's probably easiest if I just merge the changes and let my CI build images.  This change won't break any existing capabilities as it is only additive.

mbentley, 2022-04-04:
I merged in the changes and the new images are now on Docker Hub.  Could you give the updated `3.2` tagged image a try with the variables?

mbentley, 2022-04-04:
Re-opening until you can confirm that it works as expected.

PatrikStaehli, 2022-04-04:
Hello @mbentley 
Yes, I can confirm it's working!
Thank's a lot, it's very much appreciated.

mbentley, 2022-04-04:
Awesome, thanks for the quick test!

commit 9c06df0e (2022-04-04): Merge pull request #196 from mbentley/add-custom-ports-3.2

Added custom ports for management interfaces; fixes #195
commit d883ee96 (2022-04-04): Added custom ports for management interfaces; fixes #195
closed 2022-04-04 by mbentley