mbentley/docker-omada-controller #197: mongo out of memory because of 32-bit version?

tacognito, 2022-04-05:
**Describe the bug**

I have the container running on an rpi4 and it corrupts everyday, after having run smoothly for a year. Restoring from full file backup of mounted volumes solves it. But reoccurs within a day. 

The mongo db log states: 
`ERROR: mmap private failed with out of memory. You are using a 32-bit build and probably need to upgrade to 64.`

Isn't 32-bit only for dev purposes only normally? Not an mongo expert though.

Container runs in host mode (so not expecting issues with recent port addition), but anyway the log so clearly states mongodb running out of memory that definitely mongo is the place to look for a fix. 

**To Reproduce**
Steps to reproduce the behavior:

1. restore container from backup of mounted volumes
2. let run for a day
3. mongo fails, container restarts, keeps hanging in starting state

**Expected behavior**

Mongo not running out of memory.

**How you're launching your container**

docker-compose

```
omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:latest-armv7l
    environment:
      - TZ=Europe/Amsterdam
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
    network_mode: host
    volumes:
      - /opt/omada-controller/data:/opt/tplink/EAPController/data
      - /opt/omada-controller/work:/opt/tplink/EAPController/work
      - /opt/omada-controller/logs:/opt/tplink/EAPController/logs
    dns:
      - 1.1.1.1
    restart: always
```

**Container Logs**

mongodb.log:
```
2022-04-05T12:58:38.285+0200 ***** SERVER RESTARTED *****
2022-04-05T12:58:38.294+0200 [initandlisten] MongoDB starting : pid=38 port=27217 dbpath=../data/db 32-bit host=raspberrypi
2022-04-05T12:58:38.294+0200 [initandlisten] 
2022-04-05T12:58:38.294+0200 [initandlisten] ** NOTE: This is a 32 bit MongoDB binary.
2022-04-05T12:58:38.294+0200 [initandlisten] **       32 bit builds are limited to less than 2GB of data (or less with --journal).
2022-04-05T12:58:38.294+0200 [initandlisten] **       See http://dochub.mongodb.org/core/32bit
2022-04-05T12:58:38.294+0200 [initandlisten] 
2022-04-05T12:58:38.294+0200 [initandlisten] db version v2.6.10
2022-04-05T12:58:38.294+0200 [initandlisten] git version: nogitversion
2022-04-05T12:58:38.294+0200 [initandlisten] OpenSSL version: OpenSSL 1.0.2g  1 Mar 2016
2022-04-05T12:58:38.294+0200 [initandlisten] build info: Linux kishi10 3.2.0-84-highbank #121-Ubuntu SMP PREEMPT Tue May 5 19:44:15 UTC 2015 armv7l BOOST_LIB_VERSION=1_58
2022-04-05T12:58:38.294+0200 [initandlisten] allocator: tcmalloc
2022-04-05T12:58:38.294+0200 [initandlisten] options: { net: { bindIp: "127.0.0.1", port: 27217 }, processManagement: { pidFilePath: "../data/mongo.pid" }, storage: { dbPath: "../data/db", journal: { enabled: true } }, systemLog: { destination: "file", logAppend: true, path: "../logs/mongod.log" } }
2022-04-05T12:58:38.295+0200 [initandlisten] 
2022-04-05T12:58:38.295+0200 [initandlisten] ** WARNING: Readahead for ../data/db is set to 65532KB
2022-04-05T12:58:38.295+0200 [initandlisten] **          We suggest setting it to 256KB (512 sectors) or less
2022-04-05T12:58:38.295+0200 [initandlisten] **          http://dochub.mongodb.org/core/readahead
2022-04-05T12:58:38.300+0200 [initandlisten] journal dir=../data/db/journal
2022-04-05T12:58:38.300+0200 [initandlisten] recover : no journal files present, no recovery needed
2022-04-05T12:58:38.314+0200 [initandlisten] ERROR: mmap private failed with out of memory. You are using a 32-bit build and probably need to upgrade to 64
2022-04-05T12:58:38.315+0200 [initandlisten] Assertion: 13636:file /opt/tplink/EAPController/lib/../data/db/omada_data.4 open/create failed in createPrivateMap (look in log for more information)
2022-04-05T12:58:38.323+0200 [initandlisten] 0x7639be 0x72180a 0x70a5f0 0x70a68c 0x5cb9b0 0x5cbac4 0x5c9cd6 0x5d3f7c 0x297d6e 0x29bf7e 0x29f6fa 0x2afeb4 0x2b002c 0x1ee366 0x1ef78a 0x1f009c 0x1c4f8c 0xb671d8aa 
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo15printStackTraceERSo+0x25) [0x7639be]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo10logContextEPKc+0x15d) [0x72180a]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11msgassertedEiPKc+0x8f) [0x70a5f0]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo11msgassertedEiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x7) [0x70a68c]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo17DurableMappedFile13finishOpeningEv+0x283) [0x5cb9b0]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo17DurableMappedFile4openERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb+0xb3) [0x5cbac4]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo8DataFile12openExistingEPKc+0xed) [0x5c9cd6]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo13ExtentManager4initEv+0xab) [0x5d3f7c]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo8Database12openAllFilesEv+0x1d) [0x297d6e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo8DatabaseC1EPKcRbRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x1ad) [0x29bf7e]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo14DatabaseHolder11getOrCreateERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES8_Rb+0x649) [0x29f6fa]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo6Client7Context11_finishInitEv+0x2b) [0x2afeb4]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo6Client7ContextC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_b+0x67) [0x2b002c]
 /opt/tplink/EAPController/bin/mongod() [0x1ee366]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo14_initAndListenEi+0x5ed) [0x1ef78a]
 /opt/tplink/EAPController/bin/mongod(_ZN5mongo13initAndListenEi+0x13) [0x1f009c]
 /opt/tplink/EAPController/bin/mongod(main+0x197) [0x1c4f8c]
 /lib/arm-linux-gnueabihf/libc.so.6(__libc_start_main+0x9d) [0xb671d8aa]
2022-04-05T12:58:38.323+0200 [initandlisten] warning database ../data/db omada_data could not be opened
2022-04-05T12:58:38.323+0200 [initandlisten] DBException 13636: file /opt/tplink/EAPController/lib/../data/db/omada_data.4 open/create failed in createPrivateMap (look in log for more information)
2022-04-05T12:58:38.324+0200 [initandlisten] exception in initAndListen: 13636 file /opt/tplink/EAPController/lib/../data/db/omada_data.4 open/create failed in createPrivateMap (look in log for more information), terminating
2022-04-05T12:58:38.325+0200 [initandlisten] dbexit: 
2022-04-05T12:58:38.325+0200 [initandlisten] shutdown: going to close listening sockets...
2022-04-05T12:58:38.325+0200 [initandlisten] shutdown: going to flush diaglog...
2022-04-05T12:58:38.325+0200 [initandlisten] shutdown: going to close sockets...
2022-04-05T12:58:38.325+0200 [initandlisten] shutdown: waiting for fs preallocator...
2022-04-05T12:58:38.325+0200 [initandlisten] shutdown: lock for final commit...
2022-04-05T12:58:38.325+0200 [initandlisten] shutdown: final commit...
2022-04-05T12:58:38.325+0200 [initandlisten] shutdown: closing all files...
2022-04-05T12:58:38.326+0200 [initandlisten] closeAllFiles() finished
2022-04-05T12:58:38.326+0200 [initandlisten] journalCleanup...
2022-04-05T12:58:38.326+0200 [initandlisten] removeJournalFiles
2022-04-05T12:58:38.331+0200 [initandlisten] shutdown: removing fs lock...
2022-04-05T12:58:38.332+0200 [initandlisten] dbexit: really exiting now
```

**Additional context**

tacognito, 2022-04-05:
Just realizing that armv7l is a 32 bit processor. Would that mean that the only way is to reduce info stored in mongo? And how? Maybe disable/reduce reporting?

tacognito, 2022-04-05:
> What does `uname -a` return from your raspberry pi?

armv7l

mbentley, 2022-04-05:
I believe that starting with the Raspberry Pi 3, they supported 64 bit instructions (and operating systems) so if you did have the ability to re-install with a 64 bit version of an OS, it likely wouldn't be an issue, assuming you have the memory on your pi.  I'll have to look to see if there is any way to reduce the memory footprint at all.

tacognito, 2022-04-05:
Thanks for the great container and a wowie for your amazingly swift responses. Obviously, raspbian is very common for the pi and I just read [here](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/) that:

1. You are right (obviously, I guess)
2. They just released a 64 bit version this February

It would be a huge pain to reinstall the pi, so would love to be able to keep the mongo db manageable. No journaling could help, hopefully writing will be stable. I'm running it on an ssd so that probably helps. Would implementing sharding do the trick? Though I can expect that might be way too much of a pain in the ass to implement.

tacognito, 2022-04-05:
I'll do some research tomorrow, will let you know!

tacognito, 2022-04-06:
Well, this [link](https://stackoverflow.com/questions/12319579/how-to-keep-32-bit-mongodb-memory-usage-down-on-changing-dataset#12321768) explains very well that running mongodb on 32-bit in production is just in all means a bad idea. So what I did is migrate the container to my windows Intel NUC. Also, not ideal, but 64-bit which should keep it stable. I'll try to gather the nerve to upgrade my pi in the coming weeks :-)

Thanks for all the info, advice and awesome work!

tacognito, 2022-04-06:
Btw, had to restore from config which doesn't contain all the client hostname renaming. Which is a big shame. Do you happen to know the credentials with which I could peek in the mongo db of the old installation and see if I can salvage it?

mbentley, 2022-04-06:
There are no credentials - it's just `mongo mongodb://127.0.0.1:27217` after installing the mongo db client in the container.

tacognito, 2022-04-06:
Cool, I guess we can close this issue. Thanks!

closed 2022-04-06 by tacognito