mbentley/docker-omada-controller #199: Data directory grows too much and quickly

jorgejams88, 2022-04-20:
**Describe the bug**
After regular usage the  data directory increases size too quick

**To Reproduce**
Normally use the container

**Expected behavior**
The size should not increase drastically

**How you're launching your container**
```
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.0
    environment:
      - TZ=America/Guayaquil
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - PGID=1000
      - PUID=1000
#      - SSL_CERT_NAME="tls.crt"
#      - SSL_KEY_NAME="tls.key"
    network_mode: host
    depends_on:
      - "fluentd_host"

    volumes:
      - ./omada/data:/opt/tplink/EAPController/data
      - ./omada/work:/opt/tplink/EAPController/work
      - ./omada/logs:/opt/tplink/EAPController/logs
    restart: unless-stopped  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.0
    environment:
      - TZ=America/Guayaquil
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - PGID=1000
      - PUID=1000
    network_mode: host
    volumes:
      - ./omada/data:/opt/tplink/EAPController/data
      - ./omada/work:/opt/tplink/EAPController/work
      - ./omada/logs:/opt/tplink/EAPController/logs
    restart: unless-stopped
```

**Additional context**
After a month, the "data" directory consumes 500MB, making  backups larger and larger. Is there a way to purge something to decrease backup size?

mbentley, 2022-04-20:
I don't think there is anything that can be done there but I have a long running controller, managing 3 APs and it's not using much in the way of disk space:

```
# du -hs *
14K     cert
339M    data
3.8M    logs
512     work
```

That's something that is probably better asked on the forums as the only previous thing that could be done is using the `--smallfiles` arg for [MongoDB but as the readme states](https://github.com/mbentley/docker-omada-controller#mongodb-small-files), that is no longer valid for newer version.

jorgejams88, 2022-04-20:
Thanks, understood!!!

closed 2022-04-20 by jorgejams88