mbentley/docker-omada-controller #200: Ubuntu 20.04 support for Omada Software Controller V5

markus-gitdev, 2022-05-07:
**Is your feature request related to a problem? Please describe.**
Your docker image still uses Ubuntu 18.04, but regarding the information from [TP-Link](https://www.tp-link.com/us/support/download/omada-software-controller/#Controller_Software) Ubuntu 20.04 is also supported for V5 of their Omada Software Controller.

**Describe the solution you'd like**
It would be great if you could update the docker image to use Ubuntu 20.04.

mbentley, 2022-05-07:
Just curious for understanding priority - any other specific needs beyond just that it isn't the latest Ubuntu?  I do plan on having it updated prior to the end of standard support for 18.04 which is April 2023 though.

markus-gitdev, 2022-05-07:
No urgency.
I just wanted to draw attention to the topic, as Ubuntu is 2 LTS releases behind.

mbentley, 2022-05-07:
Good deal. Just making sure you weren't doing anything like extending the image where a package from 20.04 was preferred.

closed 2022-05-08 by mbentley

markus-gitdev, 2022-05-08:
That was fast! Thank you for the fast and smooth implementation.

commit bb300b56 (2022-05-08): Updated to Ubuntu 20.04 for 5.x; fixes #200
commit 9572178a (2022-05-08): Merge pull request #203 from mbentley/update-ubuntu-5.x

Updated to Ubuntu 20.04 for 5.x; fixes #200