mbentley/docker-omada-controller #207: pem certificates

Anzic23, 2022-06-02:
**Is your feature request related to a problem? Please describe.**
I have a container that renews certificates every 3 months. I use these certificates with all my containers by referencing them. But the problem is that certificates are created with *.pem extension
In the interface of the controller, it is possible to download a PEM certificate, but then it will become obsolete after 3 months and will not be automatically updated. Is it possible to specify links specifically to *.pem files?

**Describe the solution you'd like**
create links to pem certificates in docker compose

**Additional context**
version: "3.1"

services:
   omada-controller:
     container_name: omada-controller
     hostname: omada
     image: mbentley/omada-controller:latest
     networks:
      default:
        ipv4_address: 172.23.0.10
     ports:
     - 8088:8088
     - 8043:8043
     - 8843:8843
     - 29810:29810/udp
     - 29811:29811
     - 29812:29812
     - 29813:29813
     - 29814:29814
     environment:
      - TZ=Europe/Moscow
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=true
      - SSL_CERT_NAME="fullchain.pem"
      - SSL_KEY_NAME="privkey.pem"
     
     volumes:
      - /volume2/doker_mapping/omada/omada-data:/opt/tplink/EAPController/data
      - /volume2/doker_mapping/omada/omada-work:/opt/tplink/EAPController/work
      - /volume2/doker_mapping/omada/omada-logs:/opt/tplink/EAPController/logs
      - /volume2/doker_mapping/letsencrypt/nginx/live/huako.ru:/cert:ro
     restart: unless-stopped

networks:
    default:
        external:
            name: nginx-proxy

mbentley, 2022-06-02:
So I also use LetsEncrypt for my certs.  I let the the cert renew via cron and then I have a reload command configured in my letsencrypt client that automatically runs to copy the certs from my acme client's directory over to where I store the certs specifically for my controller.  In my case, my controller has the certs stored in `/zfs/apps/omada-controller/cert` which I then bind mount `/zfs/apps/omada-controller/cert` to `/cert`.  

My reload command executes the following in a shell script:

```
cp /root/.acme.sh/\*.casa.mbentley.net/fullchain.cer /zfs/apps/omada-controller/cert/tls.crt
cp /root/.acme.sh/\*.casa.mbentley.net/\*.casa.mbentley.net.key /zfs/apps/omada-controller/cert/tls.key
docker restart -t 30 omada-controller
```

This will automatically restart the controller after the certs are renewed and they will be automatically applied.  I bet your LE client has a capability to automatically execute a reload command.  You might have to do some additional trickery like adding the docker client and bind mounting the docker socket to get it to work because you renew in a container but it should be possible.

Anzic23, 2022-06-02:
in truth, I suffer with certificates to get the container working through Nginx Proxy Manager. According to this [post](https://github.com/mbentley/docker-omada-controller/issues/168#issuecomment-1053762649), it is possible, but I can't change the port in the controller settings to 443, even though I run it in privileged mode. Maybe you can add some kind of inventory so that you can specify ports below 1024?

mbentley, 2022-06-02:
Did you see this note about using [unprivileged ports](https://github.com/mbentley/docker-omada-controller#unprivileged-ports)?

Anzic23, 2022-06-02:
> Did you see this note about using [unprivileged ports](https://github.com/mbentley/docker-omada-controller#unprivileged-ports)?

yes, of course, but I'm creating a container using docker compose, and for it, it seems like you need to use this line:
```
my_service:
  privileged: true
```
Or is this not the right decision? How can I run a container in privileged mode via docker compose?

mbentley, 2022-06-02:
That's not the correct thing to use.  Check out the docs: https://docs.docker.com/compose/compose-file/#sysctls and here is an example:

```
        sysctls:
            - net.ipv4.ip_unprivileged_port_start=0
```

Anzic23, 2022-06-02:
> sysctls:
>             - net.ipv4.ip_unprivileged_port_start=0

Recreating omada-controller ... error

ERROR: for omada-controller  Cannot start service omada-controller: OCI runtime create failed: container_linux.go:367: starting container process caused: process_linux.go:495: container init caused: write sysctl key net.ipv4.ip_unprivileged_port_start: open /proc/sys/net/ipv4/ip_unprivileged_port_start: no such file or directory: unknown

so i tried, i get error:
```
ERROR: for omada-controller  Cannot start service omada-controller: OCI runtime create failed: container_linux.go:367: starting container process caused: process_linux.go:495: container init caused: write sysctl key net.ipv4.ip_unprivileged_port_start: open /proc/sys/net/ipv4/ip_unprivileged_port_start: no such file or directory: unknown
ERROR: Encountered errors while bringing up the project.

```

mbentley, 2022-06-02:
No idea why that would be happening.  You should be able to `cat /proc/sys/net/ipv4/ip_unprivileged_port_start` from the host and the container and it should work fine unless there is something about your linux distro that doesn't have that for some reason.

Anzic23, 2022-06-02:
`cat: /proc/sys/net/ipv4/ip_unprivileged_port_start: No such file or directory`
i use synology. I'll try to find out how this command is done in it

Anzic23, 2022-06-03:
I found that on synology there is no prohibition on using ports below 1024, at least this docker-compose creates a container without problems:
```
version: "2"
services:
  adguard:
    image: adguard/adguardhome:latest
    networks:
      default:
        ipv4_address: 172.23.0.13
    container_name: adguardhome
    hostname: adguardhome
    ports:
      - '53:53/tcp'
      - '53:53/udp'
      - '67:67/udp'
#      - '68:68/tcp'
#      - '68:68/udp'
      - '5280:80/tcp'
      - '8943:443/tcp'
      - '853:853/tcp'
      - '4000:3000/tcp'
    volumes:
      - /volume2/doker_mapping/adguard/data:/opt/adguardhome/work
      - /volume2/doker_mapping/adguard/config:/opt/adguardhome/conf
      - /volume2/doker_mapping/letsencrypt/nginx:/opt/adguardhome/work/certs:ro
    restart: unless-stopped

networks:
    default:
        external:
            name: nginx-proxy
```
I don't need the container to use host ports below 1024, because I'm associating host port 8043 with container port 443, so I just need the ability to specify port 443 inside the container. Outside it will be 8043. But as I said, the software does not allow me to do this.

mbentley, 2022-07-14:
Closing due to inactivity - let me know if I can be of further help.

closed 2022-07-14 by mbentley