mbentley/docker-omada-controller #212: Change user/group of container

laheym, 2022-07-05:
**Is your feature request related to a problem? Please describe.**
Change the user/group (defined with an environment variable) of the container after startup.  This will allow administrators to set appropriate permissions on the persistent volumes

**Describe the solution you'd like**
Add two environment variables to the container (e.g. linuxserver.io containers typically uses PUID and PGID) that define the user and group that the container runs as.

mbentley, 2022-07-05:
Shouldn't be a problem to implement - I'm already using `gosu` to change to the `omada` user; would just need to move the creation of the `omada` user from the install.sh script to the entrypoint script to execute as a part of the entrypoint; likely the very beginning.

I'll probably only backport it to the 4.x and 5.x versions; maybe also 3.2.x just because it is last of the older version that supports some older versions of hardware.

laheym, 2022-07-05:
Thanks Matt.  And thanks for creating the container - it made entry into the omada ecosystem much easier.

commit 4591ebca (2022-07-06): Added PUID/PGID env vars to set UID/GID at runtime; fixes #212
commit e0cb7bc8 (2022-07-06): Merge pull request #213 from mbentley/runtime-uid-gid

Added PUID/PGID env vars to set UID/GID at runtime; fixes #212
commit 1378a363 (2022-07-11): Added PUID/PGID env vars to set UID/GID at runtime; fixes #212
closed 2022-07-06 by mbentley