mbentley/docker-omada-controller #214: Cannot generate reports on Omada Software Controller 5.3.1

Adandu, 2022-07-11:
Hello,

I need some help in exporting the reports from the Omada Software Controller v5.3.1
I run the controller in a Docker container on a Synology NAS.
When I try to export the report as PDF from the Report tab i get the following error:
**Please install Chromium before exporting the network report. Make sure you can run Chromium as root.**

[image]

If i try to schedule the report I get the below error in the Log tab under alerts.
[image]

I have opened a thread on the TP-Link Forum and someone pointed out to this issue/solution:
https://github.com/microsoft/playwright/issues/2205

I have no idea how I could do that with the docker image you provide. I'm not that good with docker or coding :)
But, maybe it helps you.

Any help is much apreciated.

Thank you!

mbentley, 2022-07-11:
Thanks for the report.  I'm trying to see if I can get a version working with Chromium but I am running into some controller software issues and no helpful logs so I posted [on your post on the forums](https://community.tp-link.com/en/business/forum/topic/569686?replyId=1111892) to see if I can get help from someone who understands what the controller is trying to do.

mbentley, 2022-07-11:
I found a container related bug related to persistent data in the `data` directory where some of the needed files were missing because of how the software is packaged in the container and how the suggested mounts are configured.

I also did add a Dockerfile that extends the existing `5.3` image to add Chromium [here](https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.v5.3.x-chromium).  There is an image that has been pushed to Docker Hub as well (`mbentley/omada-controller:5.3-chromium`) but it throws the error as I posted on the forums.

zagub59, 2022-08-19:
Hi !

Also following this thread as it'd be great to generate reports. 

Tried out the 5.4.6 chromium version but I get an error "Failed to export report due to damaged software. It is recommended to reinstall the controller."

Any update on this ? 

By the way, thank for your great work ! Been using your container for over a year in multiple environments and it's been working flawlessly. Keep it up !

mbentley, 2022-08-19:
Nope, no updates yet on the [forums post](https://community.tp-link.com/en/business/forum/topic/569686?replyId=1111892) as I am stuck not knowing what the dependencies are until they update the documentation.

mbentley, 2022-08-22:
Oh well, this explains a lot...stupid snaps.

```
root@09e4f77588d9:/usr/bin# chromium-browser --version

Command '/usr/bin/chromium-browser' requires the chromium snap to be installed.
Please install it with:

snap install chromium
```

Will have to work around this.

closed 2022-08-22 by mbentley

mbentley, 2022-08-22:
OK, so I have `amd64` builds working now for versions 5.1 - 5.5, such as `mbentley/omada-controller:5.5-chromium`.

mbentley, 2022-08-22:
The PPA I used only has `amd64` packages so if other architectures like `arm64` are required, I will have to come up with another solution.

zagub59, 2022-08-23:
Big thanks for the time spent on this, it's all working now ! :)

commit 2f923a37 (2022-08-22): Merge pull request #244 from mbentley/fix-chromium-for-reports

Changed to use PPA for chromium; fixes #214
commit bb3ef49c (2022-08-22): Changed to use PPA for chromium; fixes #214