mbentley/docker-omada-controller #217: Can't enable data retention when using port 443 for management

BJD1997, 2022-07-13:
**Describe the bug**
When using port 443 for management I can't turn on data retention under controller settings.
It complains to use port 1024 or higher. So I can't save the setting.
If you change the management port and then change the ports in the docker container the web interface works again.
But when changing it back to 443 using docker the controller seems broken.

**To Reproduce**

1. run controller in docker using ports in configuration below
2. Setup controller like you normally would do
3. Go to settings -> Controller
4. tick the "Collect Clients' History Data" box and save

**Expected behavior**
Save the setting

**How you're launching your container**
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  --sysctl net.ipv4.ip_unprivileged_port_start=0 \
  -p 80:80 \
  -p 443:443 \
  -p 8088:8088 \
  -p 8843:8843 \
  -p 29810:29810/udp \
  -p 29811:29811 \
  -p 29812:29812 \
  -p 29813:29813 \
  -p 29814:29814 \
  -e MANAGE_HTTP_PORT=80 \
  -e MANAGE_HTTPS_PORT=443 \
  -e PGID="508" \
  -e PORTAL_HTTP_PORT=8088 \
  -e PORTAL_HTTPS_PORT=8843 \
  -e PUID="508" \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e SSL_CERT_NAME="tls.crt" \
  -e SSL_KEY_NAME="tls.key" \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  -v omada-certs:/cert \
  mbentley/omada-controller:5.3

**Container Logs**

```
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'Etc/UTC'
INFO: Setting 'manage.http.port' to 80 in omada.properties
INFO: Setting 'manage.https.port' to 443 in omada.properties
WARN: ownership not set correctly on '/opt/tplink/EAPController/work'; setting correct ownership (omada:omada)
INFO: Importing cert from /cert/tls.[key|crt]
INFO: Starting Omada Controller as user omada
07-13-2022 12:50:08.120 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(32): success to load configuration omada.properties
07-13-2022 12:50:08.191 INFO [main] [] c.t.s.o.s.OmadaBootstrap(86): going to start local mongod.
07-13-2022 12:50:11.217 INFO [main] [] c.t.s.o.s.s.b(161): mongodb process id is 71
07-13-2022 12:50:11.220 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Mongo DB server started
07-13-2022 12:50:11.221 INFO [main] [] c.t.s.o.s.s.b(142): Mongo DB server started

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
[... 89 lines cut ...]
07-13-2022 12:51:08.736 INFO [manage-work-group-9] [] c.t.s.o.m.d.p.t.c(129): Device 54-AF-97-72-AE-FE OmadacId(34e1a567cdb6364a4192adf8f1370002) changed to status CONNECTED, which don't need to handle.
07-13-2022 12:51:08.791 INFO [manage-work-group-10] [] c.t.s.o.m.d.d.m.i.a(86): got first inform of OmadacId(34e1a567cdb6364a4192adf8f1370002) DeviceMac(54-AF-97-72-AE-FE)
07-13-2022 12:51:08.809 INFO [manage-work-group-11] [] c.t.s.o.m.d.d.m.i.d(194): first inform send same version config to OmadacId(34e1a567cdb6364a4192adf8f1370002) DeviceMac(54-AF-97-72-AE-FE)
07-13-2022 12:51:08.867 INFO [manage-work-group-12] [] c.t.s.o.m.d.d.m.d.a.N(100): syncConfigurationForSameVersion to OmadacId(34e1a567cdb6364a4192adf8f1370002) DeviceMac(54-AF-97-72-AE-FE), result:SendDeviceMsgResult(success=true, deviceResponse=com.tplink.smb.omada.device.manage.message.body.manage.BaseConfigRespBody@1e6edde)
07-13-2022 12:51:19.637 WARN [manage-work-group-15] [] c.t.s.o.m.d.d.m.i.a.UpdateApImageCommonInfoService(112): omadacId:OmadacId(34e1a567cdb6364a4192adf8f1370002), siteId:SiteId(62cea573d78794263c9445ca), deviceMac:54-AF-97-72-AE-FE, apImage wlanId is null, need query database
07-13-2022 12:51:30.797 INFO [check-update-work-group-0] [] c.t.s.o.m.b.e(153): Start sync Cloud Users OmadacId(34e1a567cdb6364a4192adf8f1370002).
07-13-2022 12:51:30.938 INFO [check-update-work-group-0] [] c.t.s.o.m.b.e(118): Start checking last firmwares of managed devices.
07-13-2022 12:51:30.951 INFO [check-update-work-group-0] [] c.t.s.o.m.b.b(95): Checking Device Last Firmware...
07-13-2022 12:51:31.942 INFO [check-update-work-group-0] [] c.t.s.o.m.b.e(128): Checking last firmwares of managed devices succeed.
07-13-2022 12:51:51.950 INFO [Thread-11] [] c.t.e.c.c.q(891): send tp cloud heartbeat
07-13-2022 12:51:52.051 INFO [Thread-11] [] c.t.e.c.c.q(893): heartbeat reponse: {"error_code":0,"id":3}
07-13-2022 12:51:53.775 INFO [log-device-status-push-info-event-pool-1] [] c.t.s.o.l.c.b(50): Start pushing new connected devices.
07-13-2022 12:52:52.052 INFO [Thread-11] [] c.t.e.c.c.q(891): send tp cloud heartbeat
07-13-2022 12:52:52.153 INFO [Thread-11] [] c.t.e.c.c.q(893): heartbeat reponse: {"error_code":0,"id":4}
07-13-2022 12:53:52.154 INFO [Thread-11] [] c.t.e.c.c.q(891): send tp cloud heartbeat
07-13-2022 12:53:52.255 INFO [Thread-11] [] c.t.e.c.c.q(893): heartbeat reponse: {"error_code":0,"id":5}
07-13-2022 12:54:52.256 INFO [Thread-11] [] c.t.e.c.c.q(891): send tp cloud heartbeat
07-13-2022 12:54:52.358 INFO [Thread-11] [] c.t.e.c.c.q(893): heartbeat reponse: {"error_code":0,"id":6}
07-13-2022 12:55:52.358 INFO [Thread-11] [] c.t.e.c.c.q(891): send tp cloud heartbeat
07-13-2022 12:55:52.460 INFO [Thread-11] [] c.t.e.c.c.q(893): heartbeat reponse: {"error_code":0,"id":7}
07-13-2022 12:56:52.460 INFO [Thread-11] [] c.t.e.c.c.q(891): send tp cloud heartbeat
07-13-2022 12:56:52.562 INFO [Thread-11] [] c.t.e.c.c.q(893): heartbeat reponse: {"error_code":0,"id":8}
07-13-2022 12:57:52.563 INFO [Thread-11] [] c.t.e.c.c.q(891): send tp cloud heartbeat
07-13-2022 12:57:52.664 INFO [Thread-11] [] c.t.e.c.c.q(893): heartbeat reponse: {"error_code":0,"id":9}
```

mbentley, 2022-07-13:
Interesting.  I will have to see if I can reproduce this.  I use ports 80/443 and I have `Collect Clients' History Data` enabled in `History Data Retention`.  I know that at one point, I was using the default ports so I may have had this setting already set before switching the ports over to 80/443 so I will have to setup a new controller to test when I have the chance.

BJD1997, 2022-07-13:
Thanks for your reply. Let me know if I can help :)

mbentley, 2022-07-13:
Yeah, I can definitely reproduce this.  So I can see it's the software blocking the settings from being saved.  Not sure if they added a check to validate the port range at some point; I'll have to check previous versions to see if they had a port check in the UI as well.

mbentley, 2022-07-14:
OK, so after checking version 4.1 to the latest, they removed 443 from the list of allowed ports in 5.1:

5.1
[image]

The best I can do is file an issue on the forums to see if they can add it back, remove the restriction, or just do a proper check to see if they can bind to privileged ports.

BJD1997, 2022-07-14:
So as a workaround it would be enough to setup the controller on a older version and after that upgrade to the latest version?

BJD1997, 2022-07-14:
Also another quick question (unrelated to this). Since you have some experience packaging docker containers (of existing applications). Do you have a good resource for me to try and build my own docker containers?

mbentley, 2022-07-14:
So a better workaround is by using the API.  Below is a script that will utilize `curl` and `jq` which you need to install to enable the data retention.  If you need a linux CLI where you can get those tools, you can just run this container which has both and will drop you in a shell:
```
docker run -it --rm mbentley/swissarmy
```

The last command that does a `PATCH` will set the retention.  You need to adjust the value (it appears to be in days) from 180 to whatever you want and `clientsDataEnable":true` will enable it:

```
#!/bin/bash

# set variables
OMADA_URL="https://localhost:443"
USERNAME="admin"
PASSWORD="password"

# get controller id from the API
CONTROLLER_ID="$(curl -sk "${OMADA_URL}/api/info" | jq -r .result.omadacId)"

# login, get token, set & use cookies
TOKEN="$(curl -sk -X POST -c "/tmp/omada-cookies.txt" -b "/tmp/omada-cookies.txt" -H "Content-Type: application/json" "${OMADA_URL}/${CONTROLLER_ID}/api/v2/login" -d '{"username": "'"${USERNAME}"'", "password": "'"${PASSWORD}"'"}' | jq -r .result.token)"

# set data retention (180 = 6mo)
curl -sk -X PATCH -b "/tmp/omada-cookies.txt" -H "Content-Type: application/json" -H "Csrf-Token: ${TOKEN}" "${OMADA_URL}/${CONTROLLER_ID}/api/v2/controller/history?token=${TOKEN}" -d '{"retention":180,"clientsDataEnable":true}'
```

When you run this, you should see output like:

```
$ bash test.sh
{
  "errorCode": 0,
  "msg": "Success.",
  "result": {
    "login": true
  }
}
{"errorCode":0,"msg":"Success."}
```

And then you should be able to see in the UI that the value has changed.

mbentley, 2022-07-14:
> Also another quick question (unrelated to this). Since you have some experience packaging docker containers (of existing applications). Do you have a good resource for me to try and build my own docker containers?

Will have to get back to you on that.  Been a looooong time since I've gone down that route as I started Dockerizing apps in 2014 and am mostly self taught and learning through others while I was working at Docker.

mbentley, 2022-07-15:
I was wondering if it was possible to set the port to 443 from the Omada Controller API but it looks like they perform the validation from the API, not just the UI:

```
#!/bin/bash

# set variables
OMADA_URL="https://localhost:8043"
USERNAME="admin"
PASSWORD="password"

# get controller id from the API
CONTROLLER_ID="$(curl -sk "${OMADA_URL}/api/info" | jq -r .result.omadacId)"

# login, get token, set & use cookies
TOKEN="$(curl -sk -X POST -c "/tmp/omada-cookies.txt" -b "/tmp/omada-cookies.txt" -H "Content-Type: application/json" "${OMADA_URL}/${CONTROLLER_ID}/api/v2/login" -d '{"username": "'"${USERNAME}"'", "password": "'"${PASSWORD}"'"}' | jq -r .result.token)"

# change controller port
curl -sk -X PATCH -b "/tmp/omada-cookies.txt" -H "Content-Type: application/json" -H "Csrf-Token: ${TOKEN}" "${OMADA_URL}/${CONTROLLER_ID}/api/v2/controller/setting?token=${TOKEN}" -d '{"webPort":{"manageHttpsPort":443}}'
```

Then to run it:

```
$ bash test.sh
{"errorCode":-30033,"msg":"The HTTP or HTTPS port is out of range."}
```

BJD1997, 2022-07-18:
> So a better workaround is by using the API. Below is a script that will utilize `curl` and `jq` which you need to install to enable the data retention. If you need a linux CLI where you can get those tools, you can just run this container which has both and will drop you in a shell:
> 
> ```
> docker run -it --rm mbentley/swissarmy
> ```
> 
> The last command that does a `PATCH` will set the retention. You need to adjust the value (it appears to be in days) from 180 to whatever you want and `clientsDataEnable":true` will enable it:
> 
> ```
> #!/bin/bash
> 
> # set variables
> OMADA_URL="https://localhost:443"
> USERNAME="admin"
> PASSWORD="password"
> 
> # get controller id from the API
> CONTROLLER_ID="$(curl -sk "${OMADA_URL}/api/info" | jq -r .result.omadacId)"
> 
> # login, get token, set & use cookies
> TOKEN="$(curl -sk -X POST -c "/tmp/omada-cookies.txt" -b "/tmp/omada-cookies.txt" -H "Content-Type: application/json" "${OMADA_URL}/${CONTROLLER_ID}/api/v2/login" -d '{"username": "'"${USERNAME}"'", "password": "'"${PASSWORD}"'"}' | jq -r .result.token)"
> 
> # set data retention (180 = 6mo)
> curl -sk -X PATCH -b "/tmp/omada-cookies.txt" -H "Content-Type: application/json" -H "Csrf-Token: ${TOKEN}" "${OMADA_URL}/${CONTROLLER_ID}/api/v2/controller/history?token=${TOKEN}" -d '{"retention":180,"clientsDataEnable":true}'
> ```
> 
> When you run this, you should see output like:
> 
> ```
> $ bash test.sh
> {"errorCode":0,"msg":"Success."}
> ```
> 
> And then you should be able to see in the UI that the value has changed.

I've turnede the data rentention back on this way and works perfectly!

mbentley, 2022-07-18:
Awesome, thanks for the follow up.  I hope that I can still convince them to remove the restriction in the API in the forum post.  I understand why they did it since it could leave people with a controller that is unable to start with no easy way to revert the port but it stinks for those who know how to work around it.

mbentley, 2022-07-27:
Looks like a permanent fix will be coming [according to Fae](https://community.tp-link.com/en/business/forum/topic/570870?replyId=1116564):
> Thanks so much for your valuable feedback! I've shared the situation to the R&D team for further investigation and evaluation. It's confirmed that Omada Controller 5.5 (Linux) will add the support of port 443 back, additional note for using port 443 will be added. Cheers!

BJD1997, 2022-07-27:
That's great news! For me this issue is solved :) Thanks for the help!

BJD1997, 2022-08-23:
I've updated my controller to 5.5.6 and port 443 is fixed but now it's bugging me about port 80 usage....

mbentley, 2022-08-23:
Just curious - which ports do you have configured for port 80?  Just `MANAGE_HTTP_PORT`?  If so, I believe that one should be fairly low value to actually be port 80 since I am pretty confident that it only handles a redirect from http to https and nothing else.

BJD1997, 2022-08-23:
Yes have port 80 configured. But this now shows up in the interface as before it wasn't shown. So it fails exactly the same as 443 dit before.

mbentley, 2022-08-23:
Oh geez, I didn't make that connection.  I'll see if they can do the same for port 80 then 🤦

mbentley, 2022-08-23:
Re-opening for now.

mbentley, 2022-10-10:
Port 80 is still a problem.  Bumping back up in the forums thread.

BJD1997, 2022-11-25:
Just updated form 5.6.3 to 5.7.4 and looked at the settings page. The issue with port 80 has been resolved now.
I don't know which release it was that fixed it but it's solved now :D

rickdgray, 2023-06-07:
For anyone landing here with a broken omada install after fussing with port numbers, I was able to fix my install using [this guide](https://www.tp-link.com/us/support/faq/3429/).

mbentley, 2023-06-07:
Oh that’s fantastic. I didn’t know about the override option!

BJD1997, 2023-09-21:
Upgraded to the 5.12 release and tested again. Issue is resolved now.

closed 2023-09-21 by BJD1997