mbentley/docker-omada-controller #221: New Version Available - 5.4.6

dli417, 2022-07-29:
**Is your feature request related to a problem? Please describe.**
The Linux archive formats of 5.4.6 are finally available - see below

**Describe the solution you'd like**
Update Dockerfile, entrypoint, etc as needed (whenever you can 🙂)

**Additional context**
Omada_SDN_Controller_v5.4.6_Linux_x64.tar.gz: `https://static.tp-link.com/upload/software/2022/202207/20220729/Omada_SDN_Controller_v5.4.6_Linux_x64.tar.gz`
Omada_SDN_Controller_V5.4.6_Linux_x64.deb: `https://static.tp-link.com/upload/software/2022/202207/20220729/Omada_SDN_Controller_v5.4.6_Linux_x64.deb`

mbentley, 2022-07-29:
Thanks!  My version checker script noticed the same this morning.  I'll be working on the new version when I am able to...pesky work 😄

dli417, 2022-07-29:
Great to hear - thanks for all the time you put into maintaining this project!

nestyxx, 2022-07-29:
Parfait ! Merci pour le suivi du projet sur docker.

closed 2022-07-29 by mbentley

dsteinkopf, 2022-07-30:
I am using the "latest" image and am not able to start the new 5.4 version. `sed: can't read /etc/java-8-openjdk/security/java.security: No such file or directory` is popping to my eyes. Can this be a problem on my side? Shall I open a new issue?

mbentley, 2022-07-30:
Ah shoot, I know why that is.

mbentley, 2022-07-30:
https://github.com/mbentley/docker-omada-controller/pull/223 fixes this; latest version uses OpenJDK 17 for the first time so  a path changed.

mbentley, 2022-07-30:
Fixed version should now be on Docker Hub.

dsteinkopf, 2022-07-30:
Many many thanks for that quick fix and for everything. Now it's working again.

alehmann76, 2022-07-30:
Hello,
sorry for my maybe stupid question....
i´m not very firm with docker-compose and so im quite happy that i have my omada controller running with your great work here!
Thankx for your time and work you spend here! Respect!

but now i´m would like to update my 5.1.7 to the current 5.4.6.
can you give me some directions how to perform this without loosing any configs?
(some weeks ago i tried to upgrade to the previous version and i lost....)

Thanks in advance for your help....

mbentley, 2022-08-02:
Do a: 

```
docker compose down -t 30
```

Then update the image tag in your compose file then do:

```
docker compose up -d
```

Which will bring the stack back up.  It will re-use whatever persistent data volumes you have configured in your compose file.

dli417, 2022-08-02:
If you are using docker volumes in your docker-compose.yml, and they are created with `docker compose up ...`, then `docker compose down` may delete those volumes (causing data loss). You can create docker volumes using `docker volume create omada-data` for example, then add `external:true` to each docker volume in docker-compose.yml so volumes are not deleted with `docker compose down`:

```yml
volumes:
  omada-data:
    external: true
```

mbentley, 2022-08-02:
`docker compose down` should only do anything with volumes when the `-v` or `--volumes` flag is used but you bring up a great point - I would typically recommend using bind mounts to a known paths on the host over use docker volumes of any kind since you'd have to explicitly delete the files and directories in the filesystem vs an accidental removal if someone were to prune volumes or something like that.

commit 0016056a (2022-07-29): Added 5.4.6; fixes #221
commit 7aecfa82 (2022-07-29): Merge pull request #222 from mbentley/add-5.4.6

Added 5.4.6; fixes #221