mbentley/docker-omada-controller #228: Fail to start omada-controller

azuttre, 2022-08-11:
Hello,

I noticed the controller was down this morning. The docker will not start anymore. I am not sure what changed since yesterday when it was working.

Here are the logs:

08-11-2022 15:27:11.791 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(55): Starting OmadaLinuxMain v5.4.6 using Java 17.0.4 on mbentley-omada-controller1 with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.4.6.jar started by omada in /opt/tplink/EAPController/lib)
08-11-2022 15:27:11.832 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(640): No active profile set, falling back to 1 default profile: "default"
08-11-2022 15:27:39.750 WARN [main] [] o.s.d.c.CustomConversions(261): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
08-11-2022 15:27:48.391 WARN [main] [] o.s.d.c.CustomConversions(261): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
08-11-2022 15:28:03.624 INFO [main] [] c.t.s.l.c.s.d.CacheLogConsumeHandler(61): log mq consume task is start...
08-11-2022 15:28:03.633 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(61): Started OmadaLinuxMain in 54.687 seconds (JVM running for 74.304)
08-11-2022 15:28:06.780 INFO [main] [] c.t.s.e.s.c.c(153): start schedule remove expire device... period = 10
08-11-2022 15:28:06.800 INFO [main] [] c.t.s.e.s.c.c(145): update rateLimiterCache, permitsPerSecond = 5.0
08-11-2022 15:28:07.396 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(97): manager workGroup core thread  num is 16, max thread num is 16
08-11-2022 15:28:07.603 INFO [main] [] c.t.s.o.m.d.d.m.m.c.DeviceMsgConfig(41): setMsgThreadPool thread coreSize is 4, maxSize is 4,queue size is 4500
08-11-2022 15:28:12.668 INFO [main] [] c.t.s.o.m.d.p.t.TransportConfiguration(154): upgradeSendReq workGroup core thread num is 1, max thread num is 1
08-11-2022 15:28:13.966 INFO [main] [] c.t.s.o.m.c.d.m.s.w.s.c.WirelessGroupConfig(37): ssidWorkerGroup thread size is 4, queue size is 1500
08-11-2022 15:28:16.485 ERROR [main] [] c.t.s.o.s.d.a(90): Controller install failed. Failed to start.
08-11-2022 15:28:16.489 ERROR [main] [] c.t.s.f.c.FacadeUtils(68): facadeMsgEnable is not enable, msg: Data abnormal due to install failed. Failed to start.
08-11-2022 15:28:16.489 ERROR [main] [] c.t.s.o.s.t.FailExitTask(19): Failed to start omada controller, going to exit
08-11-2022 15:28:16.526 INFO [Thread-0] [] c.t.s.o.s.s.b(213): Going to stop mongod which pid is 74
08-11-2022 15:28:16.675 INFO [SpringApplicationShutdownHook] [] c.t.s.e.p.b.a(226): going to shutdown proxy executors
08-11-2022 15:28:16.676 INFO [SpringApplicationShutdownHook] [] c.t.s.e.p.a(73): going to shutdown proxy thread pools.

Any idea how to debug?

mbentley, 2022-08-11:
Could you provide the full details requested in the bug issue template, specifically how you're launching the container (docker run command or compose would be helpful) as well as what platform, OS, architecture, anything that can tell me about the environment in which this is running?  Any logs from around when the container may have died or been killed?  The `Controller install failed. Failed to start.` message is something that I have no idea about what it means as it's from the controller software.

Getting the logs from mongodb may also be helpful as if there is an error related to startup, there is a good chance there is a problem in the persistent data.  Those can be found in the `logs` volume or where you have a bind mount set to.  Or worst case, just execute a `docker cp omada-controller:/opt/tplink/EAPController/logs/mongod.log .` to grab the logs from inside the container and save the `mongod.log` to your current folder.

azuttre, 2022-08-11:
Hello Matt,

Thanks! I am running the container on Synology DSM (7.1-42661 Update 4). I did the install via the DSM frontend, not sure if there's a way for me to get the docker run command from there.

**Describe the bug**
I upgraded the container yesterday to mbentley/omada-controller:latest after receiving a notification for an update by the controller. It worked right away, and the controller was running the latest version yesterday all day.
This morning I had a notification from the controller on the iOS app that the controller was down.

On the Synology DSM, I noticed that the container crashed every time at start-up.
I tried setting up a new container from the image, but no luck. Even my old container from last month crashes btw.

**To Reproduce**
Start the container via the DSM docker interface. Wait for about a minute or so until the container crashes.

**How you're launching your container**
Launching from the Synology UI.

**Container Logs**
[mongod.log](https://github.com/mbentley/docker-omada-controller/files/9310170/mongod.log)
[server.log](https://github.com/mbentley/docker-omada-controller/files/9310260/server.log)

mbentley, 2022-08-11:
So I can reproduce this.  What seems to be happening is that you were running an older version, like 5.3, you started 5.4, and then you started 5.3 again.  When you then try to go back to 5.4, it causes the error that you're seeing around `Controller install failed`.  I have no idea how this can be recovered (or if it can).  I'd have to suggest asking in the [community forums](https://community.tp-link.com/en/business/forum/582).

azuttre, 2022-08-11:
Thanks a lot for the insights!
Worst case I would have to delete the content of the omada folders manually, restart from scratch and import my latest cfg file.

mbentley, 2022-08-11:
Sure thing.  This gives me an idea for a new feature in the entrypoint - blocking starting the app if you try to run an older version of the image.  I'd just need to add a way to make the container aware of the version of the controller and to be able to write the version to a file.

azuttre, 2022-08-11:
It was an easy mistake to make. I had kept the old container as "auto-restart" by mistake. That container wasn't running yesterday, but it must have started automatically this morning.

Thanks for your help. I posted on the [community page](https://community.tp-link.com/en/business/forum/topic/577334?page=1), let's see if there's a quick fix before I do a full clean-up.

mbentley, 2022-08-11:
I'm adding https://github.com/mbentley/docker-omada-controller/pull/229 which makes sure that the version being run is never older to prevent this scenario.

closed 2022-10-10 by mbentley

guitarmanusa, 2023-02-06:
> It was an easy mistake to make. I had kept the old container as "auto-restart" by mistake. That container wasn't running yesterday, but it must have started automatically this morning.
>
> Thanks for [...]

azuttre, how did you completely purge the docker container?  Even after running docker rmi and docker volume rm, whenever I pull down a "fresh" image via docker-compose it's still not starting.