mbentley/docker-omada-controller #231: Better versioning scheme for Docker images?

ankushg, 2022-08-12:
**Is your feature request related to a problem? Please describe.**
Currently, the docker image is republished under the same tag when new patch releases come out from TP Link.

E.g., the image/tag `mbentley/omada-controller:5.4` has been reused as 5.4
- you build a version of your addon against the image `mbentley/omada-controller:5.4`
- TPLink releases a new version of the Omada Controller at `5.4.7`
- a new Docker image is pushed to the **already existing** `mbentley/omada-controller:5.4` tag, resulting in a new hash, but the same tag name

This is problematic for workflows that use Docker container tag names as an indicator of updates

**Describe the solution you'd like**
Easiest option would probably be to keep your existing `mbentley/omada-controller:Major.Minor` tag scheme, but also create a `mbentley/omada-controller:Major.Minor.Patch` tag

`mbentley/omada-controller:Major.Minor` can point to the latest patch release under the same Minor version.

I'm sure there are other options too though!

**Additional context**
This is of particular interest to this project which converts your docker image into an addon for Home Assistant

mbentley, 2022-08-12:
In the past, I have considered doing `major.minor.patch` tags.  The main issue or challenge that I have with that today is that my current "policy" (using that term loosely as it isn't published) is that any image tags that I have available on Docker Hub for an image I publish will be maintained and kept up to date.  As a part of that policy, I am scanning the images daily for vulnerabilities using Trivy and forcing re-builds of the images when OS level vulnerabilities are found.

Applying that policy to this repository, I am currently building 33 unique image tags across three architectures for this project.  That includes only the latest major.minor version tags as I stop supporting and providing a previous version of an image when a new version of the software is available.  If I were to tag with `major.minor.patch` as well, I wouldn't be able to continue to provide security patching of older patch version releases (e.g. - if we have `5.4.6` and `5.4.7`, I wouldn't be able to make sure that the `5.4.6` tag is being updated regularly just because it would increase the number of builds that I would need to do in order to support every release.

So that leads to the question - if I were to tag the versions but only regularly update the `major.minor.patch`, would there be a significant benefit that would make it worth doing?  Due to the complexity of supporting so many images, I would want to understand if supporting every single version that TP-Link releases would be worth the effort as I wouldn't want to deviate from the policy of providing patched images that can be trusted.  I ask from a standpoint of understanding what the goal of this downstream project is.

Based on how I am tagging the images for the controller today, there shouldn't be any expectation that an image tag is an indicator of a newly released version.  Due to me patching the images regularly, what would be an indicator of a change of the images I publish would be the digest of the image has changed and that is something that can be counted on.

For example, here are the last two `5.4` tags:

```
$ docker images --filter reference=mbentley/omada-controller --digests
REPOSITORY                  TAG       DIGEST                                                                    IMAGE ID       CREATED        SIZE
mbentley/omada-controller   5.4       sha256:902dcd980a37463a8a37a8ea41ead43abc8f5dc49dfa4c1aa3d997fd3410221d   88e428f52c53   15 hours ago   574MB
mbentley/omada-controller   <none>    sha256:650f53a315c173b55d9c1ca75cebef2caed47b6cd6855d98a1a57013385009eb   00757aa0f8be   17 hours ago   574MB
```

These two images have different digests because their contents are different.  That's an indicator that something has changed.  One could also check the manifests from Docker Hub to look for changes.  Just as an example, if I were to get the manifests for the `5.4` tag at two different times, I could save and compare them over time:

```
docker manifest inspect -v mbentley/omada-controller:5.4 > manifest_5.4_v1.txt
docker manifest inspect -v mbentley/omada-controller:5.4 > manifest_5.4_v2.txt
```

In this case, I re-ran my image builds in between pulling the image manifests.  If I had either 1.) rebuilt the images due to vulnerabilities or 2.) rebuilt the images due to a change in version of the controller software, the manifest would have changed.  Because I am making use of caching, there were no changes to the image digests.

```
$ md5sum manifest_5.4_v1.txt manifest_5.4_v2.txt
c868b6b6bd671823b47adf9982409137 manifest_5.4_v1.txt
c868b6b6bd671823b47adf9982409137 manifest_5.4_v2.txt
```

So what exactly are your goals when it comes to extending this image/building from it?  Would you also be regularly rebuilding your images to ensure that you are not shipping vulnerabilities or are you just interested in building when a new version comes out?  Is there an explicit need to make changes to your image due to a new version of the software or what's the expectation?

adhawkins, 2022-08-12:
I think it's fairly common practice for (say) a 5.4 tag to automatically change from 5.4.1 to 5.4.2. Some projects to maintain (for example) 5.4.1 and 5.4.2 tags, but as @mbentley says, that implies that these 'older' version tags are still supported with new fixes and the like.

Personally, I think the scheme is just fine. I use [diun](https://github.com/crazy-max/diun) to monitor my containers, and have it notify me when new versions are available. I can then take the decision whether to update them or not.

mbentley, 2022-08-12:
Something to note is that yesterday I added a feature where the entrypoint will validate that the version of the controller in the image isn't an older version than was previously ran.  This is accomplished by having the image write the version of the controller to a file in the image.  This could be read pretty easily:

```
$ docker run --rm --entrypoint cat mbentley/omada-controller:5.4 /opt/tplink/EAPController/IMAGE_OMADA_VER.txt
5.4.6
```

Of course, that does rely on having to pull and run the image but it is an option if knowing the version is required.

ankushg, 2022-08-12:
Thanks for all this context on your versioning scheme! It’s great to know
that even older versions have security issues patched!

The approach that we ended up using was just checking the last modified
time of the image instead of relying on the tag.

If you were feeling up to it, it might be interesting to have a tagging
scheme of Major.Minor.DateLastUpdated if versions like 5.4.20220812 aren’t
too ugly haha

dli417, 2022-08-22:
Another approach could be running a service such as Watchtower or Diun to alert/automatically update docker images if a tag has been updated. I have Diun set up to send me a Slack notification whenever my current Omada Controller docker tag gets updated, or a new tag is released.

mbentley, 2022-08-23:
> Thanks for all this context on your versioning scheme! It’s great to know that even older versions have security issues patched! The approach that we ended up using was just checking the last [...]

Yeah, a tagging scheme like that would be a really nasty to deal with, especially given the number of tags build built and in the current build format, there isn't a great, predictable way for me to know when an image is ending up being rebuilt or not due to using buildkit + caching mechanisms.  If you feel like you have a workable solution in terms of figuring out the latest updates, I'll go ahead and close this.

Bockiii, 2022-09-21:
I agree with adhawkins: It's a normal use case that semver versioned containers update.

If you run a "debian:11" container, you dont want 11.0.0 right? You would want the current patch level of 11. Same goes for "debian:11.3". You wouldn't want to run 11.3.0, you would want the latest patch level of 11.3.

So I would keep it as you have it now. tag 5.4 will change if 5.4.X is released, just as tag 5 will change if the latest 5.x.x is released.

mbentley, 2022-09-21:
Your comment brings up another question - is it helpful for anyone to have just a major version tag like `5`?  I don't recall there being breaking changes in minor versions but I could be wrong about that.

Bockiii, 2022-09-21:
I guess it would since it will become very helpful if they decide to release a v6 which breaks everyone's setups again like with 4->5.

I'm running "latest" with your container, but I'm always aware that that could break at any point :) I would set it to :5 if you had it because then other people can to the 6.0 beta testing ;)

mbentley, 2022-10-10:
Closing for now unless there is action required.

closed 2022-10-10 by mbentley