mbentley/docker-omada-controller #246: [Bug]: SMTP FROM address not set

jgardner-qha, 2022-09-19:
### Describe the Bug

In the controller, after setting the SMTP settings with Amazon SES credentials, sending a test message appears to be successful, but does not actually send. Amazon SES reports an "Invalid FROM" address that matches the SMTP username, rather than the sender's address.

### Expected Behavior

When using the SMTP settings in the Omada controller, emails should be sent FROM the specified "Sender Address" rather than from the SMTP Username.

### Steps to Reproduce

1. Complete the SMTP settings screen with Amazon SES credentials.
2. Send a test e-mail. "Success" message appears on the screen, but mail is not sent.
3. Error 501 Invalid MAIL FROM address provided appears in container logs (shown below).
[image]
[image]

### How You're Launching the Container

```plain
#docker-compose.yml

version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.5
    restart: unless-stopped
    network_mode: host
    environment:
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PGID=508
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - PUID=508
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=Etc/UTC
    volumes:
      - omada-data:/opt/tplink/EAPController/data
      - omada-logs:/opt/tplink/EAPController/logs
      - /home/omada/.acme.sh/omada.qualitytechnd.org/fullchain.cer:/cert/tls.crt
      - /home/omada/.acme.sh/omada.qualitytechnd.org/omada.qualitytechnd.org.key:/cert/tls.key

volumes:
  omada-data:
  omada-logs:
```

### Container Logs

```plain
omada-controller  | DEBUG: setDebug: JavaMail version 1.6.2
omada-controller  | DEBUG: getProvider() returning javax.mail.Provider[TRANSPORT,smtps,com.sun.mail.smtp.SMTPSSLTransport,Oracle]
omada-controller  | DEBUG SMTP: useEhlo true, useAuth true
omada-controller  | DEBUG SMTP: trying to connect to host "email-smtp.us-east-1.amazonaws.com", port 465, isSSL true
omada-controller  | 220 email-smtp.amazonaws.com ESMTP SimpleEmailService-d-S9HCEPLYJ 6jSQihYSl28ImmtQXmOx
omada-controller  | DEBUG SMTP: connected to host "email-smtp.us-east-1.amazonaws.com", port: 465
omada-controller  | EHLO vomada01.ndhcri.local
omada-controller  | 250-email-smtp.amazonaws.com
omada-controller  | 250-8BITMIME
omada-controller  | 250-AUTH PLAIN LOGIN
omada-controller  | 250 Ok
omada-controller  | DEBUG SMTP: Found extension "8BITMIME", arg ""
omada-controller  | DEBUG SMTP: Found extension "AUTH", arg "PLAIN LOGIN"
omada-controller  | DEBUG SMTP: Found extension "Ok", arg ""
omada-controller  | DEBUG SMTP: protocolConnect login, host=email-smtp.us-east-1.amazonaws.com, user=AXXXXXXXXXXXXXXES, password=<non-null>
[... 77 lines cut ...]
omada-controller  |                             "curState":     1,
omada-controller  |                             "eventCauseStr":        "INNER_ERR:[_do_rebuild_state_v2():3009] @ rebuild failed for long times, goto reset"
omada-controller  |                     }, {
omada-controller  |                             "timestamp":    "2022-09-16 22:56:10",
omada-controller  |                             "prevState":    4,
omada-controller  |                             "curState":     5,
omada-controller  |                             "eventCauseStr":        "INFORM_TMO"
omada-controller  |                     }]
omada-controller  |     }
omada-controller  | },additionalProperties={}]),addressDTO=<null>] failed.
omada-controller  | java.lang.NullPointerException: Cannot invoke "com.tplink.smb.omada.device.gateway.core.datatrack.a.b.a(java.lang.Long)" because "datatrackImage" is null
omada-controller  |     at com.tplink.smb.omada.device.gateway.core.datatrack.h.a(SourceFile:63) ~[device-gateway-core-5.5.6.jar:5.5.6]
omada-controller  |     at com.tplink.smb.omada.device.gateway.core.datatrack.h.a(SourceFile:51) ~[device-gateway-core-5.5.6.jar:5.5.6]
omada-controller  |     at com.tplink.smb.omada.device.gateway.core.datatrack.a.a(SourceFile:36) ~[device-gateway-core-5.5.6.jar:5.5.6]
omada-controller  |     at com.tplink.smb.omada.device.gateway.core.datatrack.b.a(SourceFile:60) ~[device-gateway-core-5.5.6.jar:5.5.6]
omada-controller  |     at com.tplink.smb.omada.device.gateway.core.datatrack.d.c(SourceFile:61) ~[device-gateway-core-5.5.6.jar:5.5.6]
omada-controller  |     at org.apache.skywalking.apm.toolkit.trace.RunnableWrapper.run(RunnableWrapper.java:34) ~[apm-toolkit-trace-8.1.0.jar:8.1.0]
omada-controller  |     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[?:?]
omada-controller  |     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[?:?]
omada-controller  |     at java.lang.Thread.run(Thread.java:833) [?:?]
```

### Additional Context

This is likely not be a problem with your Docker container, but I figured I'd post it here in case you have any suggestions. I'll also post this on the forum. Are there any environment variables that can control SMTP settings for the Omada Controller?

mbentley, 2022-09-19:
This seems like something that would be ideal to post to the [TP-Link Community forums](https://community.tp-link.com/en/business/forum/582).  The 501 message appears to be coming from Amazon SES.  In a test, I see the format it is using when sending is `MAIL FROM:<noreply@mbentley.net>` from the logs.  It could be that SES doesn't like that format or maybe it is the domain that is being used as the sender.  I'm not well versed in SES but I imagine that it doesn't just let you send mail as anyone.

What's the format for your email address?  Is it `name@domain.com`?  I would expect so since the form validation seems to only let me put something like that in but figured I would check.  If the format that is being send by the controller isn't valid for SES, then it is a fix that needs to happen in the controller software.

jgardner-qha, 2022-09-19:
I will be posting to the forums also.

You are correct, the 501 is coming from SES. I censored out the actual usernames in the logs, because they are a hash, not an e-mail address. The problem is that the controller is using the SMTP username as the FROM address instead of the Sender address. The SES SMTP username is _not_ an e-mail address. SES requires "validated" addresses for the FROM address, which of course needs to be an email address.

It's as if there are two separate problems... probably both with the controller, not the docker container. One is that it does not seem to be honoring the specified From address, and other that it reports "Successful" sending a test message when there is a clear failure in the logs.

Do you know if any of the SMTP settings can be specified by environment variables rather than within the controller settings?

Big thanks for the quick reply and for the wonderful docker image!

mbentley, 2022-10-10:
Sorry, I thought I replied to this.  No, you can't set them by environment variables as far as I know.  Hopefully the forums have been helpful.

closed 2022-10-10 by mbentley