mbentley/docker-omada-controller #247: [Bug]: Drops Wan every 48 hours ER-605 latest omada controller running on pi

jonas740, 2022-09-27:
### Describe the Bug

I have TP-Link ER605 V1

Running version 1.2.1

Recently it drops wan every 48 hours.

3 things that was changed during last month

1. Updated to version 1.2.1 on ER605
2. Updated to latest mbentley/omada controller a few weeks ago
3. my whole town had a power outage

### Expected Behavior

I do not expect my wan to drop because it has never done so and i do not know if the controller is the issue. Controller has been running fine on my raspberry pi for overe a year. 

Today i saw that 16 hours ago a new docker image of the controller was pushed so i have recently updated to that. 

### Steps to Reproduce

wait 48 hours

### How You're Launching the Container

```plain
services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:latest
     healthcheck:
      disable: true
     environment:
      - TZ=Etc/UTC
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
     network_mode: host

     volumes:
      - omada-data:/opt/tplink/EAPController/data
      - omada-work:/opt/tplink/EAPController/work
      - omada-logs:/opt/tplink/EAPController/logs
     restart: unless-stopped
volumes:
  omada-data:
  omada-work:
  omada-logs:
```

### Container Logs

```plain
09-27-2022 23:08:20.002 INFO [pool-4-thread-1] [] c.t.s.e.s.c.c(): [STAT] {"statName":"OmadaDeviceStat","deviceType":"gateway","ecspVer":"2.2.0","deviceNum":"1"},
09-27-2022 23:08:20.002 INFO [pool-4-thread-1] [] c.t.s.e.s.c.c(): [STAT] {"statName":"OmadaDeviceStat","deviceType":"ap","ecspVer":"2.3.0","deviceNum":"1"},
09-27-2022 23:14:41.800 INFO [Thread-0] [] c.t.s.o.s.s.b(): Going to stop mongod which pid is 64,
09-27-2022 23:14:41.829 INFO [SpringApplicationShutdownHook] [] c.t.s.e.s.EcspTransporterStarter(): going to shutdown ecsp servers & executors,
09-27-2022 23:14:41.830 INFO [SpringApplicationShutdownHook] [] c.t.s.e.t.a.t.AbstractServer(): Close NettyUdpServer, bind /0.0.0.0:29810, export localhost/127.0.0.1:29810,
09-27-2022 23:14:41.832 INFO [SpringApplicationShutdownHook] [] c.t.s.e.t.a.t.AbstractServer(): Close NettyTcpServer, bind /0.0.0.0:29811, export localhost/127.0.0.1:29811,
09-27-2022 23:14:41.837 INFO [SpringApplicationShutdownHook] [] c.t.s.e.t.a.t.AbstractServer(): Close NettyTcpServer, bind /0.0.0.0:29812, export localhost/127.0.0.1:29812,
09-27-2022 23:14:41.839 INFO [SpringApplicationShutdownHook] [] c.t.s.e.t.a.t.AbstractServer(): Close NettyTcpServer, bind /0.0.0.0:29813, export localhost/127.0.0.1:29813,
09-27-2022 23:14:41.841 INFO [SpringApplicationShutdownHook] [] c.t.s.e.t.a.t.AbstractServer(): Close NettyTcpServer, bind /0.0.0.0:29814, export localhost/127.0.0.1:29814,
09-27-2022 23:14:41.847 INFO [manage-work-group-12] [] c.t.s.o.m.d.p.t.c(): Device D8-07-B6-A6-E8-66 OmadacId(64cde7237e224b81b1a287f7de0034f4) changed to status CONNECTED_ERROR, which don't need to handle.,
09-27-2022 23:14:41.847 INFO [manage-work-group-2] [] c.t.s.o.m.d.p.t.c(): Device C0-06-C3-BF-67-40 OmadacId(64cde7237e224b81b1a287f7de0034f4) changed to status CONNECTED_ERROR, which don't need to handle.,
[... 69 lines cut ...]
09-27-2022 23:15:29.467 INFO [Thread-5] [] c.t.e.c.c.CloudClient(): Close connection to SEF server.,
09-27-2022 23:15:29.472 INFO [Thread-8] [] c.t.e.c.c.n(): ssl recv thread is stopped.,
09-27-2022 23:15:29.490 INFO [Thread-5] [] c.t.e.c.c.CloudClient(): Connect service server automatically, ConnectionType is PERSISTENT_CONNECTION.,
09-27-2022 23:15:29.491 INFO [Thread-5] [] c.t.e.c.c.q(): set connection status: CONNECTING,
09-27-2022 23:15:29.897 INFO [Thread-5] [] c.t.e.c.c.q(): set connection status: CONNECTED,
09-27-2022 23:15:29.897 INFO [Thread-5] [] c.t.e.c.c.q(): The result of connection is true.,
09-27-2022 23:15:33.483 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): MANAGED_BY_OWN Device C0-06-C3-BF-67-40 on omadac 64cde7237e224b81b1a287f7de0034f4 is discoveried.,
09-27-2022 23:15:36.256 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): MANAGED_BY_OWN Device D8-07-B6-A6-E8-66 on omadac 64cde7237e224b81b1a287f7de0034f4 is discoveried.,
09-27-2022 23:15:40.905 INFO [adopt-work-group-0] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(C0-06-C3-BF-67-40),
09-27-2022 23:15:41.022 INFO [adopt-work-group-1] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(D8-07-B6-A6-E8-66),
09-27-2022 23:15:41.208 INFO [manage-work-group-1] [] c.t.s.o.m.d.p.t.c(): Device D8-07-B6-A6-E8-66 OmadacId(64cde7237e224b81b1a287f7de0034f4) changed to status CONNECTED, which don't need to handle.,
09-27-2022 23:15:41.233 INFO [manage-work-group-2] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(D8-07-B6-A6-E8-66),
09-27-2022 23:15:41.352 INFO [manage-work-group-3] [] c.t.s.o.m.d.d.m.i.d(): first inform send same version config to OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(D8-07-B6-A6-E8-66),
09-27-2022 23:15:41.517 INFO [manage-work-group-4] [] c.t.s.o.m.d.d.m.d.a.R(): syncConfigurationForSameVersion to OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(D8-07-B6-A6-E8-66), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={})),
09-27-2022 23:15:44.199 INFO [manage-work-group-5] [] c.t.s.o.m.d.p.t.c(): Device C0-06-C3-BF-67-40 OmadacId(64cde7237e224b81b1a287f7de0034f4) changed to status CONNECTED, which don't need to handle.,
09-27-2022 23:15:46.525 INFO [manage-work-group-6] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(C0-06-C3-BF-67-40),
09-27-2022 23:15:46.597 INFO [manage-work-group-7] [] c.t.s.o.m.d.d.m.i.e(): first inform no need send full config to OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(C0-06-C3-BF-67-40),
09-27-2022 23:15:47.398 INFO [manage-work-group-8] [] c.t.s.o.m.d.d.m.d.b.z(): syncConfigurationForSameVersion to OmadacId(64cde7237e224b81b1a287f7de0034f4) DeviceMac(C0-06-C3-BF-67-40), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={})),
09-27-2022 23:16:00.169 WARN [https-jsse-nio-8043-exec-9] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.,
09-27-2022 23:16:00.188 WARN [https-jsse-nio-8043-exec-9] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
```

### Additional Context

I just pulled the latest image and hoping that will solve the issue.

mbentley, 2022-09-30:
To the best of my knowledge, a controller outage should not impact the router's functionality.  The only instance where outages were occurring was when it was bug related on the controller software where it was forcing a re-provisioning due to what I believe ended up being a time zone issue or something of that nature, but it was on older controllers and I am only aware of the issue from access points, or at least that is where people reported the issue as it was dropping clients to re-provision each WAP.

Any idea what version of the controller you were running previously before you updated?

jonas740, 2022-09-30:
I ran the previous version of what is available now.

/Jonas

mbentley, 2022-10-04:
Does the controller container end up getting killed/restarted when the outage occurs or is there any other behavior that can be seen that happens around the same time?  It does look like from the logs that the container is being stopped with the typical shutdown logs and then restarts back up.  Are there any processes that would perform any actions on your containers when updates are available or anything like that?  What about automatic upgrades of packages?

jinkazph, 2022-10-13:
I have same issue, so irevert back my er605 to 1.1.1 Build202110723

jonas740, 2022-10-13:
> I have same issue, so irevert back my er605 to 1.1.1 Build202110723 

Well I went and bought er605 hardware version 2 and are using 2.0.1 firmware at the moment.

Do you think reverting to 2.0.0 would do the trick since I guess they released firmware update for all hardware versions?

jinkazph, 2022-10-13:
For reference..
https://community.tp-link.com/en/business/forum/topic/581548

jonas740, 2022-10-16:
> For reference..
> https://community.tp-link.com/en/business/forum/topic/581548

Ye I read that. Might not be the exact same issue but now I am running the V2 of the ER605 with firmware 2.0.1 and it happened again tonight.

Rebooted the router and it worked again. Then it happened again so I rebooted again.

Now I am downgrading to first available firmware version of the V2 of ER605.

I also wrote about the issue to tp-link since this seems to be a firmware issue.

So strange that it happens at almost the same time every Sunday to Monday around midnight.

I have to wait a week now with the downgraded version to see if it happens again.

For me WAN drops out as stated above.

jonas740, 2022-10-17:
> For reference..
> https://community.tp-link.com/en/business/forum/topic/581548

Below is the exact message I can read out from the logs in the omada interface. 

"Failed to obtain the ip address for wan because no response from the server"

But if i power cycle the ER605 it comes back online.

Last night I rebooted through omada controller and got wan connection again but minutes later same thing happened so I did another reboot and connection was back. I then downgraded to firmware 2.0.0 and since then it did not reproduce. [image]
[image]
[image]
[image]

Hoping you can see the screenshots I posted.

jonas740, 2022-10-24:
Just to be sure. But updating the controller with this docker is as simple as stopping the docker container and removing it and then pulling the latest right?

I am not a pro at docker just yet hence I am asking.

There was an update today I saw.

So i did update but I am not sure if the image has the latest controller software?

mbentley, 2022-10-24:
I would suggest that you don't use the latest tag but use one of the specific version tags (e.g. - 5.6) just to be safe.  While they currently have the same contents, it's just a better practice to not use "latest".  But anyway, the main thing you want to do is to stop and remove the container and then re-create the container with the new image.

Make sure that you give the container plenty of time to stop!  MongoDB, especially the version that is required for a Raspberry Pi, is notorious for losing data if the container is forcefully killed, which happens when the process doesn't stop within the default allotted stop window of 10 seconds so you'll want to set that timeout, something like "docker stop -t 60 <container-name-or-id>" (e.g. -`docker stop -t 60 omada-controller`).  Then you can just remove the container and start up a new one.

mbentley, 2022-11-26:
Are you still having stability issues?  Anything else that I can do to help troubleshoot to see if it is related to the software being containerized?

jonas740, 2022-11-27:
> Are you still having stability issues?  Anything else that I can do to help troubleshoot to see if it is related to the software being containerized?

It actually seems to have been resolved, TP-Link themselves had remote access and did something. I am not sure what they did but it seems to have fixed the issue.

So this thread can be marked as resolved now I guess.

closed 2022-11-27 by mbentley