mbentley/docker-omada-controller #248: Failed to obtain the ip address for wan because no response from the server

jonas740, 2022-10-03:
### Describe the Bug

I have been getting this error "Failed to obtain the ip address for wan because no response from the server" every Sunday/Monday for the last 3-4 weeks now. 

I had the same problem with V1 of the ER605 as well. I switched to V2 this weekend just to see if that would make any difference but at 00:39-01:47AM Monday it happened again. 

I am running Omada controller on a Linux server https://hub.docker.com/r/mbentley/omada-controller wich has been working fine. 

Running the latest image of it BTW. 

When I made the switch from Er605 v1 to V2 I just removed v1 in omada and adopted the v2. 

Should I have made a total reset of the omada install and do it all over again? 

Although I don't think that this is the cause for it but it is very frustrating. 

Last Monday I did a power cycle of my er605 v1 when this occurred and within 5 minutes I had wan connection again. 

I have my isp troubleshooting if something is incorrect at their end aswell since it always happens Sunday/Monday during night. 

### Expected Behavior

This should not happen

### Steps to Reproduce

It happens sunday/Monday every week. 

### How You're Launching the Container

```plain
Docker compose file as found on your github.

It has always worked before. But since the last two updates this has started.
```

### Container Logs

```plain
I will post them when I get home
```

mbentley, 2022-10-04:
Hmm, hard to say.  I found a [reddit thread from a year ago](https://www.reddit.com/r/TPLink_Omada/comments/oxczxq/failed_to_obtain_the_ip_address_for_wan/) and this guy seemed to solve it by powering off his modem for a period of time.  Otherwise, you'll probably want to post on the [TP-Link forums](https://github.com/mbentley/docker-omada-controller#:~:text=TP%2DLink%20community%20forums) as it isn't likely this is an issue with the Docker image packaging.

I see [this issue on the forums](https://community.tp-link.com/en/business/forum/topic/513724) but it sounds like this should have been resolved in a 1.2 version of the firmware which I wouldn't expect would be an issue with the v2 firmware you're using unless there is a regression.

jonas740, 2022-10-04:
I found this also.

https://community.tp-link.com/en/business/forum/topic/513724

But that is pretty far fetched.

No I have not changed anything I have only done the three last updates of the image now just as I always done.

But that's about when it started.

I should add that I during the weekend bought the ER605 hardware version 2 running firmware 2.0.1

Before I had ER605 hardware version 1 and firmware 1.2.1 wich is the latest.

Had to try something but same thing with the v2 of the router.

And I should add that one of the times I power cycled the router and then it managed to obtain wan as it should.

But it does not seem to be able to do so automatically.

I don't have any scheduled reboots or anything like that.

I have yet to hear anything from my ISP and they are also investigating this but at their end ofc.

mbentley, 2022-10-10:
That's hard to say.  Doesn't sound like an issue with the controller containerization through so I am going to close this.  Let me know if I can be of further assistance.

mbentley, 2022-10-10:
Will take a look and see if it may be anything container related.  If you have approximate timelines to line up from when you noted outages or issues to match up timelines with the logs, that would be most helpful.

jonas740, 2022-10-10:
This is from the Omada Controller Events interface. Just copied a few lines. My controller was offline for a few minutes this morning and i was not the one logging in as it says there. Also the issues with EAP245

`<html>
<body>

Master Administrator admin logged in to the controller from 192.168.1.55. | Oct 10, 2022 18:25:50
-- | --
Master Cloud Administrator j**********dt@gmail.com logged in to the controller from Cloud Access. | Oct 10, 2022 07:00:18
TP-Link ER605 V2 was connected. | Oct 10, 2022 06:38:54
TP-Link EAP245 was connected. | Oct 10, 2022 06:38:49
TP-Link EAP245 was readopted automatically. | Oct 10, 2022 06:38:48
TP-Link ER605 V2 was readopted automatically. | Oct 10, 2022 06:38:48
Master Cloud Administrator j**********dt@gmail.com logged in to the controller from Cloud Access. | Oct 10, 2022 06:35:20
Master Administrator admin logged in to the controller from 192.168.1.55. | Oct 08, 2022 22:43:38
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon24" on channel 11 because WPA Authentication times out/failed.(1 time in a minute) | Oct 08, 2022 02:43:44
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(2 times in a minute) | Oct 08, 2022 02:42:44
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(1 time in a minute) | Oct 08, 2022 02:40:14
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(1 time in a minute) | Oct 08, 2022 02:35:14
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(1 time in a minute) | Oct 08, 2022 02:33:44
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(1 time in a minute) | Oct 08, 2022 02:28:14
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(2 times in a minute) | Oct 08, 2022 02:21:44
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(2 times in a minute) | Oct 08, 2022 02:18:14
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(1 time in a minute) | Oct 08, 2022 02:13:14
[Failed]00-04-4B-4A-FC-FC failed to connected to TP-Link EAP245 with SSID "Agnoon50" on channel 36 because WPA Authentication times out/failed.(2 times in a minute)

</body>
</html>`

mbentley, 2022-10-12:
Nothing of any helpful relevance there from what I can see.  I'm not likely to be able to find whatever issue it is you're having based on the information provided so the forums or TP-Link support will still be your best bet.

closed 2022-10-12 by mbentley