mbentley/docker-omada-controller #249: [Bug]: Controller "Id" changes after creating a new container using a stored volume from a previous container.

andyslucky, 2022-10-09:
### Describe the Bug

I created a container using docker run a few months ago to test out the controller, and I have been very pleased. Recently I needed to stop that container to do some system maintenance. Using portainer (lol the first mistake), I tried update the image used and restart the container, but some problems were encountered and the container was lost. I make regular backups of my container file systems, so I copied the docker compose script, created a persistent volume, and copied the contents of /opt/tplink/EAPController/data from my backup into the _data dir of the persistent volume and changed the owner and group of all the files to 508 (the uid and gid I am using). After starting the controller up, I noticed the controller "id" (not sure if that is correct) had changed, and I was greeted with a "Let's get started" screen. The state, devices, and config were not preserved. Have I missed some crucial step in copying the state and configuration from my old installation? At the moment I cannot see the devices in the setup page; I assume this is because they are still associated with another controller?

### Expected Behavior

One might expect, after copying the contents of a previous container's file system to a persistent volume with the appropriate permissions that they should be able to start the controller and all state, connected devices, configuration, etc. should be preserved. Essentially and I should be able to use the controller as though nothing has changed.

### Steps to Reproduce

1. Create a container using the latest version of the image / or use an existing container you may have
2. Export that container's file system using `docker export --output="omada-controller.tar"`
3. Create a new persistent volume, and copy the contents from /opt/tplink/EAPController/data inside the .tar file into the new persistent volume's `_data` dir
4. Using docker-compose and the attached docker compose file, create a new stack, replacing the volume `omada-controller-data` with the name of the persistent volume created in step 3.
5. Monitor the logs of this new stack's "omadacontroller" container to confirm the id of the controller is different and the state,  associated devices, and admins from the original container were not preserved in the new container.

### How You're Launching the Container

```plain
version: "3.7"
services:  
  omada:
    container_name: omadacontroller
    hostname: omadacontroller
    image: mbentley/omada-controller:latest
    restart: unless-stopped
    network_mode: host
    environment:
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043

      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - PUID=508
      - PGID=508
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=America/New_York
    volumes:
     - omada-controller-data:/opt/tplink/EAPController/data

volumes:
  omada-controller-data:
    external: true
    name: omada-controller-data
```

### Container Logs

```plain
INFO: Validating user/group (omada:omada) exists with correct UID/GID (508:508)
INFO: Group (omada) doesn't exist; creating
INFO: User (omada) doesn't exist; creating
INFO: Time zone set to 'America/New_York'
INFO: Setting 'portal.https.port' to 8043 in omada.properties
INFO: Version check passed; image version (5.5.6) >= the last version ran (5.5.6); writing image version to last ran file...
INFO: Starting Omada Controller as user omada
10-09-2022 12:08:17.543 INFO [log4j-thread] [] c.t.s.o.c.o.a.b(): success to load configuration omada.properties
10-09-2022 12:08:17.559 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): going to start local mongod.
10-09-2022 12:08:18.568 INFO [main] [] c.t.s.o.s.s.b(): mongodb process id is 79
10-09-2022 12:08:18.571 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Mongo DB server started
10-09-2022 12:08:18.572 INFO [main] [] c.t.s.o.s.s.b(): Mongo DB server started

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v2.6.6)
[... 27 lines cut ...]
10-09-2022 12:08:31.515 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyUdpServer bind /0.0.0.0:29810, export localhost/127.0.0.1:29810
10-09-2022 12:08:31.553 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29812 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
10-09-2022 12:08:31.557 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29812, export localhost/127.0.0.1:29812
10-09-2022 12:08:31.558 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29811 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
10-09-2022 12:08:31.558 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
10-09-2022 12:08:31.559 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
10-09-2022 12:08:31.560 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
10-09-2022 12:08:31.561 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
10-09-2022 12:08:31.562 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
10-09-2022 12:08:31.562 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Device connector server started.
10-09-2022 12:08:31.562 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Omada Controller started
10-09-2022 12:08:31.562 INFO [main] [] c.t.s.o.s.t.SuccessExitTask(): Omada Controller started
10-09-2022 12:08:34.206 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): MANAGE_BY_OTHERS Device 10-27-F5-97-CC-B2 on omadac bb666f246316ed06cafecddfb84ef65f is discoveried.
10-09-2022 12:08:37.047 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
10-09-2022 12:08:37.047 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
10-09-2022 12:08:38.855 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): MANAGE_BY_OTHERS Device 5C-A6-E6-F5-10-18 on omadac bb666f246316ed06cafecddfb84ef65f is discoveried.
10-09-2022 12:08:47.067 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
10-09-2022 12:08:47.069 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
10-09-2022 12:08:47.163 WARN [https-jsse-nio-8043-exec-5] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
10-09-2022 12:08:47.173 WARN [https-jsse-nio-8043-exec-5] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
10-09-2022 12:08:57.093 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
10-09-2022 12:08:57.094 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
10-09-2022 12:09:07.117 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
10-09-2022 12:09:07.118 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.c(): Invalid wireless ap discovery without db record or parent ap on B0-A7-B9-14-E8-A6 with omadac bb666f246316ed06cafecddfb84ef65f.
```

mbentley, 2022-10-10:
Doing a `docker export...` isn't the way to create a backup; it doesn't create backups of any volumes, [only the copy on write filesystem](https://docs.docker.com/engine/reference/commandline/export/#:~:text=The%20docker%20export%20command%20does,the%20contents%20of%20the%20volume.) (essentially you're backing up the disposable parts of the container).  So doing that exports the _container's_ filesystem, not the persistent data.  The proper way to backup the persistent data when using a docker named volume [can be found in this issue comment](https://github.com/mbentley/docker-omada-controller/issues/198#issuecomment-1186160230).  You should slightly modify the command is that `work` volume is no longer necessary.  For completeness, here is an updated example:

Stop existing container:

```
docker stop -t 60 omada-controller
```

Run a temporary container to tar up the persistent data from docker volumes (this assumes your volumes are named `omada-data` and `omada-logs`; modify as needed):

```
docker run -it --rm \
  -w /opt/tplink/EAPController \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-logs:/opt/tplink/EAPController/logs \
  -v "${HOME}:/tmp/transfer" \
  alpine \
  tar cvf /tmp/transfer/omada_data.tar data logs
```

Verify the tar exists on your source system in your home directory:

```
$ ls -l "${HOME}/omada_data.tar"
-rw-r--r--  1 mbentley  staff  319035392 Jul 16 07:15 /Users/mbentley/omada_data.tar
```

Store the tar file wherever you want; of course you can rename it to fit your needs.

Run a temporary container to import the tar data in the volumes.  This assumes you have the tar file in your user's home directory on the new machine and it is the same name:

```
docker run -it --rm \
  -w /opt/tplink/EAPController \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-logs:/opt/tplink/EAPController/logs \
  -v "${HOME}:/tmp/transfer" \
  alpine \
  tar xvf /tmp/transfer/omada_data.tar
```

Start the new omada controller container on the new host using the same volume paths.

Inside the persistent data directory, the folder structure should look like this:

```
# ls -la
total 61
drwxr-xr-x 8 omada omada   9 Oct 10 04:25 ./
drwxr-x--- 5 omada omada   5 Oct 10 06:19 ../
drwxr-xr-x 2 omada omada  16 Oct 10 01:15 autobackup/
drwxr-xr-x 4 omada omada 844 Oct 10 06:21 db/
drwxr-xr-x 3 omada omada   3 Sep 10 15:43 device-firmware/
drwxr-xr-x 2 omada omada   5 Jul 11 07:27 html/
drwxr-xr-x 2 omada omada   3 Oct 10 04:25 keystore/
-rw-r--r-- 1 root  root    6 Oct 10 04:25 LAST_RAN_OMADA_VER.txt
drwxr-xr-x 2 omada omada   2 Jul 11 07:22 pdf/
```

If it doesn't, you do not have the correct data.

mbentley, 2022-10-10:
Closing as this should get you set right.  Let me know if it doesn't work.  If you do not have any other backups of your volume data, you'll have to re-configure the controller.

closed 2022-10-10 by mbentley

andyslucky, 2022-10-10:
Thank you very much for your help and education. I had forgotten the details about `docker export`, once I navigated to my backup of docker volumes (as I weekly backup `var` and other crucial root fs folders) and found the temporary volume that had been used for the omada data mount, copying the data over and setting permissions worked like a charm. Thanks again!

mbentley, 2022-10-10:
Excellent - good to hear!