mbentley/docker-omada-controller #253: [Bug]: SMTP Sending using SendGrid

TheLaundryBoss, 2022-10-21:
### Describe the Bug

I am getting success on the test send however it does not get sent. I will put the log below. The error in the log states that the from email address is not verified although it is. This works on the Unifi controller on the same ubuntu server. 

### Expected Behavior

Email sent and recieved

### Steps to Reproduce

Send a test email 

### How You're Launching the Container

```plain
sudo
docker run -d \
  --name omada-controller \
  --restart unless-stopped \
  --net host \
  -e MANAGE_HTTP_PORT=8088 \
  -e MANAGE_HTTPS_PORT=8043 \
  -e PGID="508" \
  -e PORTAL_HTTP_PORT=8088 \
  -e PORTAL_HTTPS_PORT=8943 \
  -e PUID="508" \
  -e SHOW_SERVER_LOGS=true \
  -e SHOW_MONGODB_LOGS=false \
  -e SSL_CERT_NAME="tls.crt" \
  -e SSL_KEY_NAME="tls.key" \
  -e TZ=Etc/UTC \
  -v omada-data:/opt/tplink/EAPController/data \
  -v omada-work:/opt/tplink/EAPController/work \
  -v omada-logs:/opt/tplink/EAPController/logs \
  mbentley/omada-controller:5.3
```

### Container Logs

```plain
"log":"354 Continue\n","stream":"stdout","time":"2022-10-21T19:57:48.302405922Z"}
{"log":"Date: Fri, 21 Oct 2022 19:57:48 +0000 (UTC)\n","stream":"stdout","time":"2022-10-21T19:57:48.303040213Z"}
{"log":"From: Omada SDN Controller \u003capikey\u003e\n","stream":"stdout","time":"2022-10-21T19:57:48.303055131Z"}
{"log":"To: scott@hosiermail.com\n","stream":"stdout","time":"2022-10-21T19:57:48.30305966Z"}
{"log":"Message-ID: \u003c1662041817.16.1666382268302@f727df802aec\u003e\n","stream":"stdout","time":"2022-10-21T19:57:48.303063247Z"}
{"log":"Subject: [Omada] Controller Test Email\n","stream":"stdout","time":"2022-10-21T19:57:48.303067394Z"}
{"log":"MIME-Version: 1.0\n","stream":"stdout","time":"2022-10-21T19:57:48.303070971Z"}
{"log":"Content-Type: text/plain;charset=UTF-8\n","stream":"stdout","time":"2022-10-21T19:57:48.303074247Z"}
{"log":"Content-Transfer-Encoding: 7bit\n","stream":"stdout","time":"2022-10-21T19:57:48.303089616Z"}
{"log":"\n","stream":"stdout","time":"2022-10-21T19:57:48.303093423Z"}
{"log":"This is a message generated by Omada SDN Controller [Omada Controller_TNBI].\n","stream":"stdout","time":"2022-10-21T19:57:48.303096709Z"}
{"log":"Getting this message means that your SMTP server settings are correct.\n","stream":"stdout","time":"2022-10-21T19:57:48.303100145Z"}
{"log":".\n","stream":"stdout","time":"2022-10-21T19:57:48.303121425Z"}
{"log":"550 The from address does not match a verified Sender Identity. Mail cannot be sent until this error is resolved. Visit https://sendgrid.com/docs/for-developers/sending-email/sender-identity/ to see the Sender Identity requirements\n","stream":"stdout","time":"2022-10-21T19:57:48.338237217Z"}
{"log":"DEBUG SMTP: got response code 550, with response: 550 The from address does not match a verified Sender Identity. Mail cannot be sent until this error is resolved. Visit https://sendgrid.com/docs/for-developers/sending-email/sender-identity/ to see the Sender Identity requirements\n","stream":"stdout","time":"2022-10-21T19:57:48.338270058Z"}
{"log":"\n","stream":"stdout","time":"2022-10-21T19:57:48.338274656Z"}
{"log":"RSET\n","stream":"stdout","time":"2022-10-21T19:57:48.338278123Z"}
{"log":"250 I remember nothing\n","stream":"stdout","time":"2022-10-21T19:57:48.373642454Z"}
{"log":"DEBUG SMTP: MessagingException while sending, THROW: \n","stream":"stdout","time":"2022-10-21T19:57:48.37372601Z"}
{"log":"com.sun.mail.smtp.SMTPSendFailedException: 550 The from address does not match a verified Sender Identity. Mail cannot be sent until this error is resolved. Visit https://sendgrid.com/docs/for-developers/sending-email/sender-identity/ to see the Sender Identity requirements\n","stream":"stdout","time":"2022-10-21T19:57:48.373783457Z"}
{"log":"\n","stream":"stdout","time":"2022-10-21T19:57:48.373789419Z"}
{"log":"\u0009at com.sun.mail.smtp.SMTPTransport.issueSendCommand(SMTPTransport.java:2358)\n","stream":"stdout","time":"2022-10-21T19:57:48.373800509Z"}
{"log":"\u0009at com.sun.mail.smtp.SMTPTransport.finishData(SMTPTransport.java:2110)\n","stream":"stdout","time":"2022-10-21T19:57:48.373804396Z"}
{"log":"\u0009at com.sun.mail.smtp.SMTPTransport.sendMessage(SMTPTransport.java:1316)\n","stream":"stdout","time":"2022-10-21T19:57:48.373807553Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.a(SourceFile:252)\n","stream":"stdout","time":"2022-10-21T19:57:48.373810909Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.b(SourceFile:204)\n","stream":"stdout","time":"2022-10-21T19:57:48.373814215Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.a(SourceFile:192)\n","stream":"stdout","time":"2022-10-21T19:57:48.373817581Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.a(SourceFile:131)\n","stream":"stdout","time":"2022-10-21T19:57:48.373820617Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.a$b.a(SourceFile:131)\n","stream":"stdout","time":"2022-10-21T19:57:48.373823692Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.a$b.call(SourceFile:116)\n","stream":"stdout","time":"2022-10-21T19:57:48.373826758Z"}
[... 57 lines cut ...]
{"log":"DEBUG SMTP: MessagingException while sending, THROW: \n","stream":"stdout","time":"2022-10-21T19:58:19.738340786Z"}
{"log":"com.sun.mail.smtp.SMTPSendFailedException: 550 The from address does not match a verified Sender Identity. Mail cannot be sent until this error is resolved. Visit https://sendgrid.com/docs/for-developers/sending-email/sender-identity/ to see the Sender Identity requirements\n","stream":"stdout","time":"2022-10-21T19:58:19.738436364Z"}
{"log":"\n","stream":"stdout","time":"2022-10-21T19:58:19.738444219Z"}
{"log":"\u0009at com.sun.mail.smtp.SMTPTransport.issueSendCommand(SMTPTransport.java:2358)\n","stream":"stdout","time":"2022-10-21T19:58:19.738473483Z"}
{"log":"\u0009at com.sun.mail.smtp.SMTPTransport.finishData(SMTPTransport.java:2110)\n","stream":"stdout","time":"2022-10-21T19:58:19.738479475Z"}
{"log":"\u0009at com.sun.mail.smtp.SMTPTransport.sendMessage(SMTPTransport.java:1316)\n","stream":"stdout","time":"2022-10-21T19:58:19.738558933Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.a(SourceFile:252)\n","stream":"stdout","time":"2022-10-21T19:58:19.738565275Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.b(SourceFile:204)\n","stream":"stdout","time":"2022-10-21T19:58:19.738568681Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.a(SourceFile:192)\n","stream":"stdout","time":"2022-10-21T19:58:19.738571807Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.b.a(SourceFile:131)\n","stream":"stdout","time":"2022-10-21T19:58:19.738575354Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.a$b.a(SourceFile:131)\n","stream":"stdout","time":"2022-10-21T19:58:19.73857876Z"}
{"log":"\u0009at com.tplink.smb.omada.common.smtpclient.a$b.call(SourceFile:116)\n","stream":"stdout","time":"2022-10-21T19:58:19.738582116Z"}
{"log":"\u0009at java.util.concurrent.FutureTask.run(FutureTask.java:266)\n","stream":"stdout","time":"2022-10-21T19:58:19.738585984Z"}
{"log":"\u0009at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)\n","stream":"stdout","time":"2022-10-21T19:58:19.738629685Z"}
{"log":"\u0009at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)\n","stream":"stdout","time":"2022-10-21T19:58:19.738634715Z"}
{"log":"\u0009at java.lang.Thread.run(Thread.java:750)\n","stream":"stdout","time":"2022-10-21T19:58:19.73863784Z"}
{"log":"10-21-2022 19:58:19.738 WARN [email-task-0] [] c.t.s.o.c.s.b(256): Failed to send a email: 550 The from address does not match a verified Sender Identity. Mail cannot be sent until this error is resolved. Visit https://sendgrid.com/docs/for-developers/sending-email/sender-identity/ to see the Sender Identity requirements\n","stream":"stdout","time":"2022-10-21T19:58:19.738888859Z"}
{"log":"\n","stream":"stdout","time":"2022-10-21T19:58:19.7389109Z"}
{"log":"QUIT\n","stream":"stdout","time":"2022-10-21T19:58:19.739034521Z"}
{"log":"221 See you later\n","stream":"stdout","time":"2022-10-21T19:58:19.757842411Z"}
```

### Additional Context

I am using the exact same settings on Unifi controller without issue.  Ubuntu version 20.04.5 LTS

mbentley, 2022-10-21:
So I don't have control over the software so you may have to post on the forums but it looks like SendGrid doesn't like the from format of whatever you're entering.  It's valid from an address standpoint, I'm assuming you replaced it with `apikey` since it has your API key.  Not know how SendGrid expects to work, I am not totally sure what you should enter but I would expect it would just need to be a valid email address that is already validated through SendGrid to allow it to send on your behalf as documented on the page that message mentions: https://docs.sendgrid.com/for-developers/sending-email/sender-identity

TheLaundryBoss, 2022-10-21:
Thanks for the quick reply. That exact email is in the verified list and works on Unify. Oh well,  do you have a recommendation on an smtp relay that you know works

Regards
Scott Hosier

mbentley, 2022-10-21:
The only thing I can think of is that it doesn't like the complete email address format in the `from:` line which would be something like `Omada SDN Controller <noreply@mbentley.net>` so maybe they're doing something further with the validation that may be including the name.

As for a relay service, I am not sure what works directly.  I use Mailgun myself but I also have a local postfix install that I use as a single point to send email ƒrom my network and then postfix relays to Mailgun.  Works nicely because then I have caching on my network, if needed.

TheLaundryBoss, 2022-10-21:
I'll figure something out. I just migrated everything to Azure last night. Thanks for your help and making the Omada Docker!
Have a great weekend

-sh

mbentley, 2022-10-21:
Please do let me know if you find something out, if you continue to use SendGrid and figure out a solution.

closed 2022-10-21 by mbentley

TheLaundryBoss, 2022-10-21:
Will do..