mbentley/docker-omada-controller #256: [Bug]: Fails adoption after updating controller from 5.0 to 5.6

randomcuriouscode, 2022-10-29:
### Describe the Bug

I just updated my omada controller from 5.0 to 5.6. I have 3 EAP245 v3's (5.0.6 Build 20220429 Rel. 44315) in my network. They were previously adopted fine to the 5.0 controller.

After updating to 5.6, they are now stuck adopting. The logs don't show anything obvious. I am running the controller on a raspberry pi in docker, the mbentley/omada-controller/5.6-arm64 image. It was previously running the 5.0-arm64 image which worked fine.

The container is running in host mode so ports are exposed, and it does not appear any ports changed. Nothing else has changed on my network, and I did not change any controller settings. 

I don't see an obvious hint to what's wrong in the controller logs 

Any ideas what's going on?

### Expected Behavior

It works

### Steps to Reproduce

Upgrade 5.0 to 5.6 with an EAP245 (v5.0.6 Build 20220429 Rel. 44315) adopted on 5.0. After updating to 5.6 the controller tries to adopt it again, and fails.

### How You're Launching the Container

```plain
This was my old 5.0 docker-compose.yml:

version: "3.1"

services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:5.0-arm64
     environment:
      - TZ=America/Denver
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
     network_mode: host

     volumes:
      - omada-data:/opt/tplink/EAPController/data
      - omada-work:/opt/tplink/EAPController/work
      - omada-logs:/opt/tplink/EAPController/logs
     restart: unless-stopped
volumes:
  omada-data:
  omada-logs:
  omada-work:
```

This is my 5.6 docker-compose.yml:
```
version: "3.1"

services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:5.6-arm64
     environment:
      - PUID=508
      - PGID=508
      - TZ=America/Denver
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
[... 110 lines cut ...]
10-29-2022 15:33:51.051 INFO [https-jsse-nio-8043-exec-4] [] c.t.s.o.c.u.d.a(): list local interface macs: [mac1, mac2]
10-29-2022 15:34:08.676 INFO [comm-pool-15] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
10-29-2022 15:34:08.719 INFO [https-jsse-nio-8043-exec-7] [] c.t.s.c.l.a.AbstractReadWriteLockService(): [readWriteLockService]businessId:omadac.id:id get writeLock module:manager:upgrade.online.check fail, execute onFail.
10-29-2022 15:34:15.520 INFO [comm-pool-15] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
10-29-2022 15:34:15.546 INFO [https-jsse-nio-8043-exec-5] [] c.t.s.c.l.a.AbstractReadWriteLockService(): [readWriteLockService]businessId:omadac.id:id get writeLock module:manager:upgrade.online.check fail, execute onFail.
10-29-2022 15:34:31.390 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device1) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:34:59.510 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device2) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:35:10.797 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device3) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:35:55.528 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device1) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:36:16.639 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device2) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:36:22.052 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device3) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:37:08.708 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device1) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:37:36.798 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device2) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:37:38.419 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device3) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:37:45.925 INFO [https-jsse-nio-8043-exec-10] [] c.t.s.o.p.p.a.k(): Received invalid PortalBatchQueryDTO: PortalBatchQueryDTO(omadacId=id, siteId=id, portalIds=[])
10-29-2022 15:38:19.840 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device1) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:38:42.926 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device2) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:39:02.592 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device3) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:39:40.002 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device1) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:40:13.155 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device2) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:40:34.798 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device3) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:40:41.134 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device1) receive discovery when adopting, times[3] reach limit, set adopt fail
10-29-2022 15:41:17.255 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): OmadacId(id) DeviceMac(device2) receive discovery when adopting, times[3] reach limit, set adopt fail
```
```

mbentley, 2022-10-30:
You have a port conflict:

```
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29811
      - PORT_MANAGER_V1=29811
```

randomcuriouscode, 2022-11-01:
@mbentley thank you for the quick response!

closed 2022-11-01 by randomcuriouscode

mbentley, 2022-11-01:
Good deal - just to confirm, I'm assuming that was the problem then?

randomcuriouscode, 2022-11-01:
Yep, the duplicate port number was the problem