mbentley/docker-omada-controller #259: [Bug]: AI WLAN Optimization Not working

mike391, 2022-11-02:
### Describe the Bug

Enabling the AI WLAN optimization (Wireless Networks > AI WLAN Optimization) for any channels and clicking 'Save' gives a popup that says 'General Error'.

Docker logs below

### Expected Behavior

Should save the AI WLAN Configuration

### Steps to Reproduce

Go to Wireless Networks
Click AI WLAN Optimization
Enable a few radios
click Save
Observe error pop up in webui and errors in docker logs

### How You're Launching the Container

```plain
version: "3.1"

services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:latest
     user: "0:0"
     environment:
      - TZ=America/Toronto
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
     network_mode: host
     volumes:
      - /srv/dev-disk-by-label-RAID/Docker/.config/omada/data:/opt/tplink/EAPController/data
      - /srv/dev-disk-by-label-RAID/Docker/.config/omada/work:/opt/tplink/EAPController/work
      - /srv/dev-disk-by-label-RAID/Docker/.config/omada/logs:/opt/tplink/EAPController/logs
     restart: unless-stopped
```

### Container Logs

```plain
2022-11-02T15:09:56.836614536Z 11-02-2022 11:09:56.836 INFO [https-jsse-nio-8043-exec-10] [] c.t.s.o.p.p.a.k(): Received invalid PortalBatchQueryDTO: PortalBatchQueryDTO(omadacId=335662b0ab8f171407aa9939e668a5eb, siteId=62979755dae24f1fa37c60b8, portalIds=[])
2022-11-02T15:09:58.902988667Z 11-02-2022 11:09:58.901 ERROR [https-jsse-nio-8043-exec-8] [] c.t.s.o.d.h.i.ReflectionApiMethodHandler(): Invoke API method a failed.
2022-11-02T15:09:58.903016605Z java.lang.reflect.InvocationTargetException: null
2022-11-02T15:09:58.903021878Z 	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
2022-11-02T15:09:58.903035156Z 	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77) ~[?:?]
2022-11-02T15:09:58.903039926Z 	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
2022-11-02T15:09:58.903043731Z 	at java.lang.reflect.Method.invoke(Method.java:568) ~[?:?]
2022-11-02T15:09:58.903047357Z 	at com.tplink.smb.omada.dispatcher.handler.impl.ReflectionApiMethodHandler.handle(SourceFile:70) ~[dispatcher-web-support-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903051136Z 	at com.tplink.smb.omada.dispatcher.handler.impl.ReflectionApiMethodHandler.handle(SourceFile:38) ~[dispatcher-web-support-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903054841Z 	at com.tplink.smb.omada.dispatcher.chain.HandlerChain.execute(SourceFile:97) ~[dispatcher-web-support-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903058490Z 	at com.tplink.smb.omada.dispatcher.RestDispatcher.dispatch(SourceFile:66) ~[dispatcher-web-support-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903062295Z 	at com.tplink.smb.omada.apigateway.dispatch.ApiController.apiHandler(SourceFile:111) ~[api-gateway-core-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903066288Z 	at jdk.internal.reflect.GeneratedMethodAccessor422.invoke(Unknown Source) ~[?:?]
2022-11-02T15:09:58.903069828Z 	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
2022-11-02T15:09:58.903073492Z 	at java.lang.reflect.Method.invoke(Method.java:568) ~[?:?]
2022-11-02T15:09:58.903077005Z 	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.18.jar:5.3.18]
2022-11-02T15:09:58.903080752Z 	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.18.jar:5.3.18]
2022-11-02T15:09:58.903084565Z 	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.18.jar:5.3.18]
2022-11-02T15:09:58.903089691Z 	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.18.jar:5.3.18]
2022-11-02T15:09:58.903093847Z 	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.18.jar:5.3.18]
2022-11-02T15:09:58.903097697Z 	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.3.18.jar:5.3.18]
[... 70 lines cut ...]
2022-11-02T15:09:58.903546514Z 	at org.apache.catalina.valves.ErrorReportValve.invoke(ErrorReportValve.java:92) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903550134Z 	at org.apache.catalina.core.StandardEngineValve.invoke(StandardEngineValve.java:78) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903553782Z 	at org.apache.catalina.connector.CoyoteAdapter.service(CoyoteAdapter.java:360) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903557418Z 	at org.apache.coyote.http11.Http11Processor.service(Http11Processor.java:399) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903561086Z 	at org.apache.coyote.AbstractProcessorLight.process(AbstractProcessorLight.java:65) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903564742Z 	at org.apache.coyote.AbstractProtocol$ConnectionHandler.process(AbstractProtocol.java:889) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903568400Z 	at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.doRun(NioEndpoint.java:1743) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903572152Z 	at org.apache.tomcat.util.net.SocketProcessorBase.run(SocketProcessorBase.java:49) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903631026Z 	at org.apache.tomcat.util.threads.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1191) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903638314Z 	at org.apache.tomcat.util.threads.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:659) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903642268Z 	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61) ~[tomcat-embed-core-9.0.60.jar:9.0.60]
2022-11-02T15:09:58.903646033Z 	at java.lang.Thread.run(Thread.java:833) [?:?]
2022-11-02T15:09:58.903649562Z Caused by: java.lang.NullPointerException: Cannot invoke "java.lang.Integer.intValue()" because the return value of "org.apache.commons.lang3.tuple.Pair.getKey()" is null
2022-11-02T15:09:58.903657438Z 	at com.tplink.smb.omada.monitor.rfplanning.domain.service.j.c.b(SourceFile:165) ~[omada-rfplanning-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903661205Z 	at com.tplink.smb.omada.monitor.rfplanning.domain.service.d.c.b(SourceFile:163) ~[omada-rfplanning-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903669656Z 	at com.tplink.smb.omada.monitor.rfplanning.domain.service.d.c.a(SourceFile:115) ~[omada-rfplanning-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903673604Z 	at com.tplink.smb.omada.monitor.rfplanning.domain.service.h.f.a(SourceFile:59) ~[omada-rfplanning-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903677244Z 	at com.tplink.smb.omada.monitor.rfplanning.a.a.c(SourceFile:83) ~[omada-rfplanning-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903680811Z 	at com.tplink.smb.omada.monitor.rfplanning.port.e.a.a(SourceFile:78) ~[omada-rfplanning-5.6.3.jar:5.6.3]
2022-11-02T15:09:58.903684426Z 	... 100 more
```

### Additional Context

I checked tplink forums and didnt see any notices about this happening to anyone else. Curious if its an issue with the docker image or my docker compose setup

mbentley, 2022-11-03:
I don't seem to be able to reproduce this myself.  I just stood up a new controller with no devices (maybe that has something to do with it) and I can set the options:

[image]

What Omada devices do you have under the control of the software controller?

mike391, 2022-11-03:
That is odd. I have a router (ER605) and 2 APs Eap245 & Eap670.

I did just notice I got an error for the channel optimization settings if any of the APs had their channel setting set to something other than Auto. However power settings for these APs don't have an Auto option. I wonder if that could be why... Not 100% just a guess

ryaneford, 2022-11-10:
Having the same issue. Same hardware setup, 1 EAP245 and 1 EAP670

mbentley, 2022-11-12:
I will have to take a look when I don't have other people around.  I have my 3 APs set to specific channels so I'll have to test it out as well.  @ryaneford - do you also have your APs set to specific channels or are your set to auto?

mbentley, 2022-11-13:
So just setting automatically channel optimization for both 2.4 and 5ghz and clicking `Save` worked as expected - it saved just fine here.  I only have 2.4 and 5ghz devices.  Also enabling 6 works fine as well.  At this point, I did not try to hit "Scan and optimize" and my 3 WAPs are all set to specific channels as well.

[image]

All three of my WAPs are EAP245 v3.0.  I wonder if it has something to do with mixed devices possibly.

mbentley, 2022-11-13:
Looks like this may be an AP firmware issue according to this [Reddit thread](https://reddit.com/r/TPLink_Omada/comments/yu7kib/ai_wlan_optimization_feature_general_error/).  Apparently a beta firmware is required or waiting for a GA version.

mike391, 2022-11-14:
Nice, excellent find. I had a suspicion this was probably a tp-link issue.
Thanks for looking into it!

closed 2022-11-14 by mbentley