mbentley/docker-omada-controller #262: [Bug]: Unable to update firmware using port 443

cdonovan, 2022-11-04:
### Describe the Bug

I'm not totally sure if this is a bug with the container or controller software itself. I've been running the docker container and accessing it via Traefik using the default management port (8043) for a while. Using some redirect/header modification rules in Traefik, I don't have any issues updating device firmware. However, I recently switched to using port 443 as the management port to simplify my Traefik config. After the switch, device firmware updates stopped working. The file starts uploading but never seems to complete and eventually times out after 10 minutes.

I wound up reverting my Traefik/Omada updates, so it was using port 8043 again, and once I did that, the firmware update completed successfully. At this point, I feel like the issue is likely related to the Omada software, but when I brought it up on their forum, I was advised to come here instead.

### Expected Behavior

Device firmware updates should complete successfully.

### Steps to Reproduce

1. Run the docker container/omada software on port 443
2. Access via a reverse proxy such as Traefik
3. Start a device firmware update

### How You're Launching the Container

```plain
omada:
  container_name: omada
  image: mbentley/omada-controller:5.6
  restart: unless-stopped
  ports:
    - 8088:8088
    - 8843:8843
    - 29810:29810/udp
    - 29811:29811
    - 29812:29812
    - 29813:29813
    - 29814:29814
  environment:
    - PUID=1000
    - PGID=1000
    - MANAGE_HTTPS_PORT=443
    - TZ=America/Los_Angeles
  volumes:
    - ./omada/data:/opt/tplink/EAPController/data
    - ./omada/work:/opt/tplink/EAPController/work
    - ./omada/logs:/opt/tplink/EAPController/logs
  sysctls:
    - net.ipv4.ip_unprivileged_port_start=0
  labels:
    traefik.http.routers.omada.rule: Host(`omada.home.lan`)
    traefik.http.routers.omada.entrypoints: web-secure
    traefik.http.services.omada.loadbalancer.server.scheme: https
    traefik.http.services.omada.loadbalancer.server.port: 443
    traefik.http.services.omada.loadbalancer.serversTransport: ignoreSSL@file
```

### Container Logs

```plain
10-28-2022 10:56:50.622 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.DeviceUpgradeStaticTask(): upgrade task upgradeId [x] of OmadacId(x) is running...
10-28-2022 10:56:50.666 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.g(): wait for all switch check file complete, upgradeId[x] deviceCount[1]
10-28-2022 10:56:55.622 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.DeviceUpgradeStaticTask(): upgrade task upgradeId [x] of OmadacId(x) is running...
10-28-2022 10:56:55.666 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.g(): wait for all switch check file complete, upgradeId[x] deviceCount[1]
10-28-2022 10:57:00.622 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.DeviceUpgradeStaticTask(): upgrade task upgradeId [x] of OmadacId(x) is running...
10-28-2022 10:57:00.623 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.DeviceUpgradeStaticTask(): v2 device macx of omadacIdx upgradeIdx finished upgrade process in FIRMWARE_DOWNLOAD_FAILED
10-28-2022 10:57:00.631 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.c.c(): OmadacId(x)  upgradeId[x] DeviceMac(x) upgrade status change from[false] to [true]
10-28-2022 10:57:00.635 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.DeviceUpgradeStaticTask(): OmadacId(x) upgrade[id=x] finish.
10-28-2022 10:57:00.638 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.DeviceUpgradeStaticTask(): upgrade task upgradeId [x] of OmadacId(x) is finished
10-28-2022 10:57:00.641 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.c.c(): OmadacId(x) upgradeId[x] releaseUpgradeCatelog finished
10-28-2022 10:57:00.666 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.g(): wait for all switch check file complete, upgradeId[x] deviceCount[1]
10-28-2022 10:57:00.669 INFO [comm-pool-14] [] c.t.s.o.m.d.d.m.u.t.g(): all switch check file complete, upgradeId[x] check ok device count[0], start reboot
```

### Additional Context

I'd like to try running the controller software natively, but on port 443 to try and rule out Docker or Traefik as the culprit, but am not sure how easy it would be to convert the docker install to bare-metal. Would it be just installing the native client (probably on a spare machine) and copying the three bind mount folders to the same locations after the install is complete?

mbentley, 2022-11-05:
Well, when looking at this [FAQ document](https://www.tp-link.com/us/support/faq/3281/), I am not sure what it means when it comes to port `8043` as it mentions the use for the port is:
>1. When you visit the management page of Omada Software Controller via an HTTPS connection.
>2. When upgrading the firmware for the Omada devices with Omada Software Controller.

The usage for number two there is what I am not exactly sure of.  When I uploaded a firmware to perform a downgrade on one of my WAPs, it didn't start listening on any other ports so I don't think it needs to do that specifically during an upgrade or anything.

I use port 443 using macvlan on my docker container but do so directly without Traefik in front.  What part of the upgrade does it fail on? I was able to upload a firmware fine and it applied successfully.

When you say that you're having problems uploading the firmware, do you get to the point where you can upload the firmware, it shows a uploaded and then you get a button to click to perform the upgrade or does it fail just uploading the file?  If the problem is when uploading the file, could you open your browser's developer tools to see if there are any errors reported on the network tab while uploading?

You could try to run the controller natively - you would either need to try to do a backup/restore via the software or you could just try to copy over the persistent data directories to where the controller expects them and that ought to work, as long as you have matching OS versions of packages, especially MongoDB; not sure about how picky it might be with Java.

mbentley, 2022-11-26:
Any update on where exactly it was failing?

SamMousa, 2022-11-27:
I have the same issue. It likely has to do with the way the device tries to connect to the controller.

I'm using traefik reverse proxy with a certificate from LE. The management hostname is set to a public dns record that resolves to the IP of traefik.

cdonovan, 2022-11-28:
I haven't had a chance to play with the network really the past few weeks. Glad (is that the right word???) that someone else has experienced it as well though. I might be able to find some time between work & family being home this week to bring the wifi down to test this.

mbentley, 2022-11-29:
Also, do you have a link to the forum thread where you asked about this?  I am curious if they can have someone take a look at the behavior of the update code to see the controller is advertising a non-routable address or a port that isn't exposed through the reverse proxy.  The [port documentation](https://www.tp-link.com/us/support/faq/3265/) makes it sound like it should be using the https port for upgrades but it wouldn't surprise me if there is something untested with a reverse proxy in place.

cdonovan, 2022-11-30:
Here's a link to the thread - https://community.tp-link.com/en/business/forum/topic/587358. Just to clarify though, I'm not too sure that the reverse proxy itself is causing the issue, since it works fine behind traefik using the "standard" management port (8043).

The proxy config for the using port 8043 is a bit convoluted (has to adjust some headers & redirects for the auth flow), but in the end, firmware updates still work.  It was when I changed (i.e. simplified) the config to remove the extra header/redirect stuff for the login and used port 443 directly by changing the management port in the Omada UI.  When I did that, I removed the original port 8043 from being exposed to the host network. I wonder, even though the port was changed in the Omada software (and was indeed accessed via 443), if some things related to the firmware update flow were still looking to communicate on port 8043 (which now had access cut off).  One of my planned experiments was to keep the new config (using port 443) but keeping 8043 still exposed. As soon as I get a change, I'll let you know how it goes.

cdonovan, 2022-12-30:
Over the holiday break I finally had some time to think about this issue further and realize the likely problem. Traefik is currently exposed on port 443 and I'm *mostly* using http host rules to route to the correct container.

For the Omada controller, that's something like `omada.internal`, which (although used by me to manage the network via the ui), is likely *not* what is used by the actual devices 🤦. I can't expose port 443 for Omada specifically, because it's already being used by traefik, so I think the only way around this would be to set up additional routing rules. I'm not sure if the firmware update communication uses http or just tcp and what specific criteria could be used to correctly route to the omada container. Perhaps it's actually simpler to keep the login redirect hack (utilizing the unoccupied default port) instead of adding additional routing logic that could be just as fragile as the login redirect itself.

I think this can likely be closed, although if you do have any information on the actual firmware update logic (protocol, common headers, etc.) it could prove useful in trying to set up the proxy routing.

mbentley, 2023-01-05:
This is really confusing as to where any why it would be failing.  If I am understanding correctly, you're only exposing Traefik on 443 in both scenarios, right?  I wasn't sure if your mention of it working when the controller is using 8043 was there you had 8043 also exposed on Traefik as well or if it was still only using 443 but with more complex rewrite rules.

I wish they would provide some more detailed information about how the upgrade process worked as it is hard to understand why it doesn't work.

cdonovan, 2023-01-05:
I think the problem is primarily that the device itself needs to communicate back to the controller. When the controller is running on 8043 it can do so, since nothing else is using that port. However, in the scenario where the controller is running on port 443, when the device tries to communicate, traefik is already using that port, so I imagine the packets go into traefik and die, since traefik doesn't know where to send them.

mbentley, 2023-01-06:
So if the configuration is Traefik is listening on 443 and 8043 with only the Omada Controller on 8043, that would mean that the Omada controller should be the default server that traffic is sent to if no host match is found.  Makes me curious if the device are just not passing a hostname or what exactly the devices are sending for a host header or SNI in the case of https.  I suppose the quick test would be to set the Omada Controller as the default server backend to use in case of no matches and see if it succeeds then.

That's the only scenario where I can think of where it would succeed with Traefik as the reverse proxy to 8043 but it would fail using 443, common with other services.

legonigel, 2023-02-11:
I also have hit this in my k3s cluster using Traefik ingress. I definitely remember doing upgrades to my devices sometime last year without issue, but it doesn't work any longer. I know that 5.7 added the "Device Management Hostname/IP" option, so I wonder if something changed with 5.7.

dsander, 2023-02-20:
I might have/had a very similar problem, firmware updates used to work but then didn't. I am also using traefik to handle the SSL termination. To fix it on version 5.8 I have set `Device Management Hostname/IP` to the same value as `Controller Hostname/IP` and firmware updates started working again. I have tried to roll back the change to reproduce the problem but then it still worked, not sure if the change somehow persisted on the clients or what is up with that.

Here is the slightly obfuscated docker compose service entry I am using (it is processed using ansible):

```yaml
    omada:
      image: 'mbentley/omada-controller:5.8'
      container_name: omada
      networks:
        - traefik_proxy
      restart: unless-stopped
      environment:
        - MANAGE_HTTP_PORT=8088
        - MANAGE_HTTPS_PORT=443
        - PORTAL_HTTP_PORT=8088
        - PORTAL_HTTPS_PORT=8843
        - SHOW_SERVER_LOGS=true
        - SHOW_MONGODB_LOGS=false
        - TZ={{ dc_TZ }}
      volumes:
        - '{{ appdata_path }}/omada/data:/opt/tplink/EAPController/data'
        - '{{ appdata_path }}/omada/work:/opt/tplink/EAPController/work'
        - '{{ appdata_path }}/omada/logs:/opt/tplink/EAPController/logs'
      ports:
        - '29810:29810/udp'
        - '29811:29811'
        - '29812:29812'
        - '29813:29813'
        - '29814:29814'
      labels:
        - "traefik.enable=true"
        - "traefik.http.routers.omada.entrypoints=https"
        - "traefik.http.services.omada.loadbalancer.server.port=443"
        - "traefik.http.services.omada.loadbalancer.server.scheme=https"

```

mbentley, 2023-03-21:
So while I know this isn't going to solve the issue for everyone, there is a help article related to failures of firmware upgrades that may explain some of the behaviors that people are seeing: https://community.tp-link.com/en/business/forum/topic/559150

IndiGP, 2023-03-29:
Its very interesting, that only switches and routers are not capable of updating firmware. Accses Points are still getting updates from my controller behind nginx revers proxy (manager).. Very strange behaviour.
Tested with AP653 from version 1.0.0 --> 1.0.4

IndiGP, 2023-03-30:
> I would assume that would be because certain (newer) firmwares will use the controller's HTTPS port for updates, other older firmwares will use a specific update port as documented in the linked forum post above.

Yes, but every port is open.. I set mgm port in controller to 443 and on reverse proxy on 443 and I am forwarding every port from 29810-14 to container instance itself. So very annoying thats its not working anymore. Hope there will be an update in future for that.