mbentley/docker-omada-controller #263: [Bug]:

fusioncahaos, 2022-11-05:
### Describe the Bug

Getting the following errors in order.  Randomly it seems atleast every week, my controller will not load the UI and I am forced to run through setup and configuration each time this happens.

getExpImprove failed
can't get admins list in
Cannot find omadac by id:
can not get owner accountId.

### Expected Behavior

Open the docker WebUI container and see the login screen.  I have ran the omada controller on a virtualbox environment and never had this issue before.

### Steps to Reproduce

Setup and configure omada controller intially
wait a couple days or a week and try to load into the WebUI
right click the docker from unraid and select WebUI
webpage shows invalid website

### How You're Launching the Container

```plain
Right click the container icon and select WebUI
```

### Container Logs

```plain
text  error  warn  system  array  login  

tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
tail: '/opt/tplink/EAPController/logs/server.log' has been replaced;  following new file
10-27-2022 19:47:29.910 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29811, export localhost/127.0.0.1:29811
10-27-2022 19:47:29.911 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29813 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
10-27-2022 19:47:29.912 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29813, export localhost/127.0.0.1:29813
10-27-2022 19:47:29.914 INFO [main] [] c.t.s.e.t.n.NettyTcpServer(): TCP server /0.0.0.0:29814 global traffic shaping, writeLimit: 536870912, readLimit: 0, checkInterval: 1000, max wait time: 15000
10-27-2022 19:47:29.919 INFO [main] [] c.t.s.e.t.a.t.AbstractServer(): Start NettyTcpServer bind /0.0.0.0:29814, export localhost/127.0.0.1:29814
10-27-2022 19:47:29.919 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Device connector server started.
10-27-2022 19:47:29.920 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Omada Controller started
10-27-2022 19:47:29.920 INFO [main] [] c.t.s.o.s.t.SuccessExitTask(): Omada Controller started
10-27-2022 19:47:29.920 INFO [main] [] c.t.s.o.s.OmadaBootstrap(): record: finished run omada tasks
10-27-2022 19:47:33.136 WARN [https-jsse-nio-8043-exec-5] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
10-27-2022 19:47:33.161 WARN [https-jsse-nio-8043-exec-5] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
10-27-2022 19:47:33.445 INFO [discovery-work-group-0] [] c.t.s.o.m.d.d.m.b.a(): MANAGE_BY_OTHERS Device 5C-A6-E6-EE-FA-A8 on omadac 6653083fd1ecee1d59977091b89081cf is discoveried.
10-27-2022 19:47:36.220 INFO [discovery-work-group-1] [] c.t.s.o.m.d.d.m.b.a(): MANAGE_BY_OTHERS Device 6C-5A-B0-9E-A6-8E on omadac 6653083fd1ecee1d59977091b89081cf is discoveried.
10-27-2022 19:48:57.784 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(6653083fd1ecee1d59977091b89081cf).
10-27-2022 19:48:57.788 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
10-27-2022 19:48:57.789 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
10-27-2022 19:48:57.798 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
10-27-2022 19:48:57.806 WARN [check-update-work-group-0] [] c.t.s.o.m.d.d.m.u.c.a(): can't get admins list in 6653083fd1ecee1d59977091b89081cf
10-27-2022 20:47:27.271 INFO [scheduled-pool-5] [] c.t.s.o.c.p.c.f(): Scheduling executed, set updateOmadacCacheEnable: false
10-28-2022 05:43:45.010 INFO [common-block-pool-10] [] c.t.s.o.m.d.c.a(): clear omadac data with omadacId [6653083fd1ecee1d59977091b89081cf]
10-28-2022 05:43:45.019 INFO [manager-data-retention-pool-0] [] c.t.s.c.u.l.LocalUpgradeFileCache(): delete expired upgrade file, number = 0
10-28-2022 19:48:57.780 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(6653083fd1ecee1d59977091b89081cf).
10-28-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
10-28-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
10-28-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
10-28-2022 19:48:57.783 WARN [check-update-work-group-0] [] c.t.s.o.m.d.d.m.u.c.a(): can't get admins list in 6653083fd1ecee1d59977091b89081cf
10-29-2022 05:43:45.002 INFO [common-block-pool-4] [] c.t.s.o.m.d.c.a(): clear omadac data with omadacId [6653083fd1ecee1d59977091b89081cf]
10-29-2022 05:43:45.003 INFO [manager-data-retention-pool-0] [] c.t.s.c.u.l.LocalUpgradeFileCache(): delete expired upgrade file, number = 0
10-29-2022 19:48:57.780 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(6653083fd1ecee1d59977091b89081cf).
10-29-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
10-29-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
[... 27 lines cut ...]
11-02-2022 19:48:57.780 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(6653083fd1ecee1d59977091b89081cf).
11-02-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
11-02-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
11-02-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
11-02-2022 19:48:57.783 WARN [check-update-work-group-0] [] c.t.s.o.m.d.d.m.u.c.a(): can't get admins list in 6653083fd1ecee1d59977091b89081cf
11-03-2022 04:53:01.828 INFO [monitor-data-track-pool-1] [] c.t.s.o.m.s.g(): generate report dto list:6653083fd1ecee1d59977091b89081cf, GenerateReportDTO(super=EventParamDTO(), omadacId=6653083fd1ecee1d59977091b89081cf, omadacType=0, pdfCnt=0, csvCnt=0, pdfSize=0.0, csvSize=0.0)
11-03-2022 04:53:01.828 INFO [monitor-data-track-pool-1] [] c.t.s.o.m.d.p.c.c(): getExpImprove failed, omadacId6653083fd1ecee1d59977091b89081cf
11-03-2022 05:43:45.003 INFO [common-block-pool-6] [] c.t.s.o.m.d.c.a(): clear omadac data with omadacId [6653083fd1ecee1d59977091b89081cf]
11-03-2022 05:43:45.004 INFO [manager-data-retention-pool-0] [] c.t.s.c.u.l.LocalUpgradeFileCache(): delete expired upgrade file, number = 0
11-03-2022 19:48:57.780 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(6653083fd1ecee1d59977091b89081cf).
11-03-2022 19:48:57.781 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
11-03-2022 19:48:57.781 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
11-03-2022 19:48:57.782 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
11-03-2022 19:48:57.783 WARN [check-update-work-group-0] [] c.t.s.o.m.d.d.m.u.c.a(): can't get admins list in 6653083fd1ecee1d59977091b89081cf
11-04-2022 05:43:45.003 INFO [common-block-pool-1] [] c.t.s.o.m.d.c.a(): clear omadac data with omadacId [6653083fd1ecee1d59977091b89081cf]
[... cut ...]
11-04-2022 19:48:57.783 WARN [check-update-work-group-0] [] c.t.s.o.m.d.d.m.u.c.a(): can't get admins list in 6653083fd1ecee1d59977091b89081cf
11-04-2022 20:24:36.538 WARN [https-jsse-nio-8043-exec-7] [] c.t.s.o.m.c.a.o.d(): Cannot find omadac by id: OmadacId(946b5f90b9094e2a5b701bd76e923ba7)
11-04-2022 20:26:00.227 WARN [https-jsse-nio-8043-exec-10] [] c.t.s.o.m.c.a.o.d(): Cannot find omadac by id: OmadacId(946b5f90b9094e2a5b701bd76e923ba7)
11-04-2022 20:26:03.629 WARN [https-jsse-nio-8043-exec-9] [] c.t.s.o.m.l.d.m.N(): omadacId 6653083fd1ecee1d59977091b89081cf can not get owner accountId.
```

mbentley, 2022-11-05:
Really hard to say without further info but it sounds like something is happening to the persistent data.

- Have you been trying to get the controller to upgrade through the web UI of the controller or anything like that?
- Might there be something that is updating the container automatically for you?  It's hard to tell if this is on a NAS or directly on Docker or what exactly.
- If I knew how you were launching that container, that would help as "Right click the container icon and select WebUI" doesn't help provide info about how you started the container.

mariushosting, 2022-11-05:
version 5.6.4 is out.

mbentley, 2022-11-05:
> version 5.6.4 is out.

It's out for the OC200, not for the software based controller.

fusioncahaos, 2022-11-07:
> Really hard to say without further info but it sounds like something is happening to the persistent data.
>
> * Have you been trying to get the controller to upgrade through the web UI of the [...]

- I have not tried upgrading the controller through the WebUI, this issue does occur whether there is an update or not to the controller
- no, there is no setting i have turned on to auto-update the docker container in unraid
- The container is within unraid, the container is set to a static ip within my subnet and the port is 8080 which the "Right click..." as mentioned brings me to this address.

fusioncahaos, 2022-11-07:
Specific errors i see

11-05-2022 22:54:54.113 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Device connector server started.
11-05-2022 22:54:54.113 ERROR [main] [] c.t.s.f.c.FacadeUtils(): facadeMsgEnable is not enable, msg: Omada Controller started

fusioncahaos, 2022-11-07:
Its failing to pull my profile...

11-05-2022 22:54:26.477 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): Starting OmadaLinuxMain v5.6.3 using Java 17.0.4 on 51a60ef01bae with PID 1 (/opt/tplink/EAPController/lib/local-starter-5.6.3.jar started by omada in /opt/tplink/EAPController/lib)
11-05-2022 22:54:26.480 INFO [main] [] c.t.s.o.s.OmadaLinuxMain(): No active profile set, falling back to 1 default profile: "default"
11-05-2022 22:54:37.219 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.
11-05-2022 22:54:40.222 WARN [main] [] o.s.d.c.CustomConversions(): Registering converter from class java.time.LocalDateTime to class org.joda.time.LocalDateTime as reading converter although it doesn't convert from a store-supported type! You might want to check your annotation setup at the converter implementation.****

mbentley, 2022-11-07:
Those messages are typical to see, nothing that points to an issue so far.  The logs from when you find the controller having issues, specifically before and after you see the controller asking to be set up again would be helpful.

fusioncahaos, 2022-11-09:
Closing issue.  It keeps occurring and is reached my limits of having to keep redoing the docker setup.  I have moved this effort to a VM install instead.

mbentley, 2022-11-13:
Hey - I was finally able to reproduce this!  It has nothing to do with the controller running in Docker or not.  It's that there seems to be some sort of race condition in the startup of the controller where it tries to load the page before the controller is fully up.  As you can see, the URL has `null` where it should have a controller ID:
[image]

Clearing the URL and then reloading will resolve the issue to allow it to re-discover the controller's ID.  There isn't a problem or data lost, it's a bug in the software it seems.

mbentley, 2022-11-13:
I created a post on the forums: https://community.tp-link.com/en/business/forum/topic/588910?page=1

agrogers, 2023-09-25:
I had this problem on the ER7212PC. Refreshing and clearing the Chrome browser cache didn't help. I then tried on Edge and it worked fine. Maybe just a coincidence.