mbentley/docker-omada-controller #266: [Feature]: How to upgrade version and keep settings

blueman457, 2022-11-22:
### What problem are you looking to solve?

I'm trying to upgrade the docker image for the first time, I'm using docker on a Synology NAS. I'm trying to upgrade the image, but every time I do, I lose the settings that are within the controller. 

I may be doing this wrong, but I didn't know if there was a method to keep the controller settings while upgrading the docker image.

Thanks for our work!

mbentley, 2022-11-22:
I am not sure exactly how the container is launched when you run it through the Synology but the reason that you are losing your settings would be because it is not persisting the necessary data.  Using the compose file for reference, there are [two directories that are used for persistent data](https://github.com/mbentley/docker-omada-controller/blob/master/docker-compose.yml#L29-L30) but the only one that actually matters as it relates to the controller settings and configuration would be the `data` directory which is `/opt/tplink/EAPController/data` inside the container.  You need to map that to a volume of some sorts so that when you stop and remove your controller container, you can re-attach it to your new container.  Now if Synology uses a typical Docker setup, there may be Docker managed volumes which are just assigned random IDs that have your persistent data in them, unless they are being removed when you remove the old container. They would look something like this:

```
$ docker volume ls
DRIVER    VOLUME NAME
local     7f85c6a2354f8c90ed672f3e08a939fa1cf4b451e54bd42740fce42c54c12d4e
```

It's possible to extract that data, if it exists in a unnamed volume, to a location so I will refrain from typing that out for now but what I typically do is use a [bind mount](https://docs.docker.com/storage/bind-mounts/) which does exactly that - bind mount a path from your host's file system into the container at the path specified.  That's how I typically deal with all of my persistent data across the dozens of containers I manage on my host.

blueman457, 2022-11-22:
Thanks for the reply - I'm a novice in all of this, I'll dig around synology docker and see if I can make it work. Thanks again for your work!

closed 2022-11-22 by blueman457

adhawkins, 2022-11-22:
Just to add to this, Synology does have docker-compose and the 'docker compose' plugin available. You can then manager the container using a docker-compose file, which I find is an easier way to make sure all the settings are correct and persisted correctly.

Andy

blueman457, 2022-12-04:
I ended up needing creating a new container and restoring my omada settings with a backup configuration file. Either way, thanks for making the container!