mbentley/docker-omada-controller #268: Constant disconnect from cloud.

jonas740, 2022-12-07:
### Describe the Bug

I have started to get intermittent notifications that my  ER605 is disconnected from the controller. Today its been like 10 times but everything works at home and my local controller is accessible and does not show any errors from what I can see.

It does however show that a lot of clients was disconnected and then reconnected.

I am using TP-Link ER605 v2 and a EAP-615-Wall.

I did notice that the docker compose you provided contains more things nowadays but should that be the issue.

And I did try deploy that docker compose but that made all my settings go away and I did provide a backup so some settings reappeared but far from all so I went back to the old docker-compose. 
[_omada-controller_logs.txt](https://github.com/mbentley/docker-omada-controller/files/10180454/_omada-controller_logs.txt)

### Expected Behavior

Well I expect no disconnects. 

### Steps to Reproduce

Intermittent. 

### How You're Launching the Container

```plain
version: "3.1"
services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:latest
     healthcheck:
      disable: true
     environment:
      - TZ=Etc/UTC
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
     network_mode: host
     volumes:
      - omada-data:/opt/tplink/EAPController/data
      - omada-work:/opt/tplink/EAPController/work
      - omada-logs:/opt/tplink/EAPController/logs
     restart: unless-stopped
volumes:
  omada-data:
  omada-work:
  omada-logs:
```

### Container Logs

```plain
12-07-2022 21:08:49.151 INFO [https-jsse-nio-8043-exec-5] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:09:49.155 INFO [https-jsse-nio-8043-exec-4] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:10:49.156 INFO [https-jsse-nio-8043-exec-1] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:11:40.156 WARN [quartzScheduler_Worker-1] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (log_limit_check, null) but did not execute because corresponding handler log_limit_check doesn't exist
12-07-2022 21:11:49.156 INFO [https-jsse-nio-8043-exec-5] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:12:49.159 INFO [https-jsse-nio-8043-exec-3] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:13:49.159 INFO [https-jsse-nio-8043-exec-2] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:14:49.169 INFO [https-jsse-nio-8043-exec-7] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:16:27.154 INFO [https-jsse-nio-8043-exec-7] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:16:27.411 INFO [https-jsse-nio-8043-exec-10] [] c.t.s.o.p.p.a.k(): Received invalid PortalBatchQueryDTO: PortalBatchQueryDTO(omadacId=64cde7237e224b81b1a287f7de0034f4, siteId=6337647ef54a8326902ac563, portalIds=[])
12-07-2022 21:16:40.157 WARN [quartzScheduler_Worker-2] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (log_limit_check, null) but did not execute because corresponding handler log_limit_check doesn't exist
12-07-2022 21:17:28.648 INFO [https-jsse-nio-8043-exec-7] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:18:28.667 INFO [https-jsse-nio-8043-exec-6] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 21:19:28.670 INFO [https-jsse-nio-8043-exec-10] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
[... 66 lines cut ...]
12-07-2022 22:02:28.766 INFO [https-jsse-nio-8043-exec-5] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:03:28.766 INFO [https-jsse-nio-8043-exec-1] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:04:28.782 INFO [https-jsse-nio-8043-exec-7] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:05:28.772 INFO [https-jsse-nio-8043-exec-5] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:06:28.773 INFO [https-jsse-nio-8043-exec-2] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:06:40.157 WARN [quartzScheduler_Worker-1] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (log_limit_check, null) but did not execute because corresponding handler log_limit_check doesn't exist
12-07-2022 22:07:28.777 INFO [https-jsse-nio-8043-exec-6] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:08:20.000 INFO [pool-4-thread-1] [] c.t.s.e.s.c.c(): [STAT] {"statName":"OmadaDeviceStat","deviceType":"gateway","ecspVer":"2.2.0","deviceNum":"1"}
12-07-2022 22:08:20.000 INFO [pool-4-thread-1] [] c.t.s.e.s.c.c(): [STAT] {"statName":"OmadaDeviceStat","deviceType":"ap","ecspVer":"2.3.0","deviceNum":"1"}
12-07-2022 22:08:28.781 INFO [https-jsse-nio-8043-exec-5] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:09:28.784 INFO [https-jsse-nio-8043-exec-8] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:10:28.799 INFO [https-jsse-nio-8043-exec-1] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:11:28.797 INFO [https-jsse-nio-8043-exec-8] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:11:40.157 WARN [quartzScheduler_Worker-2] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (log_limit_check, null) but did not execute because corresponding handler log_limit_check doesn't exist
12-07-2022 22:12:28.794 INFO [https-jsse-nio-8043-exec-7] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:13:28.803 INFO [https-jsse-nio-8043-exec-9] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:14:28.793 INFO [https-jsse-nio-8043-exec-4] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:15:28.810 INFO [https-jsse-nio-8043-exec-6] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:16:28.798 INFO [https-jsse-nio-8043-exec-8] [] c.t.s.o.c.u.d.a(): list local interface macs: [48-4D-7E-E7-7C-15, 02-42-D1-9A-D2-A4, 02-42-A3-C0-58-02, 02-42-55-CC-E3-DA]
12-07-2022 22:16:40.154 WARN [quartzScheduler_Worker-1] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (log_limit_check, null) but did not execute because corresponding handler log_limit_check doesn't exist
```

### Additional Context

I don't know if recent cyber attacks ongoing here and there in Europe and Sweden can be the cause aswell.

jonas740, 2022-12-08:
> I should have asked which version of the controller you are currently on.  I wasn't sure if you were pulling new versions or if you had an older version.  I can't imagine anything being a problem specifically from a container packaging standpoint as you're re-using the host's network interface so the ports should be exposed from the host, assuming no port conflicts on the host.  I don't have a TP-Liink router so I am not as familiar with those as much as APs.  I've certainly seen a number of issues with people having disconnects with their router but not frequent.
> 
> Have you also submitted a forum post with TP-Link?  I'm guessing it is more of a software issue but I don't know if any of the warnings are are important or not.

Hello.

I am using your latest controller. It has been working just great but yesterday connection with the cloud started acting weird.

And also, are the differences from your compose crucial then what I have? You have some more ports assigned if I am not mistaken.

Yes I have created an issue with tp-link. Could just be their servers acting up since the controller itself has not shown any disconnects with my router.

I did recently switch from EAP245 to the EAP-615-Wall, dunno if that could cause any issues.

mbentley, 2022-12-08:
I'm assuming by latest you mean 5.7.4 (as you could be running the `latest` tag but not have the most recent image pulled down).  So from a compose perspective, the port differences will not matter when you're using host mode as Docker ignores the ports published when in host mode as there is nothing to publish since they're directly connected to the host's network interface(s).  Main thing I would suggest doing would be to validate that the ports are not in use (using something like `netstat -ltnpu` should help show the ports in use from your host as root; may be worth stopping the controller, running netstat to capture the ports and validate none are conflicting on the host).  You can look at the ports and protocols from the latest compose example or look at the example `docker run` commands in the README.

Happy to look at the issue on the forums (assuming it isn't a private ticket) to see if they push back because the controller is running in a container.

jonas740, 2022-12-08:
yes i am running 5.7.4

And unfortunaley i made a ticket direct to TP-Link. But there are others on the tp link forum that has experianced the same issue

I did not mention this but this is something reported in the logs that is strange "The LAN IP address/mask of EAP615-Wall Livingroom were changed to 192.168.1.80/255.255.255.0" but that is the ip adress it has had from start. 

[https://community.tp-link.com/en/business/forum/topic/223458?page=2](url)

[https://community.tp-link.com/en/home/forum/topic/582160](url)

I am using the Power injector that i used for my EAP245 for the EAP615-Wall, maybe that is the cause for problems. But this started yesterday and i have had the EAP615-Wall for 5 days or something like that. I have however ordered another power injector.

I changed my docker compose a bit and removed the old pulled image so this is how it looks like now.

`
version: "3.1"

services:
   omada-controller:
     container_name: omada-controller
     image: mbentley/omada-controller:5.7
     restart: unless-stopped
     stop_grace_period: 60s
     network_mode: host
     environment:
      - PUID=1000
      - PGID=1000
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8043
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME="tls.crt"
      - SSL_KEY_NAME="tls.key"
      - TZ=Etc/UTC
     volumes:
      - omada-data:/opt/tplink/EAPController/data
      - omada-work:/opt/tplink/EAPController/work
      - omada-logs:/opt/tplink/EAPController/logs

volumes:
  omada-data:
  omada-work:
  omada-logs:
`

mbentley, 2022-12-08:
I also see the "The LAN IP address/mask..." messages when my APs reconnect, regardless of IPs actually being changed so I don't think that's an issue.  I'd only be concerned about the power injector if the device is actually going offline I suppose.

ganey, 2022-12-28:
I have this to, my ER605 keeps missing the heartbeats, so my log is full of 'ER605 was disconnected' alerts, even though everything is working fine

tophee, 2023-01-19:
Have you been able to solve this? I seem to be having the same or a similar issue. For me it's not just the router but all devices that intermittently disconnect from the controller... I suspect this is a big in the actual controller, not the docker image. Did you get a response from TP-Link?

jonas740, 2023-01-19:
Hello.

I don't want to jinx things but since I updated to chromium release it does
not seem to have behaved like it did.

I will have a look through the logs later this evening to be sure.

Tp-link did massive research and even had access to my controller but they
could not give me an explanation at the time but they said they were going
to hand it over to the development team.

And yes I might have forgotten to say so but it was a bit intermittent,
some times only er605 was dropping and sometimes all units dropped out.

I will have a look later this evening and get back to you.

Best Regards / Jonas

tophee, 2023-01-19:
I can't seem to understand what that chromium version is or does... Apart from that it seems to solve this issue, what is it for?

> it was a bit intermittent, some times only er605 was dropping and sometimes all units dropped out.

Exactly the same here: some times one, sometimes multiple devices drop out. And sometimes the Omada logs also show that they reconnected and sometimes they don't (but they still reconnect).

Here is some of the stuff I see in the container logs around the times when disonnects occur:

~~~
01-19-2023 05:37:59.337 INFO [manage-work-group-8] [] c.t.s.e.p.a.a(): Fail to send message REBUILD_RESPONSE to C0-C9-E3-4B-A3-EA, cause manage server route is null
01-19-2023 05:37:59.337 WARN [manage-work-group-8] [] c.t.s.o.m.d.p.t.g.a(): send rebuild reply to omadacId OmadacId(7a6c92d25f861217009ad14b40d49ea1) & mac DeviceMac(C0-C9-E3-4B-A3-EA) error, com.tplink.smb.ecsp.common.TransResult@42e9158[errCode=2001,msg=ERR_DEVICE_ROUTE_CACHE_NULL,result=<null>,addressDTO=<null>]
01-19-2023 05:38:00.022 INFO [device-timeout-workgroup-1] [] c.t.s.o.m.d.d.m.f.b(): Device DeviceMac(C0-C9-E3-4B-A3-EA) omadacId OmadacId(7a6c92d25f861217009ad14b40d49ea1) status change from Connected to Heartbeat Missed
01-19-2023 05:38:00.023 INFO [device-timeout-workgroup-0] [] c.t.s.o.m.d.d.m.f.b(): Device DeviceMac(C0-C9-E3-4B-A4-5C) omadacId OmadacId(7a6c92d25f861217009ad14b40d49ea1) status change from Connected to Heartbeat Missed
01-19-2023 05:38:28.690 INFO [discovery-work-group-68] [] c.t.s.o.m.d.d.m.b.a(): MANAGED_BY_OWN Device C0-C9-E3-4B-A3-EA on omadac 7a6c92d25f861217009ad14b40d49ea1 is discoveried.
01-19-2023 05:38:38.815 INFO [adopt-work-group-4] [] c.t.s.o.m.d.d.m.d.a.G(): Ap OmadacId(7a6c92d25f861217009ad14b40d49ea1) SiteId(Default) DeviceMac(C0-C9-E3-4B-A3-EA) adopt[auto=true] ok
01-19-2023 05:38:39.182 INFO [adopt-work-group-4] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(7a6c92d25f861217009ad14b40d49ea1) DeviceMac(C0-C9-E3-4B-A3-EA)
01-19-2023 05:38:44.913 WARN [quartzScheduler_Worker-1] [] c.t.s.c.s.c.TaskExecutorService(): receive scheduled event with identity (log_limit_check, null) but did not execute because corresponding handler log_limit_check doesn't exist
01-19-2023 05:38:44.934 INFO [monitor-topology-pool-14] [] c.t.s.o.c.u.d.a(): list local interface macs: [76-DB-FF-2F-A5-B1, 76-DB-FF-2F-A5-B1, CE-3F-6E-D4-A7-EC, 02-42-1F-FD-64-F3, 02-42-B0-A1-AD-36, 02-42-B2-8C-9B-11, 02-42-C5-08-55-A0, 02-42-79-DA-B8-26, 02-42-55-B8-BC-A1, 02-42-7C-2F-A0-24, 02-42-D1-40-F2-35, 02-42-27-A7-67-12, 02-42-C4-48-86-98, 02-42-77-DA-5F-38, 02-42-C5-8B-85-1C, 02-42-5A-9E-6D-53, 02-42-95-14-90-AF, 02-42-11-55-C6-7C, 02-42-81-13-3E-74, 02-42-D0-A5-28-48, 02-42-0F-CC-72-90, 02-42-9A-5B-A5-C8, 02-42-DE-7C-E5-26]
01-19-2023 05:38:45.049 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:C8-2B-96-10-DD-A9 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.049 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:28-6C-07-82-FF-CC is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.050 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:B4-E6-2D-79-F2-01 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.050 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:84-CC-A8-AD-88-F1 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.051 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:84-CC-A8-AD-29-5C is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.051 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:F6-1A-DD-17-C6-68 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.051 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:24-CE-33-A2-4E-26 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.052 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:F4-03-2A-63-73-4E is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.052 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:EC-FA-BC-C4-BC-70 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.053 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:00-55-DA-5F-3A-53 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.053 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:C8-2B-96-10-DF-9D is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
01-19-2023 05:38:45.054 WARN [monitor-topology-pool-14] [] c.t.s.o.c.d.a.n(): Client:EC-FA-BC-C4-B7-26 is not OswClientImage, Cannot modify TrafficRequireMark. omadac:7a6c92d25f861217009ad14b40d49ea1, site:Default
[... cut ...]
	at com.mongodb.client.internal.MongoClientDelegate$DelegateOperationExecutor.execute(MongoClientDelegate.java:183) ~[mongodb-driver-sync-4.4.2.jar:?]
	at com.mongodb.client.internal.MongoIterableImpl.execute(MongoIterableImpl.java:135) ~[mongodb-driver-sync-4.4.2.jar:?]
	at com.mongodb.client.internal.MongoIterableImpl.iterator(MongoIterableImpl.java:92) ~[mongodb-driver-sync-4.4.2.jar:?]
	at org.springframework.data.mongodb.core.MongoTemplate.executeFindMultiInternal(MongoTemplate.java:2853) ~[spring-data-mongodb-3.3.3.jar:3.3.3]
	... 32 more
~~~

jonas740, 2023-01-19:
> 01-19-2023 05:37:59.337 WARN [manage-work-group-8] [] c.t.s.o.m.d.p.t.g.a(): send rebuild reply to omadacId OmadacId(7a6c92d25f861217009ad14b40d49ea1) & mac DeviceMac(C0-C9-E3-4B-A3-EA) error, ***@***.***[errCode=2001,msg=ERR_DEVICE_ROUTE_CACHE_NULL,result=<null>,addressDTO=<null>]

I might have jinxed myself.

Because today I got a message that er605 was connected to controller.

So I checked the events log tab in the omada interface and I could see that
it had connected but I could not find that that the ER605 was disconnected
before that wich I find strange.

It also showed a lot of devices disconnected and connected but when using
the actual network I have not noticed anything.

This is very strange and I have googled about it a lot and we ain't alone
having this issue. An issue that not always means anything strange enough.

Are there any power saving features that causes this maybe?

There was an update a week ago for my ER605 and yesterday there was an
update for my EAP-615-Wall so the update last night might have triggered
something?

Med vänlig hälsning / Jonas Öhrfeldt
Hagfors El-Montage AB
070-370 58 26

mbentley, 2023-01-19:
> I can't seem to understand what that chromium version is or does... Apart from that it seems to solve this issue, what is it for?

The Chromium version doesn't do anything outside of [adding Chromium](https://github.com/mbentley/docker-omada-controller/blob/master/Dockerfile.chromium) by extending the base image.  So if you're on 5.7 and you use the Chromium tag to 5.7, it has `chromium-browser` which is required to generate pdf reports since I would assume TP-Link uses a library which requires a page to be rendered in a browser and then "printed" (or saved) as a pdf.

I'll have to take a look at the logs you shared to see if there is anything that sticks out to me at all.

tophee, 2023-01-22:
I still don't really understand the Chromium thing, but since I don't need to generate any pdfs, I guess I don't need it. And I don't really see how it could possibly solve the issue here. 

>  it looks like the controller isn't able to connect to MongoDB. Is that something that happens when the controller is starting up or is it during normal operation?

That is during normal operation. I searched the logs for mongodb and found that these errors are all over the place. I'd almost say that it is a mere coincidence that it was in the log at the same time as the device disconnect. 

At the end of december, there were days completely littered with the mongodb issue, but now it hasn't occurred since the ones I posted above. So maybe it fixed itself at some point when I restarted the container? No, `docker ps` tells me that it has been running for three weeks. So the restart three weeks ago probably fixed the problem that was filling the logs in december, but it's not completely fixed because it occasionally occurred also after the last restart.

> Anything in the mongodb logs?

I am not sure what to look for, so I searched for `error` Most of the occurrences are of the type
~~~
2023-01-19T05:15:38.302+0000 I COMMAND  [conn850] command admin.$cmd command: getLastError { getlasterror: 1, $db: "admin" } numYields:0 reslen:79 locks:{} protocol:op_msg 953ms
~~~
and I assume that is irrelevant.

The only three real errors are these (with some context before and after: 

~~~
2023-01-19T06:23:32.633+0000 I NETWORK  [conn935] received client metadata from 127.0.0.1:59846 conn935: { driver: { name: "mongo-java-driver|sync|spring-boot", version: "4.4.2" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.39-3-pve" }, platform: "Java/Private Build/17.0.5+8-Ubuntu-2ubuntu120.04" }
2023-01-19T06:23:33.291+0000 I WRITE    [LogicalSessionCacheRefresh] update config.system.sessions command: { q: { _id: { id: UUID("d3617bd5-dc81-475e-98bc-737d18e0bf62"), uid: BinData(0, E3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855) } }, u: { $currentDate: { lastUse: true } }, multi: false, upsert: true } planSummary: IDHACK keysExamined:1 docsExamined:1 nMatched:1 nModified:1 keysInserted:1 keysDeleted:1 numYields:2 locks:{ Global: { acquireCount: { r: 3, w: 3 } }, Database: { acquireCount: { w: 3 } }, Collection: { acquireCount: { w: 3 } } } 446317ms
2023-01-19T06:24:14.462+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:59850 #937 (5 connections now open)
2023-01-19T06:24:14.462+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:59852 #938 (6 connections now open)
2023-01-19T06:24:14.462+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:59854 #939 (7 connections now open)
2023-01-19T06:24:14.462+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:59856 #940 (8 connections now open)
2023-01-19T06:24:14.462+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:59858 #941 (9 connections now open)
2023-01-19T06:24:14.463+0000 I NETWORK  [listener] connection accepted from 127.0.0.1:59860 #942 (10 connections now open)
2023-01-19T06:24:14.465+0000 I NETWORK  [conn933] Error sending response to client: SocketException: Broken pipe. Ending connection from 127.0.0.1:59842 (connection id: 933)
2023-01-19T06:24:14.465+0000 I NETWORK  [conn933] end connection 127.0.0.1:59842 (9 connections now open)
2023-01-19T06:24:14.473+0000 I COMMAND  [conn934] command admin.$cmd command: isMaster { isMaster: 1, helloOk: true, client: { driver: { name: "mongo-java-driver|sync|spring-boot", version: "4.4.2" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.39-3-pve" }, platform: "Java/Private Build/17.0.5+8-Ubuntu-2ubuntu120.04" }, $db: "admin" } numYields:0 reslen:223 locks:{} protocol:op_query 275506ms
2023-01-19T06:24:14.473+0000 I NETWORK  [conn934] Error sending response to client: SocketException: Broken pipe. Ending connection from 127.0.0.1:59844 (connection id: 934)
2023-01-19T06:24:14.473+0000 I NETWORK  [conn934] end connection 127.0.0.1:59844 (8 connections now open)
2023-01-19T06:24:14.473+0000 I COMMAND  [conn935] command admin.$cmd command: isMaster { isMaster: 1, helloOk: true, client: { driver: { name: "mongo-java-driver|sync|spring-boot", version: "4.4.2" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.39-3-pve" }, platform: "Java/Private Build/17.0.5+8-Ubuntu-2ubuntu120.04" }, $db: "admin" } numYields:0 reslen:223 locks:{} protocol:op_query 89729ms
2023-01-19T06:24:14.473+0000 I NETWORK  [conn935] Error sending response to client: SocketException: Broken pipe. Ending connection from 127.0.0.1:59846 (connection id: 935)
2023-01-19T06:24:14.473+0000 I NETWORK  [conn935] end connection 127.0.0.1:59846 (7 connections now open)
2023-01-19T06:24:14.473+0000 I NETWORK  [conn936] received client metadata from 127.0.0.1:59848 conn936: { driver: { name: "mongo-java-driver|sync|spring-boot", version: "4.4.2" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.39-3-pve" }, platform: "Java/Private Build/17.0.5+8-Ubuntu-2ubuntu120.04" }
2023-01-19T06:24:14.473+0000 I NETWORK  [conn939] received client metadata from 127.0.0.1:59854 conn939: { driver: { name: "mongo-java-driver|sync|spring-boot", version: "4.4.2" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.39-3-pve" }, platform: "Java/Private Build/17.0.5+8-Ubuntu-2ubuntu120.04" }
[... cut ...] 36] end connection 127.0.0.1:59848 (6 connections now open)
2023-01-19T06:24:14.474+0000 I NETWORK  [conn938] received client metadata from 127.0.0.1:59852 conn938: { driver: { name: "mongo-java-driver|sync|spring-boot", version: "4.4.2" }, os: { type: "Linux", name: "Linux", architecture: "amd64", version: "5.15.39-3-pve" }, platform: "Java/Private Build/17.0.5+8-Ubuntu-2ubuntu120.04" }
2023-01-19T06:24:14.474+0000 I NETWORK  [conn939] end connection 127.0.0.1:59854 (5 connections now open)
~~~

mbentley, 2023-01-22:
Sorry, I over-explained the Chromium thing.  Chromium is just the purely open source version of Google Chrome and as you said, not relevant unless you're generating pdfs.

Just to check to see if there is anything that sticks out from my own logs, I checked to see if I have those sorts of disconnects and errors to see if they're errors but something that is seen regularly but I am not seeing those in mine.  I'm also regularly hitting my controller to talk to the APIs to get client data via Telegraf so I have a pretty active system despite only having 3 APs.

I'm probably at the end of my ability to debug anything further as I am not sure that this is a problem related to packaging in a container and it would probably require more in depth knowledge about the controller than I have.

tophee, 2023-01-22:
OK, thanks a lot for looking into this. I guess I'll just leave it and hope for TP-Link to eventually fix it (if there is anything broken).

If I could ask you a quick question about the permissions of the log files. I noticed that while the `server.log` has `644` permissions (on host) but `mongod.log` has `600`, which made it difficult for me to access it. Is there any reason why the permissions are set like this?

(Totally off-topic, but if you are curious about ChatGPT, [here ](https://shareg.pt/l68UUZ6) is my conversation with it, trying to figure out how to get access to `mongod.log` earlier today. Since all my questions in that conversation are genuine, I cannot evaluate how good a "teacher" ChatGPT was in this case, but in areas where I am more knowledgeable, I have found it interesting to explore the limitations of the bot to better understand how it works and from that, I know, that it does make some grave mistakes sometimes. So if you share any similar interest in this new technology, take a look at the conversation. Otherwise, just ignore it.  ;-)

mbentley, 2023-01-22:
Hmm, not really sure why the logs would have different permissions set unless it's something about how the permissions are set on the parent directory maybe. I don't believe I'm doing anything within the image itself isn't setting the permissions so it must be the controller and mongodb itself that are setting the permissions upon creation.  Oddly enough, mine are 644 for both:

```# ls -la | grep -v "log.gz"
total 4888
drwxr-xr-x 2 omada omada       96 Jan 22 00:00 ./
drwxr-x--- 5 omada omada        6 Nov 14 09:06 ../
-rw-r--r-- 1 omada omada 32694061 Jan 22 01:15 mongod.log
-rw-r--r-- 1 omada omada   109493 Jan 22 08:00 server.log```

mbentley, 2023-12-10:
Closing as it seems to be an upstream issue; may no longer be relevant.  Reopen if there is something I can do.

closed 2023-12-10 by mbentley