mbentley/docker-omada-controller #272: [Bug]: Potential thread leak

xinbinhuang, 2022-12-12:
### Controller Version

v5.7.4

### Describe the Bug

Observer a potential thread leak, not sure what is the exact cause yet. Will monitor for a few days and try to reproduce it, then report back. Not sure if this is a bug on the image or on omada itself 🤷 

Memory & thread change in the last 7 days
[image]

Manually kill & restart the pod
[image]

### Expected Behavior

na

### Steps to Reproduce

na

### How You're Launching the Container

```plain
k8s
```

### Container Logs

```plain
na
```

mbentley, 2022-12-12:
I'd be curious if there is anything in the logs that might give an idea as to if it was executing a process when the memory jump happened.  I'm also interested in your usage of the controller - what sort & how many devices are connected to your controller?

I only have three EAP245 APs but I haven't seen overly excessive memory jumps on my container; I've been running 5.7.4 since it's been released.  I have a 4GB memory limit set on the container; currently sitting at 1.193GiB used at the moment:

[image]

Will have to check but I don't think I'm currently pulling # of threads or even processes through the Docker Telegraf plugin but I'm currently seeing 305 PIDs in the container from `docker stats` for a quick check.

Stupco, 2023-01-17:
I've just seen something similar, this memory leak doesn't happen when the Omada Container isn't running:
[image]
Controller Version: 5.7.4

mbentley, 2023-01-17:
I just double checked the java process arguments to make sure there isn't something new that I missed but I don't see anything; only things really different here is passing `-Xms128m -Xmx1024m` and omitting `/usr/share/java/commons-daemon.jar` from the classpath which I am pretty certain is only needed with jsvc.

from the Dockerfile CMD:
```
/usr/bin/java -server -Xms128m -Xmx1024m -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true -cp /opt/tplink/EAPController/lib/*::/opt/tplink/EAPController/properties: com.tplink.smb.omada.starter.OmadaLinuxMain
```

from the `control.sh` typically used on a native install:
```
-server -XX:MaxHeapFreeRatio=60 -XX:MinHeapFreeRatio=30 -XX:+HeapDumpOnOutOfMemoryError -XX:HeapDumpPath=/opt/tplink/EAPController/logs/java_heapdump.hprof -Djava.awt.headless=true -cp /usr/share/java/commons-daemon.jar:/opt/tplink/EAPController/lib/*:/opt/tplink/EAPController/properties com.tplink.smb.omada.starter.OmadaLinuxMain
```

Since there is another data point, anything in the logs indicating that a process is running in particular when you see jumps in memory usage?  When it comes to things like memory leaks, those are probably things that would have to be addressed to [TP-Link through their forums](https://community.tp-link.com/en/business/forum/582) since they have the ability to fix things in the software itself.

closed 2023-05-04 by mbentley