mbentley/docker-omada-controller #275: [Upgrade]: 5.8.4 released

evolucja, 2023-01-13:
### What problem are you looking to solve?

Hi, it looks like there is a version 5.8.4 for Windows released. How long does it usually take to release also the Linux version so the Docker image could be updated as well?

Thanks
Dave

mbentley, 2023-01-13:
It's whenever they (TP-Link) releases it - seems pretty random though.  As soon as they release a new version though, I am usually able to update the Docker image pretty quickly - sometimes in a few hours but I definitely aim to have it done within 24 hours unless there are major changes.  Usually the runtime environment changes are minimal so it's a quick process.

evolucja, 2023-01-13:
Thank you. I was just curious how long it usually takes as I've been recently onboarded to the Omada ecosystem :)

mbentley, 2023-01-13:
No problem - I've seen them release a Windows release one month, Linux the next - there is no rhyme or reason for when and how they release.  Seems independent from each other.  Sometimes the Linux release comes out first, other times the Windows.  One consistent thing is that the hardware controller from TP-Link seems to lag behind much further than either the Linux or Windows software controllers.

guitarmanusa, 2023-01-14:
On the latest-chromium (5.7.4) when logging into the controller I get a pop message on the dashboard saying a "newer" version is available: v5.7.4 Build 202211051_1668995905916, if I'm reading the build date of this version correctly, 05 Nov 2022, would you say this is actually older than the build date of 5.7.4 posted on TP Link's site? (11/21/2022)

mbentley, 2023-01-14:
Not sure why that would be exactly - I just started a new controller from the `latest-chromium` tag and it's reporting as being version 5.7.4 and that it is up to date:

[image]

As far as I know, there have not been any updates to the package since 5.7.4 was released - it's been on `https://static.tp-link.com/upload/software/2022/202211/20221121/Omada_SDN_Controller_v5.7.4_Linux_x64.tar.gz` [since I merged support for 5.7.4](https://github.com/mbentley/docker-omada-controller/commit/ef38cd36591b6d918b1518f3528d24d3f8e45368#diff-043df5bdbf6639d7a77e1d44c5226fd7371e5259a1e4df3a0dd5d64c30dca44fR45) and the tarball available from the TP-Link site is still reporting as the same.

guitarmanusa, 2023-01-17:
Of course after I post this the message goes away :( I swear I'm not crazy!  But I'm glad I'm now also showing the same!  Thanks!

bartoszkaminski, 2023-01-30:
Linux version is ready 🙂

closed 2023-01-30 by mbentley