mbentley/docker-omada-controller #276: [Feature]: How To Interact with the Docker Image

eyramm, 2023-01-20:
### What problem are you looking to solve?

This is not really a feature and I wish I reached out to you in person. 
I am running the docker image, but I am unable to launch the Controller in my browser as I don't see any clear process to
Can you please assist with this? What do I do after serving the image?

Thanks

mbentley, 2023-01-20:
Part of that is going to depend on how you started the container - either the run command, compose command, or however you launched it.  You access the web interface by connecting to the port that is the value of the `PORTAL_HTTPS_PORT` variable (defaults to 8843) so it would typically be `https://<ip-or-hostname-to-docker-host>:8843`.  The ports can be modified on the **_very first_** run time of the container by the [optional variables](https://github.com/mbentley/docker-omada-controller#optional-variables).  There are a few gotchas when changing ports as the controller works a bit differently than some other apps I've used and how it routes traffic via the ports as using Docker's port exposing feature to re-map a port from one on the host to another inside the container **does not work** with the controller as it is a software design limitation the best I can tell.

mbentley, 2023-02-06:
Let me know if I can help further.  Closing.

closed 2023-02-06 by mbentley

xeoncross, 2024-01-12:
This call-out would be very helpful in the readme or docker page