mbentley/docker-omada-controller #279: [Bug]: Can't connect/manage/login to the cloud controller with the tplink cloud account

zeropoint46, 2023-02-03:
### Controller Version

v5.8.4

### Describe the Bug

login to omada.tplinkcloud.com, look for the connected controller in the list. Use the "house/launch" icon to the right to login via your tplink cloud account and it's fails. opens new tab, immediately closes it and gives an error that says:
`Connection failed. Please refresh your browser and try again.`
When I look at the browser request it looks like this:
`https://use1-api-omada.tplinkcloud.com/api/v2/startLaunch?=&token=MYTOKENHERE`
if I manually go there I see this:
`{"errorCode":-1500,"msg":"Omada server error.","result":null}`

on the omada controller I have this error in the logs a lot:
```
Exception in thread "Thread-11" java.lang.NullPointerException: Cannot invoke "com.tplink.smb.omada.identityaccess.domain.model.user.d.m.y()" because "aUser" is null
	at com.tplink.smb.omada.identityaccess.domain.model.user.R.a(SourceFile:113)
	at com.tplink.smb.omada.identityaccess.port.common.d.a.a(SourceFile:233)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1625)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at com.tplink.smb.omada.identityaccess.port.common.d.a.b(SourceFile:234)
	at com.tplink.smb.omada.identityaccess.port.common.d.a.a(SourceFile:146)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b$d.a(SourceFile:718)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b$d.a(SourceFile:698)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b.a(SourceFile:386)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b.a(SourceFile:263)
	at com.tplink.smb.omada.cloud.platform.pf.b.b(SourceFile:243)
	at com.tplink.smb.omada.cloud.platform.pf.b.a(SourceFile:150)
	at com.tplink.smb.omada.cloud.platform.a.a.b.onRequest(SourceFile:58)
	at com.tplink.eap.cloudsdk.client.q.a(SourceFile:268)
	at com.tplink.eap.cloudsdk.client.p.a(SourceFile:138)
	at com.tplink.eap.cloudsdk.client.s.run(SourceFile:831)
	at java.base/java.lang.Thread.run(Thread.java:833)
02-03-2023 08:57:44.406 WARN [Thread-13] [] c.t.e.c.c.q(): send heartbeat getting timeout response: {"error_code":-20002}
02-03-2023 08:57:54.415 WARN [Thread-13] [] c.t.e.c.c.q(): send heartbeat getting timeout response: {"error_code":-20002}
02-03-2023 08:57:56.460 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start sync Cloud Users OmadacId(randomidstring).
02-03-2023 08:57:56.648 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
02-03-2023 08:57:56.650 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
02-03-2023 08:57:57.415 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
02-03-2023 08:58:04.421 INFO [Thread-13] [] c.t.e.c.c.q(): unsuccessful heartbeat response: {"error_code":-20002}
02-03-2023 08:58:04.422 INFO [Thread-13] [] c.t.e.c.c.q(): Thread 'heartBeatThread' is stopped
02-03-2023 08:58:04.422 INFO [Thread-7] [] c.t.e.c.c.CloudClient(): Close connection to service server.
02-03-2023 08:58:04.423 INFO [Thread-7] [] c.t.e.c.c.q(): set connection status: DISCONNECTED_NORMAL
02-03-2023 08:58:04.423 INFO [Thread-12] [] c.t.e.c.c.q(): expiredRequestCleanThread is interrupted.
02-03-2023 08:58:04.423 INFO [Thread-12] [] c.t.e.c.c.q(): Thread 'expiredRequestCleanThread' is stopped
02-03-2023 08:58:04.439 INFO [Thread-7] [] c.t.e.c.c.CloudClient(): Connect service server automatically, ConnectionType is PERSISTENT_CONNECTION.
02-03-2023 08:58:04.439 INFO [Thread-7] [] c.t.e.c.c.q(): set connection status: CONNECTING
02-03-2023 08:58:04.925 INFO [Thread-7] [] c.t.e.c.c.q(): set connection status: CONNECTED
02-03-2023 08:58:04.925 INFO [Thread-7] [] c.t.e.c.c.q(): The result of connection is true.

```

### Expected Behavior

should be able to successfully login via cloud

### Steps to Reproduce

the prerequisite to this issue was actually me installing this controller on a new docker host and restoring from backup from my old one. Here are the steps:
1. setup new controller on new host
2. login to start initial setup
3. select restore from backup
4. restore from back and proceed
5. add cloud connection
6. try to login

### How You're Launching the Container

```plain
I use portainer and docker and the docker-compose you have listed on this repo.
```

### Container Logs

```plain
Exception in thread "Thread-14" java.lang.NullPointerException: Cannot invoke "com.tplink.smb.omada.identityaccess.domain.model.user.d.m.y()" because "aUser" is null
	at com.tplink.smb.omada.identityaccess.domain.model.user.R.a(SourceFile:113)
	at com.tplink.smb.omada.identityaccess.port.common.d.a.a(SourceFile:233)
	at java.base/java.util.stream.ReferencePipeline$3$1.accept(ReferencePipeline.java:197)
	at java.base/java.util.ArrayList$ArrayListSpliterator.forEachRemaining(ArrayList.java:1625)
	at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:509)
	at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:499)
	at java.base/java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:921)
	at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.base/java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:682)
	at com.tplink.smb.omada.identityaccess.port.common.d.a.b(SourceFile:234)
	at com.tplink.smb.omada.identityaccess.port.common.d.a.a(SourceFile:146)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b$d.a(SourceFile:718)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b$d.a(SourceFile:698)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b.a(SourceFile:386)
	at com.tplink.smb.omada.cloud.platform.pf.scm.b.a(SourceFile:263)
	at com.tplink.smb.omada.cloud.platform.pf.b.b(SourceFile:243)
	at com.tplink.smb.omada.cloud.platform.pf.b.a(SourceFile:150)
	at com.tplink.smb.omada.cloud.platform.a.a.b.onRequest(SourceFile:58)
	at com.tplink.eap.cloudsdk.client.q.a(SourceFile:268)
	at com.tplink.eap.cloudsdk.client.p.a(SourceFile:138)
	at com.tplink.eap.cloudsdk.client.s.run(SourceFile:831)
	at java.base/java.lang.Thread.run(Thread.java:833)
02-03-2023 09:07:55.459 INFO [https-jsse-nio-8043-exec-4] [] c.t.s.o.c.u.d.a(): list local interface macs: [CA-CB-05-B8-C5-4A, 02-42-F8-E5-0D-C0]
02-03-2023 09:08:10.940 WARN [Thread-16] [] c.t.e.c.c.q(): send heartbeat getting timeout response: {"error_code":-20002}
02-03-2023 09:08:20.001 INFO [pool-4-thread-1] [] c.t.s.e.s.c.c(): [STAT] {"statName":"OmadaDeviceStat","deviceType":"ap","ecspVer":"2.3.0","deviceNum":"2"}
02-03-2023 09:08:20.947 WARN [Thread-16] [] c.t.e.c.c.q(): send heartbeat getting timeout response: {"error_code":-20002}
02-03-2023 09:08:30.954 INFO [Thread-16] [] c.t.e.c.c.q(): unsuccessful heartbeat response: {"error_code":-20002}
02-03-2023 09:08:30.954 INFO [Thread-16] [] c.t.e.c.c.q(): Thread 'heartBeatThread' is stopped
02-03-2023 09:08:30.954 INFO [Thread-7] [] c.t.e.c.c.CloudClient(): Close connection to service server.
02-03-2023 09:08:30.954 INFO [Thread-7] [] c.t.e.c.c.q(): set connection status: DISCONNECTED_NORMAL
02-03-2023 09:08:30.955 INFO [Thread-15] [] c.t.e.c.c.q(): expiredRequestCleanThread is interrupted.
02-03-2023 09:08:30.955 INFO [Thread-15] [] c.t.e.c.c.q(): Thread 'expiredRequestCleanThread' is stopped
02-03-2023 09:08:30.967 INFO [Thread-7] [] c.t.e.c.c.CloudClient(): Connect service server automatically, ConnectionType is PERSISTENT_CONNECTION.
02-03-2023 09:08:30.967 INFO [Thread-7] [] c.t.e.c.c.q(): set connection status: CONNECTING
02-03-2023 09:08:31.444 INFO [Thread-7] [] c.t.e.c.c.q(): set connection status: CONNECTED
02-03-2023 09:08:31.444 INFO [Thread-7] [] c.t.e.c.c.q(): The result of connection is true.
```

### Additional Context

this may very well be any number of issues that have nothing to do with this repo or the container, however this is the only place that seems to actually be helpful that I've found. Additional context is the first controller I was running that I restored this one from is running from the same container image on another docker host, however it wasn't setup using docker compose, I just used normal docker run commands. Thanks again if you have any pointers.

mbentley, 2023-02-03:
Interesting.  I would say it isn't likely specifically container related but more backup and restore related.  Have you tried to re-enter your cloud credentials into the controller after restoring the configuration or disabling the cloud connection and then re-enabling it?  I just checked on my controller that's running 5.8.4 and it's connected to the cloud fine but I haven't done a restore.

I'll have to see if I can setup a new controller, connect it to the cloud, do a backup and restore to another controller and see if I see the same behavior.  My hunch is something with the backup and restore process and the cloud credentials not transferring to the new controller (maybe something related to the controller ID is used in password hashing or something like that?)

zeropoint46, 2023-02-03:
I appreciate the quick response and the help. Yeah I have no idea what is is. I even dug down into the mongoDB to look to try and see where it's populating that "aUser" obj/var from to see why it's null.. I guess I could just delete everything and try starting over, but I kinda already did that, so I'm not sure.

mbentley, 2023-02-06:
Sorry for the delay - had a busy weekend.  I just tried to reproduce this but couldn't.  I setup a brand new controller, did a minimal setup, got it cloud connected, verified it worked, then took a backup, stopped & removed the container, removed the persistent data and then started a new container.  I re-ran through the user setup wizard, I didn't enable the cloud connection at first, did the restore from backup, went into the local web interface, added the cloud credentials and then I logged in to the TP-Link cloud web interface where it showed as connected and it worked as expected.  It still had my old controller in the cloud user interface so I removed the old one.  I can only imagine it might be a bug in the software outside of the basics and haven't yet seen anyone else mention having the same issue, at least in my issues, yet.

zeropoint46, 2023-02-07:
Hey, I really appreciate your willingness to look, even though all signs point to it not being an issue with your container. So here are the steps that I took to resolve the issue.

1. logged into the current container locally and did an unbind from tplink cloud.
2. took a fresh backup/export and specifically unchecked the boxes to backup user data.
3. deleted container and volumes and redeployed completely from scratch.
4. went through setup wizard, linked tplink cloud account on initial setup, then restored from backup

voila, that worked, guess there was maybe some corrupt data in my other backup file or something odd. Either way, thanks for the help and thanks for making this container.

closed 2023-02-07 by zeropoint46

mbentley, 2023-02-07:
Good deal - thanks for documenting the steps that worked for you. I'm just glad that others are able to get some use out of things I mostly create for myself.