mbentley/docker-omada-controller #280: [Bug]: After Upgrading from 5.7.4 to 5.8.4 some pages won't work anymore

baldisos, 2023-02-07:
### Controller Version

5.8.4

### Describe the Bug

After Upgrading from 5.7.4 to 5.8.4 the Global View does not show any informations or sometimes a "404
Sorry, the page you visited does not exist.". Under Site View, the sidebar doesn't pop up anymore when selecting a Device or Client. The Light Bulb icon (forgot it's name) does only show a blank page.
[image]
[image]

### Expected Behavior

The Global View should show whatever information is supposed to be there. The sidebar should open when selecting an item.

### Steps to Reproduce

- Upgrade from 5.7.4 to 5.8.4
- Open the Omada Controller in Browser
- being greeted by a blank page

### How You're Launching the Container

```plain
version: "3.1"

services:
  omada-controller:
    container_name: omada-controller
    image: mbentley/omada-controller:5.8
    restart: unless-stopped
    stop_grace_period: 60s
    network_mode: host
    environment:
      - PUID=1026
      - PGID=100
      - MANAGE_HTTP_PORT=8088
      - MANAGE_HTTPS_PORT=8043
      - PORTAL_HTTP_PORT=8088
      - PORTAL_HTTPS_PORT=8843
      - PORT_APP_DISCOVERY=27001
      - PORT_ADOPT_V1=29812
      - PORT_UPGRADE_V1=29813
      - PORT_MANAGER_V1=29811
      - PORT_MANAGER_V2=29814
      - PORT_DISCOVERY=29810
      - SHOW_SERVER_LOGS=true
      - SHOW_MONGODB_LOGS=false
      - SSL_CERT_NAME=tls.crt
      - SSL_KEY_NAME=tls.key
      - TZ=Europe/Berlin
    volumes:
      - /volume1/docker/omada/data:/opt/tplink/EAPController/data
      - /volume1/docker/omada/logs:/opt/tplink/EAPController/logs
      - /volume1/docker/omada/work:/opt/tplink/EAPController/work
```

### Container Logs

```plain
2023-02-07T12:19:08.270343172Z INFO: Validating user/group (omada:omada) exists with correct UID/GID (1026:100)
2023-02-07T12:19:08.696051166Z INFO: Group (omada) already exists with a different name; renaming 'users' to 'omada'
2023-02-07T12:19:10.276852093Z INFO: User (omada) doesn't exist; creating
2023-02-07T12:19:13.101118356Z INFO: Time zone set to 'Europe/Berlin'
2023-02-07T12:19:13.170920641Z INFO: Value of 'manage.http.port' already set to 8088 in omada.properties
2023-02-07T12:19:13.176250980Z INFO: Value of 'manage.https.port' already set to 8043 in omada.properties
2023-02-07T12:19:13.181126197Z INFO: Value of 'portal.http.port' already set to 8088 in omada.properties
2023-02-07T12:19:13.185833950Z INFO: Value of 'portal.https.port' already set to 8843 in omada.properties
2023-02-07T12:19:13.190636377Z INFO: Value of 'port.adopt.v1' already set to 29812 in omada.properties
2023-02-07T12:19:13.195781205Z INFO: Value of 'port.app.discovery' already set to 27001 in omada.properties
[... 99 lines cut ...]
2023-02-07T12:21:14.938075812Z 02-07-2023 13:21:14.937 INFO [manage-work-group-3] [] c.t.s.o.m.d.d.m.d.a.S(): syncConfigurationForSameVersion to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(28-87-BA-84-BE-D2), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={}))
2023-02-07T12:21:14.969437753Z 02-07-2023 13:21:14.968 INFO [adopt-work-group-6] [] c.t.s.o.m.d.d.m.d.a.H(): Ap OmadacId(785a6468cb95d473061d8c2527663044) SiteId(62fe9d90272796106cf9cf93) DeviceMac(34-60-F9-53-7B-D2) adopt[auto=true] ok
2023-02-07T12:21:15.033316908Z 02-07-2023 13:21:15.029 INFO [adopt-work-group-6] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-53-7B-D2)
2023-02-07T12:21:15.202697422Z 02-07-2023 13:21:15.201 INFO [manage-work-group-5] [] c.t.s.o.m.d.p.t.c(): Device 34-60-F9-53-7B-D2 OmadacId(785a6468cb95d473061d8c2527663044) changed to status CONNECTED, which don't need to handle.
2023-02-07T12:21:15.223438533Z 02-07-2023 13:21:15.222 INFO [server-comm-pool-4] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-53-7B-D2)
2023-02-07T12:21:15.277901519Z 02-07-2023 13:21:15.272 INFO [manage-work-group-6] [] c.t.s.o.m.d.d.m.i.d(): first inform send same version config to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-53-7B-D2)
2023-02-07T12:21:15.381189843Z 02-07-2023 13:21:15.380 INFO [manage-work-group-7] [] c.t.s.o.m.d.d.m.d.a.S(): syncConfigurationForSameVersion to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-53-7B-D2), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={}))
2023-02-07T12:21:17.519339229Z 02-07-2023 13:21:17.518 INFO [adopt-work-group-3] [] c.t.s.o.m.d.d.m.d.a.H(): Ap OmadacId(785a6468cb95d473061d8c2527663044) SiteId(62fe9d90272796106cf9cf93) DeviceMac(34-60-F9-FA-C3-CE) adopt[auto=true] ok
2023-02-07T12:21:17.537995501Z 02-07-2023 13:21:17.537 INFO [adopt-work-group-3] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-FA-C3-CE)
2023-02-07T12:21:17.853604822Z 02-07-2023 13:21:17.852 INFO [adopt-work-group-5] [] c.t.s.o.m.d.d.m.d.a.H(): Ap OmadacId(785a6468cb95d473061d8c2527663044) SiteId(62fe9d90272796106cf9cf93) DeviceMac(28-87-BA-84-BE-AE) adopt[auto=true] ok
2023-02-07T12:21:18.295707784Z 02-07-2023 13:21:17.879 INFO [adopt-work-group-5] [] c.t.s.o.m.d.d.m.a.c(): send empty setting to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(28-87-BA-84-BE-AE)
2023-02-07T12:21:18.296033338Z 02-07-2023 13:21:18.040 INFO [manage-work-group-10] [] c.t.s.o.m.d.p.t.c(): Device 28-87-BA-84-BE-AE OmadacId(785a6468cb95d473061d8c2527663044) changed to status CONNECTED, which don't need to handle.
2023-02-07T12:21:18.296140504Z 02-07-2023 13:21:18.059 INFO [server-comm-pool-5] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(28-87-BA-84-BE-AE)
2023-02-07T12:21:18.296212416Z 02-07-2023 13:21:18.090 INFO [manage-work-group-11] [] c.t.s.o.m.d.d.m.i.d(): first inform send same version config to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(28-87-BA-84-BE-AE)
2023-02-07T12:21:18.322077192Z 02-07-2023 13:21:18.321 INFO [manage-work-group-12] [] c.t.s.o.m.d.d.m.d.a.S(): syncConfigurationForSameVersion to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(28-87-BA-84-BE-AE), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={}))
2023-02-07T12:21:27.517553825Z 02-07-2023 13:21:27.516 INFO [manage-work-group-5] [] c.t.s.o.m.d.p.t.c(): Device 34-60-F9-FA-C3-CE OmadacId(785a6468cb95d473061d8c2527663044) changed to status CONNECTED, which don't need to handle.
2023-02-07T12:21:32.531529828Z 02-07-2023 13:21:32.529 INFO [server-comm-pool-0] [] c.t.s.o.m.d.d.m.i.a(): got first inform of OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-FA-C3-CE)
2023-02-07T12:21:32.549726593Z 02-07-2023 13:21:32.548 INFO [manage-work-group-10] [] c.t.s.o.m.d.d.m.i.d(): first inform send same version config to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-FA-C3-CE)
2023-02-07T12:21:32.614816047Z 02-07-2023 13:21:32.613 INFO [manage-work-group-11] [] c.t.s.o.m.d.d.m.d.a.S(): syncConfigurationForSameVersion to OmadacId(785a6468cb95d473061d8c2527663044) DeviceMac(34-60-F9-FA-C3-CE), result:SendDeviceMsgResult(success=true, deviceResponse=BaseConfigRespBody(sequenceId=1, errcode=0, configVersion=null, additionalProperties={}))
2023-02-07T12:21:58.837330678Z 02-07-2023 13:21:58.836 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Cloud Access Not Enable, cancel sync cloud user.
2023-02-07T12:21:58.848696972Z 02-07-2023 13:21:58.847 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Start checking last firmwares of managed devices.
2023-02-07T12:21:58.851322821Z 02-07-2023 13:21:58.850 INFO [check-update-work-group-0] [] c.t.s.o.m.c.b(): Checking Device Last Firmware...
2023-02-07T12:22:00.137267697Z 02-07-2023 13:22:00.136 INFO [check-update-work-group-0] [] c.t.s.o.m.c.f(): Checking last firmwares of managed devices succeed.
2023-02-07T12:22:06.405605890Z 02-07-2023 13:22:06.404 INFO [https-jsse-nio-8043-exec-6] [] c.t.s.o.c.u.d.a(): list local interface macs: [00-11-32-AA-B1-EB, 02-42-E7-E9-25-F8, 02-42-2B-C0-B5-CE, 02-42-16-C6-D6-6C, 02-42-A2-65-D1-B3, 02-42-FB-2A-8A-3F, 02-42-0F-41-8F-5E, 02-42-F4-39-23-04, 02-42-EF-2F-98-5B]
```

mbentley, 2023-02-07:
If you try a different browser/incognito mode, do you see the same?  I haven't seen the same behavior on my end but I did see some folks say that they needed to clear their cache or hard refresh for the controller web interface to work.

baldisos, 2023-02-07:
I feel dumb now. I cleared the Firefox cache with CTRL+SHIFT+R becaue i already tought it was something like this. Now i installed Chromium and it worked like it should. Now after clearing the Firefox Cache from the Settings, it works again. Thanks. Sometimes things are easy, but not as clear to see.

closed 2023-02-07 by baldisos